fkanehiro/hrpsys-base #456: RobotHardwareのRemoveForceSensorOffsetで追加オフセットを与えるようにする

snozawa, 2015-01-28:
力センサが時々ゼロ点ずれしている状況があるため、現在時々removeForceSensorOffsetを使っています。
RobotHardwareのRemoveForceSensorOffsetで追加オフセットを与えたい状況がでてきましたので、ご相談させてたきたくissueをきりました。

状況として、絵をかいてみました。簡単のため、センサ姿勢が単位行列で、絶対座標系のXYZとセンサ座標系があうような状況をかいています。また、静的な状況としています。
[image]

case1が``removeForceSensorOffset()``前です。
ゼロ点ずれのドリフト分と、ハンドやフットの自重分が値にはいってきます。

case2が現行の``removeForceSensorOffset()``の実行後です。
``removeForceSensorOffset()``が呼ばれたときの力センサの値をオフセットとすると、
ドリフト分だけでなくてハンドやフットの自重分もオフセット分にはいってきます。

そうすると``read_force_sensor``でよめる``f_iob``の値は、
この絵の状況では0になり、ハンドやフットを180度回転（Z方向がうえむき）にすると
``f_iob=2mg``となります。
ドリフト分は除去されてますが、自重分がセンサの姿勢によってなかったり倍はいったりして
変な状況になってしまいます。

case3が新しく追加のオフセットを与えた場合の``removeForceSensorOffset()``の実行後です。
例えば、``removeForceSensorOffset(f_off2)``として引数をあたえ、
``f_off2=mg``とします。

そうすると、``read_force_sensor``でよめる``f_iob``の値は
この絵の状況では``mg``となり、ハンドやフットを１８０度回転させると``f_iob=-mg``となります。
こうすると、ドリフト分のみ除去できて、自重分は姿勢によって適切にはいってきます。

ちょっと似たようなことをしているRemoeForceSensorOffsetRTCでは、現状
- case2で自重分までもけしてしまった分を一度もとに戻す(https://github.com/fkanehiro/hrpsys-base/blob/master/idl/RemoveForceSensorLinkOffsetService.idl#L16のforce_offset, moment_offset)
- その上で、姿勢によってかわる自重分を補償する

としています。case３のようにすると、前者のforce_offset, moment_offsetがいらなくなります。

少しまとめるますと、RobotHardwareの出す力ポートや``read_force_sensor``の読み値の仕様として、
- 「ドリフト分は除去されているが、姿勢によって自重がはいったりはいってなかったりする」  （ 現行のcase2） 

ものから
- 「ドリフト分は除去されており、外力分（自重ふくむ）が適切にはいっている」（新しいcase3）  

とかえてたいです、
そのために``removeForceSensorOffset``でオフセット（というより自重補償ぶん）を追加でセットできるようにしたいです、という要望になっております。

また、``removeForceSensorOffset``に引数追加するのはStableRTCのAPIが壊れるので、実際は
``removeForceSensorOffset (void)``はそのままで``removeForceSesnorOffsetWithMassOffset (double* mass_offset)``のようにidlに追加することを考えています。

よろしくお願いいたします。

snozawa, 2015-01-28:
長文で失礼しましたが、細くさせていただきます。

もともとの目的は
「力センサの出力値として、ドリフト分がなくてハンドフットの自重分が入ってない、純粋な外力を計測したい」
です。

このため、今は
- RobotHardwareRTCでドリフト分とremove時の自重分がキャンセルされる（全てセンサ座標系）
- RemoveForceSensorOffsetRTCで一旦上記自重分を戻し、センサの姿勢推定値も使い絶対座標系での自重分を差し引く。

としております。

これをする方法として、他には
1. RobotHardwareで上記をすべてやる(その１)。ロボットの姿勢推定値をRobothardwareに入力するとできる。
2. RobotHardwareで上記をすべてやる(その２)。RobotHardwareの中でロボットの姿勢推定計算する。
3. 上記をRobotHardwareRTCとRemoveForceSensorOffsetRTCとで分ける。そのために、`removeForceSensorOffset()``関数に引数を追加する。

がありそうで、このissueで提案させていただいてるのは3になります。

1,2のメリットは、RobotHardwraeから出力される力センサが、ドリフト・自重分含まない純粋な外力になっていて、
RobotHawrdwareの外の人からみるとシンプルなしくみになってます。

問題としては、1だと１制御周期分おくれるので、ちょっとやりたくないです。
2ですと、RobotHardwareがKalmanFilterRTCなどの機能ももつ必要があり、
姿勢推定アルゴリズムやパラメータを更新したくしづらくなりますし、
やりすぎな感があります。
そのため、３のような方法をいかがでしょうか、という提案をさせていただいておりあmす。

fkanehiro, 2015-01-28:
実はそういった（力センサの本来の出力値を取りたい）という要望が他からも来ているので
そう言った機能を実現するのはよいと思うのですが、書かれている方法がよく分からないです。

オフセット値を設定する機能はどこに追加するのでしょうか？
RobotHardwareService?RemoveForceSensorLinkOffsetService?
RobotHardwareServiceに追加したとしても引き続きRemoveForceSensorOffsetRTCが
必要？であれば現状と変わらないのでは？

snozawa, 2015-01-28:
わかりにくく失礼しました。

上記で3
> オフセット値を設定する機能はどこに追加するのでしょうか？
オフセット値を設定する機能で、今までRobot
https://github.com/fkanehiro/hrpsys-base/pull/458
https://github.com/fkanehiro/hrpsys-base/issues/456#issuecomment-71817704

snozawa, 2015-01-28:
説明わかりにくく、大変失礼しました（少し自分でも頭がこんがらがりながらかいてた部分もありました）。

> RobotHardwareServiceに追加したとしても引き続きRemoveForceSensorOffsetRTCが 必要？であれば現状と変わらないのでは？

おっしゃるとおり、このPRで提案している方法（上記の3）では
RobotHardwareRTCとRemoveForceSensorLIinkOffsetRTCを両方つかいます。
最終的なRemoveForceSensorLinkOffsetの出力力ポートの値はかわらないという点では、現状とかわらないです。
一方、同じようなパラメータがRobotHardwareRTCとRemoveForceSensorLIinkOffsetRTCの両方にいまあって、removexxxx時にそれらの整合性とる必要があったのが必要なくなる、という点が変わります。
具体的には、現行ではRrobotHardwareRTCのiobのforce_offsetの一部分と、RemoveForceSensorLinkOffsetのforce_offset, moment_offset([IDL](http://fkanehiro.github.io/hrpsys-base/de/ddc/structOpenHRP_1_1RemoveForceSensorLinkOffsetService_1_1forcemomentOffsetParam.html "IDL")のforce_offset, moment_offset）がかぶっています。
case2のようにするとRobotHardwareRTCのfoce_offsetで余計に自重分が差し引かれて、RemoveForceSesnorLinkOffsetRTCではそれをforce_offset, moment_offsetで一旦それを戻して、、、としています。

> オフセット値を設定する機能はどこに追加するのでしょうか？

ゼロ点だしのためのオフセットはRobotHardwareの方でもち、
ハンド・フットの重さなどを差し引く部分はRemoveForceSensorLinkOffsetに持つことになります。
RobotHardwareRTC、RemoveForceSensorLinkOffsetRTCでそれぞれのオフセット除去が適切に官僚したあとでは、
- RobotHardwraeの出力は、 ドリフト分のみない値がでてきます。ハンド・フット分の重量等ははいった値がでてきます。
- RemoveForceSensorLinkOffsetの出力は、ドリフト分がなく、かつハンド・フット分の重量等もなく純粋な外力がでてきます。

まだ動作確認等しておらず大まかにこんなかんじですというものですが、#458にRobotHardware、RemoveForceSensorLinkOffsetRTCそれぞれに変更を加えたものをPRしました。

fkanehiro, 2015-01-29:
なるほど分かりました。
RemoveForceSensorLinkOffsetの機能を理解していなかったので話が通じにくかったようです。

YoheiKakiuchi, 2015-01-29:
僕もよくわかっていないのですが、RobotHardwareからはiobの読み値を何もせずにそのまま出して、
RemoveForceSesnorLinkOffsetRTCに必要な機能は全部持たせるのではダメでしょうか？

snozawa, 2015-01-29:
> 僕もよくわかっていないのですが、RobotHardwareからはiobの読み値を何もせずにそのまま出して、
> RemoveForceSesnorLinkOffsetRTCに必要な機能は全部持たせるのではダメでしょうか？

これの方法は（ほぼ）可能で、実際現状のRobotHardwareRTC+RemoveForceSensorLinkOffsetRTCの構成で実現されています。
その際に同じような変数・機能がRobotHardwareRTCとRemoveForceSensorLinkOffsetRTCとに
あるので、ゼロ点だしオフセット分の変数・機能の重複をなくしてみたい、というのが本issueの意図でした。

実用上は、RobotHardwareRTC側のゼロ点だし機能を（あまり）つかわないようにすれば、現状問題ありません。

snozawa, 2017-05-17:
https://github.com/fkanehiro/hrpsys-base/pull/1132
で解決しましたので、closeします。

closed 2017-05-17 by snozawa