fkanehiro/hrpsys-base #475: RobotHardwareにcalibrateAccelSensor, calibrateGyroSensor関数を追加する

snozawa, 2015-02-09:
RobotHardwareにcalibrateAccelSensor, calibrateGyroSensor関数を追加したいと思いました。

状況としては以下になります。
calibrateInertiaSensorですとgyro, accel両方がオフセット除去されますが、
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/robot.cpp#L175
どちらか片方だけオフセット除去したいことがあります。
更に具体的には、accelでなくgyroだけオフセット除去したいことがあります。
（accelはロボット自体や地面の傾斜をきにする必要がありますが、gyroは静止していればドリフト分が除去できるため）

APIをbreakしないとすれば、calibrateInertiaSensor(gyro+acc)を残しつつ、
calibrateAccelSensor, calibrateGyroSensorを追加するのはいかがでしょうか。

よろしくお願いいたします。

fkanehiro, 2015-02-10:
確かにそういう場合もありそうですね。
よろしいのではないでしょうか。

snozawa, 2015-02-10:
ありがとうございます。
PR作成してみます。

@k-okadaさん
ふと気づいたのですが、hrpsys-baseはRobotHardwaraを使っていないので、
もしかするとRobotHardwareは`315.1.9`とのIDL互換性チェックから漏れてたりしますでしょうか。

k-okada, 2015-02-11:
https://github.com/fkanehiro/hrpsys-base/blob/master/.travis.sh#L57
でチェックしてあるでしょうか？

snozawa, 2015-02-13:
ありがとうございます．
末尾追加がチェックされてそうですね．

こちら一旦別な方法で解決できましたので，再度どうしても必要となりましたらreopenさせていただきたいと思います．

closed 2015-02-13 by snozawa