fkanehiro/hrpsys-base #480: waitInputConfirm関数の例外について

snozawa, 2015-02-12:
waitInput.pyのwaitInputConfirm関数の結果（OKかcancelか）で、
cancelが押されたときに例外がraiseされるようになっているのはなぜでしょうか。
https://github.com/fkanehiro/hrpsys-base/blob/master/python/waitInput.py#L7

例えばhrpsys_configなどの利用例をみると、
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1519
OKならTrue, cancelならFalseをreturnするようにしても良いように見受けられますが、あってますでしょうか。

ロボット体内などdisplayがないところでも、外部PCなどdisplayがあるところでも
うごくようにプログラムをかこうとして、「displayがないところ」かどうかの判定を
waitInputConfirmからの例外判断しようとして、ふと思いました。

よろしくお願いいたします。

k-okada, 2015-02-12:
hiroでは
https://github.com/start-jsk/rtmros_hironx/blob/hydro-devel/hironx_ros_bridge/src/hironx_ros_bridge/hironx_client.py#L603
としています．体内でpythonは走らせていない気がしますが，riaseされたものをcatchしているのかな

Cc: @130s

snozawa, 2015-02-13:
ありがとうございます。
hrpsys_config.pyと基本的には同じになってますね。

お伺いしたいものとしては、
A.
この部分で、
```
try:
   waitInputConfirm
except:
   xxx
```
となっている部分が
```
if waitInputConfirm():
   yy
else:
   xx
```
のように返り値＋if文だと問題ありますでしょうか。
OKボタン、cancelボタンの結果を取得するのに、例外をraiseしてcatchすることにしているのは何か理由ございますでしょうか。

B.
そもそもAの質問の意図としましては、OKかcancelかを例外+catchでなくて返り値＋if文でとるようにすると、
例外+catchを他の用途（体内でdisplayがない）で使えるのでは、と思いました。
もともとやりたかったことは、displayのない体内でもdisplayのある体外でも、両方pythonが動くようにする、でした。
pythonでdisplayがあるかないか、のif文がかければAをしなくても目的が達成されます。
（ちょっとpythonなれしておらず、displayがあるかどうかのif文がぱっとでてこない状況でした）

fkanehiro, 2015-02-13:
例外となっている理由はGrxUIの仕様と合わせるためです。
なぜ例外にしたのかは川角さんに聞いてみないとわからないですね。

snozawa, 2015-02-13:
わかりました。
手元ではpopenで調べるようにして解決しました。
```
        xp=os.popen("xset q 2>&1 | grep unable");
        display_available = (xp.read()=='')
        xp.close()
```

ありがとうございました。

closed 2015-02-13 by snozawa

k-okada, 2015-02-13:
ディスプレイがないと waitInputConfirm でerrorがraiseされるから，A)でも動くんじゃないかな．

snozawa, 2015-02-13:
displayがないなら無いなりの挙動になってほしいですが，そうならないと思います．

例えば，displayがない体内のような場合にYかNを端末に打ち込むようにする，というのを試してみます．
今のservoOnの箇所の
```
try:
    waitInputConfirm(\
           '!! Robot Motion Warning (SERVO_ON) !!\n\n'
           'Confirm RELAY switched ON\n'
           'Push [OK] to switch servo ON(%s).' % (jname))
except: # ths needs to change
    self.rh_svc.power('all', SWITCH_OFF)
    xxxx # ここが追加部分です
    raise
```
のxxxに端末に打ち込むような処理をいれるということだと思いますが，
1. displayがなくてかつxxxを実行したいときに，そのうえの処理が強制的に行われてしまう．上記ではpowerOffされる．
2. displayがあってかつcancelボタンを押した時に，xxxのようないらない処理が行われてしまう．

となって，結局cancelボタンが押されたexceptionとdisplayがないexceptionがごっちゃになるので，
処理するのが難しそうに思います．
(1はneeds to changeとかいてあり，みたかんじいらなさそうですが，2の問題は残ります)

k-okada, 2015-02-13:
あああ，displayがない場合，というのは，もうy/nを聞かずに進める，というのがhiroでの使い方だった気がします．
足があるとやっぱy/nは聞きたいわけですね．なるほど．

snozawa, 2015-02-13:
> あああ，displayがない場合，というのは，もうy/nを聞かずに進める，というのがhiroでの使い方だった気がします． 足があるとやっぱy/nは聞きたいわけですね．なるほど．

そうですね．それプラス，displayがない場合はwaitInputConfirmでぬけちゃうので，y/n聞かずに進まず，servoOn自体からexceptで抜けてしまいます．

k-okada, 2015-02-14:
ユーザに聞くか，あるいは強制的にすすめるか，どちらにしたらいいかは難しいですね．
dashboardみたいにスクリプト（プログラム）からservoonするなら強制でいい気がしますし．
ユーザからインタプリタ的に使うなら聞きたい気がしますし．

ということで，元々の @snozawa の提案でいい気がしてきました．