fkanehiro/hrpsys-base #485: hrpsys_config.pyのservoOn関係の挙動に関して

snozawa, 2015-02-14:
hrpsys_config.pyのservoOn系の関数に関して、いくつか質問がございます。
（少々雑多になってしまい申し訳ございません）

1. servoOn/servoOff関数で単軸servoOn/Offをサポートするかどうか
　今servoOn/Off関数でjnameで軸を選べるようなインターフェースにしていますが、こちらお使いの方はいらっしゃるでしょうか。
　https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1580
　のようなものがあったり、goActualなど全軸分にのみ対応しているものがあるので、単軸での運用は現状のバージョンではできないと思います。
　またこちらでの運用上、単軸でservoをいじることはめったにないです
　あるとしても上記のgoActualが単軸対応していない問題がございますので、直接goActual相当の別な処理をして、rh_svc.poewr, rh_svc.servoを実行しています
　今動いてない機能ですので、servoOn/Off関数から消しても良いのでは、と思いました（if文などが少なくて扱いやすい）

2. isServoOnの関節指定部分
　isServoOnでjnameを指定した場合、
　https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1580
　の部分で関節名からインスタンスをもってきてますが、明示的に
　HrpsysConfiguratorに関節名などに追加しないかぎり動かないコードになってると思います。
　いまこれが動いてるロボットはありますでしょうか
　こちらのロボットはすべてこれに対応してなくて、拝見するかぎりはhironx_client.pyでも動かないきがしています。

よろしくお願いいたします。

snozawa, 2015-03-18:
追加で失礼します．

3 . servoOnダイアログの挙動
　servoOn時にダイアログでservoをいれるかどうかをきかれます．
　その後OKにするとサーボがはいりますが，cancelにするとpowerを切られてしまいます．
　https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1579
　cancelになったときはとくにサーボやパワーに操作せずそのまま関数returnしてくれるほうが都合が良いのですが，こちらのpower　offはどういったものを意図されておりますでしょうか．

よろしくお願いします．

k-okada, 2015-03-18:
元々のservoOnの実装は @emijah さんでしょうか．

snozawa, 2015-03-18:
@k-okada さん，ありがとうございます．

@emijahさん，上記３点，いかがでしょうか．
よろしくお願いいたします．

emijah, 2015-03-23:
見落としていました。昼までに回答します

emijah, 2015-03-23:
1の部分サーボONについてですが、奈良先でHRP-2を座らせて使用している場合などの特例に合わせて手を加えていました。正式なhrpsys_baseでは必要な人がいなかったため、インタフェースはこちらの特例に合わせてあるが実態がないというものに確かになっています。いろいろな使用例に合わせて部分サーボONも正式にサポートできると良い気がします。

2. isServoOnを通してスクリプトで場合に応じて何かをするロボットは今はいないと思います。今後1が正式さポートされれば、使う場合がないわけではなさそうな気はしています。

3. 単に切り忘れを防ぐための安全を考慮してそうしてあったと思います。事故例はパッとは思い浮かびません。

snozawa, 2015-03-24:
@emijahさん
ご回答いただきありがとうございます．

1にかんして

> 1の部分サーボONについてですが、奈良先でHRP-2を座らせて使用している場合などの特例に合わせて手を加えていました。
> 正式なhrpsys_baseでは必要な人がいなかったため、インタフェースはこちらの特例に合わせてあるが実態がないというものに確かになっています。
> いろいろな使用例に合わせて部分サーボONも正式にサポートできると良い気がします。

部分サーボについては，承知しました．
おそらく，「座らせて上半身のみサーボON」な状況としますと，
　jointAのサーボをいれて，次にjointBをいれる
といったような部分サーボONでなくて
　jointAのみサーボをいれて，以降jointBはずっといれない
といった形かと思います．
そうすると，挙動としては，一度いずれかの軸がサーボがはいっていれば
以降サーボONしないのが安全な挙動と思いますので（goActualが対応してないので），
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1566
の部分は
` if self.isServoOn():`
でなくて
` if self.isServoOn("some"):`
とするのはいかがでしょうか．

3に関して
>    単に切り忘れを防ぐための安全を考慮してそうしてあったと思います。事故例はパッとは思い浮かびません。

（事故というほどではないですが）事故例は実際こちらではありました．
servoONを誤って二回よんでしてしまい，１個目のダイアログでは正常にserovONまでいき，
脚型ロボットを地面までおろしました．
次に２個目のダイアログを閉じようとしてcancelか閉じるボタンをおすと
強制的にswitch_offになり，ロボットが脱力する状況になりました．
servoON関数のダイアログのキャンセルでpower offが呼ばれるのはやはり危険なきがしています
（多くのユーザが，servoOnの中でoffがよばれるとわからないため，また
ダイアログが２個でたときに現状off以外の方法がなくなってしまうため）