fkanehiro/hrpsys-base #490: servoのアラームでemergencysignalを出したい

YoheiKakiuchi, 2015-02-18:
サーボのアラームでemergency signalを出したいのですが、
iob.hのservoアラームのビット定義[1] は12bit分されていて、servo alarmのmaskとshift [2] からは
16bitがmaskされており、ほかのshiftからも16bit 使えるように見えます。

この未定義の4bitのうち１ビットを SS_EMERGENCYとさせていただき、
RobotHardwareの checkemergency [3] か servo alarmをセット[4]しているところで
このビットが立っていたら、emergency signalを出すというようにしたいと考えております。

まずは、iob.hのアラームビット定義を追加して良いかと 、
そのアラームによってemergency signalを送るようにする変更について意見をお願いします。

[1] iob.hのservoアラームのビット定義（12bit)
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/io/iob.h#L33-L46

[2] servo alarmのmaskとshift
https://github.com/fkanehiro/hrpsys-base/blob/master/idl/RobotHardwareService.idl#L9-L18

[3] checkemergency
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/RobotHardware.cpp#L220

[4] servo alarmのセット
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/RobotHardware.cpp#L289

YoheiKakiuchi, 2015-03-09:
この件に関連しまして、DataLoggerの以下の部分のコードはどのような動作をするように意図されているのでしょうか？
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/DataLogger/DataLogger.cpp#L243

```ec_id == 0``` のときは、emergency_signalを読んで、それ以外のときはログをとるようになっている
ように読めます。
しかし、手元のテストでは ```ec_id == 0``` にならないためemergency_signalを出してもログが記録されないようです。
使い方が間違っているのでしょうか？

fkanehiro, 2015-03-09:
通常DataLoggerは他のRTCと直列実行して、他のRTCのECで駆動されていると思います。
他のRTCのECで駆動される場合はec_idは1000とかになりますね。
で、DataLoggerをDataLogger自身のECでも走らせる（ec_id=0)と非常信号を受け取ってログを保存するようになります。
DataLoggerをDataLogger自身のECで走らせるには、
log.start(log.owned_ecs[0])
とかします。

k-okada, 2015-03-09:
DataLoggerをDataLogger自身のECで走らせる／走らせない，の違いは
どのようなときにつかい分けたらいいんでしょうか？

fkanehiro, 2015-03-09:
非常信号で自動的にログを取りたければ走らせる、別に取らなくてよければ走らせない、でしょうか。

YoheiKakiuchi, 2015-03-11:
ありがとうございます。
以下で、emergency signalからログが保存できるように出来ました。
https://github.com/fkanehiro/hrpsys-base/pull/510

k-okada, 2015-03-13:
> 非常信号で自動的にログを取りたければ走らせる、別に取らなくてよければ走らせない、でしょうか。 

だとすると，常に別ECで走らせておけばいい気がしますが，それだとやはり取りこぼしがあったりする，とかあるんでしょうか？

fkanehiro, 2015-03-14:
?別ECというのはどれのことだろう？

k-okada, 2015-03-14:
すいません．別ECというのは
>、DataLoggerをDataLogger自身のECでも走らせる（ec_id=0)と非常信号を受け取ってログを保存するようになります。
のことです

fkanehiro, 2015-03-14:
常に走らせておいて問題ないと思いますよ。

k-okada, 2015-03-31:
#510 でクローズでしょうか？

k-okada, 2015-04-05:
すいません．これ，テストに通っていないは，純粋にこのPRを入れるとプログラムが動かないのですが，具体的には`self.log.start(self.log.owned_ecs[0])`から帰ってきません．

1)上のコメントでは，
```
DataLoggerをDataLogger自身のECで走らせるには、
log.start(log.owned_ecs[0])
```
と書いてありますが，PR(https://github.com/fkanehiro/hrpsys-base/pull/510/files)では，
```
+        self.log.owned_ecs[0].start()
+        self.log.start(self.log.owned_ecs[0])
```
どちらがただしいでしょうか？僕の環境だと，
` self.log.owned_ecs[0].start()` だとRTCがactiveにならず，`self.log.start(self.log.owned_ecs[0])`から帰ってこないです．これをする場所はserializeする前とか，後とか条件はあるでしょうか？一応，いろいろ場所を変えてみましたが，効果なさそうです．

2) DataLoggerをDataLogger自信のECで走らせる場合には，
https://github.com/fkanehiro/hrpsys-base/blob/6c6205c938c88ad1839e768bda01254136cb9d3d/python/hrpsys_config.py#L446
のr.start() は止めるべきでしょうか？

k-okada, 2015-04-05:
あと，
https://github.com/fkanehiro/hrpsys-base/blob/6c6205c938c88ad1839e768bda01254136cb9d3d/python/hrpsys_config.py#L444
の
```
        rtm.serializeComponents(rtcList)
```
のrtcListにself.logは入っていていいのでしょうか？

fkanehiro, 2015-04-05:
おっとすいません。
PRの方が正しくて、

       self.log.owned_ecs[0].start()
       self.log.start(self.log.owned_ecs[0])

をself.logをserializeComponent()した後で、両方やるのが正しいと思います。
前者はEC自体をstartするもので、後者は指定したECでRTCを駆動すること指示するもので意味合いが
違います。
通常ECは作られた時に勝手にスタートするのですが、serializeComponent()が呼ばれた時にRTCを
駆動しているEC（通常はRTCとセットで作られたowned_ecs[0]）は止めてしまうので、再スタートする
必要があり、1行目が必要となります。
https://github.com/fkanehiro/hrpsys-base/blob/6c6205c938c88ad1839e768bda01254136cb9d3d/python/rtm.py#L437

ログを取りながら、エマー信号でログを自動的に保存するには、ロガーをserializeComponent()して
別のRTCのECで駆動しておきながら、ロガー自身のECでエマー信号を監視する、ということになります。

k-okada, 2015-04-05:
うーん．だとすると，２行目から帰ってこないんですが，，，なにかちがうんですかねぇ．

k-okada, 2015-04-05:
https://github.com/fkanehiro/hrpsys-base/blob/master/python/rtm.py#L107
で，
```
    def start(self, ec=None):
        if ec == None:
            ec = self.ec
        if ec != None:
            ec.activate_component(self.ref)
            while self.isInactive(ec):
                time.sleep(0.01)
```
となっていますが，`self.isInactivate(self.ec)`とするとかありますか？

fkanehiro, 2015-04-05:
serializeした後は、DataLoggerのself.ecは他のRTCのecになっているはずで、ここはこのままで
いいはずです。

以下のようなスクリプトで試す限りは返ってこなくなる、ということはなさそうです。

import rtm

rtm.nsport = 2809
rtm.initCORBA()

ms = rtm.findRTCmanager()
print "ms=",ms

ms.load("NullComponent")
np = ms.create("NullComponent")
print "np=",np

ms.load("DataLogger")
dl = ms.create("DataLogger")
print "dl=",dl

rtm.serializeComponents([np, dl])

np.start()
dl.start()
print "here"

dl.owned_ecs[0].start()
dl.start(dl.owned_ecs[0])

k-okada, 2015-04-05:
うーん．こちらの環境だと，今度はこのプログラムだと，`np.start()`が帰ってこないです．
```
$ ./hoge.py 
<module 'rtm' from '/home/k-okada/catkin_ws/ws_rtmros/devel/lib/python2.7/dist-packages/hrpsys/rtm.py'>
configuration ORB with  kokada-t440s : 2809
ms= <rtm.RTCmanager instance at 0x7f2a8e91fa28>
np= <rtm.RTcomponent instance at 0x7f2a8e91fcf8>
dl= <rtm.RTcomponent instance at 0x7f2a8e91ffc8>
start
<rtm.RTcomponent instance at 0x7f2a8e91fcf8>
[<OpenRTM._objref_ExtTrigExecutionContextService instance at 0x7f2a8e91fd40>]
```
実行したプログラムは以下
```
print "start"
print np
print np.owned_ecs
np.start()
print "np"
```
ちなみに，VERBOSEにしてとったログを
https://gist.githubusercontent.com/k-okada/32ef62ec2fc516ca311d/raw/rtc5265.log
に起きました

k-okada, 2015-04-05:
```
ORBgiopMaxMsgSize=2147483648 LANG=C ROS_HOME=/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot /home/k-okada/catkin_ws/ws_rtmros/devel/lib/hrpsys/hrpsys-simulator /home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.xml -o manager.is_master:YES -o naming.formats:%n.rtc -o exec_cxt.periodic.rate:1000000 -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/lib -o manager.modules.preload:HGcontroller.so -o manager.components.precreate:HGcontroller -o exec_cxt.periodic.type:SynchExtTriggerEC -f /home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.RobotHardware.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.RobotHardware.500conf -o example.SequencePlayer.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.ForwardKinematics.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.ImpedanceController.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.AutoBalancer.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.StateHolder.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.TorqueFilter.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.TorqueController.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.ThermoEstimator.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.ThermoLimiter.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.VirtualForceSensor.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.AbsoluteForceSensor.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.RemoveForceSensorLinkOffset.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.KalmanFilter.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.Stabilizer.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.CollisionDetector.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.SoftErrorLimiter.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o example.HGcontroller.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.500.conf -o corba.nameservers:localhost:2809 -endless -realtime
```
を立ち上げています

fkanehiro, 2015-04-05:
NullComponent0のECのタイプがSyncExtTriggerになっているけれども、トリガが
送られているだろうか？

k-okada, 2015-04-05:
トリガはどうやって送るのでしょうか？

k-okada, 2015-04-05:
rtcdの時は `-o exec_cxt.periodic.rate:1000000 -o exec_cxt.periodic.type:SynchExtTriggerEC `で，
hrpsys-simulatorのときは，`-o exec_cxt.periodic.rate:500 -o exec_cxt.periodic.type:PeriodicExecutionContex` とかでしょうか？

k-okada, 2015-04-05:
ちがいますね．https://code.google.com/p/hrpsys-base/issues/detail?id=14
rtcdのときはhrpExecutionContext でした．シミュレーションでこの機能をつかいたかったら，
log.confみたいなところで，ECのタイプをきめなければいけないのでしょうか．
一回createCompしたらECのタイプは変更できないのでしょうか．
とりあえず，
```
        if self.log and not isinstance(self.log.owned_ecs[0], OpenRTM._objref_ExtTrigExecutionContextService):
            self.log.owned_ecs[0].start()
            self.log.start(self.log.owned_ecs[0])
```
としました．

ところで，この場所はいまsetupLoggerの最後ですが，ここでいいでしょうか（あるいはここでないといけないでしょうか）activateCompsの最後でもいいかなとも思ったんですが．

fkanehiro, 2015-04-05:
その変更だとシミュレーション時はエマーでログを保存する、ということはできませんがそれでいいでしょうか。

場所についてはロガーのセットアップとも言えるし、コンポーネントをアクティベートしてるとも言えるし、ほぼ同じ場所で呼んでるので、まぁお好みで。serializeComponents()の後であればよいです。

hrpExecutionContextとかSyncExtTriggerとかは実時間スレッドを模擬するためのもので、システムの中で使う数は（こちらでは）限られているので、こちらではrtc.confでこれらの実行コンテキストを指定することはせずに、serializeComponentsの引数のコンポーネントリストの最初のコンポーネントに対応するxxx.confでこれらの実行コンテキストを指定するようにしています。どこにも紐付けられてない実行コンテキストを作ったり、コンポーネントに紐付けられている実行コンテキストを作りなおしたりする手段がないようなので。

k-okada, 2015-04-06:
相変わらずわかっていなくてもうしわけないですが，いまみてみると，SampleRobot.RobotHardware.confは（serializecomponentnの最初のRTCはRobotHardware)
```
exec_cxt.periodic.type: hrpExecutionContext
exec_cxt.periodic.rate: 500
```
になっていて，実行時の`-o exec_cxt.periodic.rate:1000000  -o
exec_cxt.periodic.type:SynchExtTriggerEC`
を抜いたらうまく動いていました．これでいいのでしょうか？SampleRobot.RobotHardware.conf
にSynchExtTriggerEC とかかなくていいでしょうか？

また，SampleRobot.RobotHardware.confに何も書かず，`-o exec_cxt.periodic.rate:1000000
 -o exec_cxt.periodic.type:SynchExtTriggerEC`
を抜いても動いているようですが，なにか不都合が有りますでしょうか．

k-okada, 2015-04-06:
あと，どのケースでも owned_ecsは
`[<OpenRTM._objref_ExtTrigExecutionContextService instance at
0x7f72f0c46c68>]`
なのですが，これがどこからトリガがやってくるようになっているかはわかるものなのでしょうか？
つまり，運良く誰かがトリガをかけてくれそうなら自身のECで起動し，そうでなければ，serializeのECでよい，
みたいにできるでしょうか．

fkanehiro, 2015-04-06:
う〜む、いまいちどういう使い方をしようとしているのかがわからないのですが、
RobotHardwareがある、ということは実機モードなのかな？
実機モードであれば、
`-o exec_cxt.periodic.rate:1000000 -o exec_cxt.periodic.type:
SynchExtTriggerEC`
をrtcdに渡す必要はないと思います。

実行コンテキストのタイプを指定しないとデフォルトの実行コンテキストで作られるわけですが、
これは非実時間スレッドによる周期実行をするものなので、それでよければよい、実時間でないと
だめであればダメということかと思います。

fkanehiro, 2015-04-06:
どの実行コンテキストもExtTrigExecutionContextServiceを継承しているようなので見分けつかないですね。

k-okada, 2015-04-06:
すいません．実行は
https://github.com/fkanehiro/hrpsys-base/issues/490#issuecomment-89777701
でやっています．hrpsys_simulatorをつかっていて，RobotHardwareはなくて，SampleRobot(Robot)0.rtcができています．なので，ここで，
```
-o example.RobotHardware.config_file:/home/k-okada/catkin_ws/ws_rtmros/devel/share/hrpsys/samples/SampleRobot/SampleRobot.RobotHardware.500conf 
```
としてあるのは何も効果がなく，
`-o example.SampleRoobt\(Robot\).config_file` とかしないとダメなんでしょうかね．
いずれにせよ，シミュレーションのときは，デフォルトでよいのでしょうか？
そもそも，SynchExtTriggerEC をつかわなくてよかった，ということでしょうか．

fkanehiro, 2015-04-06:
> #490 (comment)

シミュレーション時にSyncExtTriggerECが必要かどうかは、シミュレーション世界の時刻に同期した処理（実世界でいうところのリアルタイム処理）が必要か否か次第ですが、普通は少なくとも関節の制御には必要かと思います。

で、SyncExtTriggerECの使用をrtcdの引数またはrtc.confで指定するか、特定のコンポーネントのconfファイルで指定するかは動かすECの内SyncExtTriggerが多いか通常のPeriodicが多いか次第でしょうか。
動かすECが10個あって1個だけSyncExtTriggerにするのなら、rtcdの引数でECのタイプを指定することはせず、そのECが所属するRTCのconfファイルでSyncExtTriggerの使用を指定すればいいですし、逆のケースであれば、rtcdの引数でデフォルトのECのタイプとしてSyncExtTriggerを指定しておき、通常のPeriodicを使いたいRTCのconfファイルでそのデフォルト設定をオーバライドすればよいかと。

今回のケースですとおそらく、リアルタイム(SyncExtTrigger)で動くEC（SampleRobotが持っているもの）と非リアルタイムで動くEC（DataLoggerが持っているもの）が1個づつぐらいしかないと思うのでどちらでもよくて、これまでrtcdの引数でSyncExtTriggerを指定していたのであれば、それは変えずにDataLogger.confを作ってPeriodicの使用を指定してはどうでしょうか。

k-okada, 2015-04-06:
ありがとうございます．
結局ですが
https://github.com/fkanehiro/hrpsys-base/pull/544/files
みたいにして，log_use_own_ecというメンバ変数をいれてシミュレーションで
emregencyの時にログが欲しい人は，これをtrueにしてもらうというふうにしようかと
思っていますがどうでしょうか？
samperobotのサンプルはlog用のconfを作るようにしましたが，はやりそうしていない
応用プログラムはありそうなので，デフォルトはこれまでと同じにしておいたほうが
いいかともいました．（何も言わず帰ってこない，というのは結構気が付かないので）
よろしくお願いいたします．

fkanehiro, 2015-04-06:
ログ自体は他人のecを使って取るので、変数名はちょっと変えた方がいいかもしれませんが、まぁ
よろしいのではないでしょうか。

start()等、いくつか無限に返ってこなくなる可能性のある関数がありますが、これは凶悪なので
タイムアウトするようにする、とかしたほうが良さそうですね。

k-okada, 2015-04-06:
なるほど．log_use_non_serialize_ec ? log_use_owned_ec ?とかでしょうか．

fkanehiro, 2015-04-06:
ネーミングセンスが無いんで、お任せします:-)

k-okada, 2015-04-07:
ありがとうございます．ようやくクローズできます．

YoheiKakiuchi, 2015-04-07:
すいません。テスト環境で動いていないことが分かっていませんでした。
それで、実際にやりたかったのは以下のPRのhardwareのサーボアラームでログをとることになります。
https://github.com/fkanehiro/hrpsys-base/pull/556

closed 2015-07-23 by YoheiKakiuchi