fkanehiro/hrpsys-base #493: ErrorLimit警告時に力センサ情報出力周期が低くなる

mmurooka, 2015-02-18:
@snozawaさん， 本日の実験であったErrorLimitの警告printが多すぎて，/lfsensorなどのrosトピック出力周期が遅くなる問題に対処いただけますでしょうか．

mmurooka, 2015-02-18:
https://github.com/fkanehiro/hrpsys-base/pull/491 と関係あったかもしれません，見落としていました．

snozawa, 2015-02-18:
出力がちょっと多いRTCはあるのですが，トピックが遅い状況がいまいち再現できてません．
外からアクセスする何かテストプログラムがいるかもしれません．

mmurooka, 2015-02-18:
首ヨー軸にeusから50度くらいを指令して，ErrorLimitの警告音がなっている状態で
`rostopic echo /lfsensor` で1Hzくらいまで周期が落ちていたと思います．

snozawa, 2015-02-18:
そうですね．その条件で試してみて，うまく再現しなかったです．

mmurooka, 2015-02-19:
こちらでも試して再現させられませんでした．
一応，テレオペプログラムを昨日同様に起動して，abc,stを入れてポイントクラウドをRvizで描画して，同じ環境にしたつもりですが，問題は生じませんでした．
長く起動し続けていることなどが関係あるのかもしれません．

また問題の起きる条件など分かりましたら報告させていただきます．

mmurooka, 2015-02-19:
以下のようなprint文が大量にでているところで，/lfsensorが遅くなる現象再現しました．

```
spdlmt: 2.81967e-05 -1.72812e-05 -0.0784243 -0.0433786 -0.000908676 -1.80788e-05
Lower joint limit error LLEG_JOINT2
Lower joint limit error LLEG_JOINT3
spdlmt: -1.42132e-05 -2.63076e-05 -0.0702643 -0.0389522 -0.000636274 -2.67555e-05
Lower joint limit error RLEG_JOINT2
Lower joint limit error RLEG_JOINT3
spdlmt: 1.61631e-05 -5.84759e-06 -0.0784241 -0.0433784 -0.000704369 -2.74965e-06
Lower joint limit error LLEG_JOINT2
Lower joint limit error LLEG_JOINT3
spdlmt: -1.04341e-05 -2.05398e-05 -0.070264 -0.038952 -0.000476818 -2.18237e-05
Lower joint limit error RLEG_JOINT2
Lower joint limit error RLEG_JOINT3
spdlmt: 8.22883e-06 1.28202e-06 -0.0784239 -0.0433783 -0.000546036 6.5101e-06
Lower joint limit error LLEG_JOINT2
Lower joint limit error LLEG_JOINT3
spdlmt: -1.94655e-05 -3.34722e-05 -0.0703431 -0.038996 -0.000847354 -3.22911e-05
Lower joint limit error RLEG_JOINT2
Lower joint limit error RLEG_JOINT3
spdlmt: 2.82124e-05 -1.72863e-05 -0.0785073 -0.0434247 -0.000909621 -1.8171e-05
Lower joint limit error LLEG_JOINT2
Lower joint limit error LLEG_JOINT3
spdlmt: -1.42532e-05 -2.6249e-05 -0.0703427 -0.0389958 -0.00063523 -2.66403e-05
Lower joint limit error RLEG_JOINT2
```

今回は以下のようにして起きて，繰り返し起きることを確認しました．
ただ，昨日はサーボONで実験中に起きたので，原因は他にもあるかもしれません．
1. 一度サーボon, start-abc, start-stをしてから，サーボOFFする．
2. リフトを下ろしてロボットの足先を地面に付ける．
3. 上のprintがでて数秒後からlfsensorが1Hz程度に落ちる．

適当に推察すると，stが切れていなくて，発散して脚関節のエラーが大きくなりprintが出ているというような感じでしょうか．