fkanehiro/hrpsys-base #529: servoOff で sc_svc がないと言われる

k-okada, 2015-03-23:
hrp2/jaxon/staro をつかっている人で，hrpsys_config.pyの servoOn/Off を使っている人いますか？以下のようなエラーがでませんでしょうか？

```
In [2]: robot.servoOff()
Turn off Hand Servo
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-47d35baf78ab> in <module>()
----> 1 robot.servoOff()

/home/k-okada/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.pyc in servoOff(self, jname, wait)
   1607 
   1608         print 'Turn off Hand Servo'
-> 1609         if self.sc_svc:
   1610             self.sc_svc.servoOff()
   1611         # if the servos aren't on switch power off
```

snozawa, 2015-03-24:
https://github.com/fkanehiro/hrpsys-base/issues/360
https://github.com/fkanehiro/hrpsys-base/issues/485
などもともとのコードで不明点・動かない点があり，
直すのに時間がまわせてない状況なので使っていないです．

また，
https://github.com/fkanehiro/hrpsys-base/issues/131
goActual相当の挙動をうわがきRTCがするようなdiffもあります．
（servoOnのコードはjaxon_hrpsys.pyなどにあります）

k-okada, 2015-03-24:
なるほど．
とりあえず本題目のものは以下で治りますね．
https://github.com/fkanehiro/hrpsys-base/pull/531

k-okada, 2015-03-31:
#531 で解決しました

closed 2015-03-31 by k-okada