fkanehiro/hrpsys-base #575: 実機が自己干渉している状態だとサーボONできない

mmurooka, 2015-04-14:
もしかして https://github.com/fkanehiro/hrpsys-base/pull/525 が怪しいでしょうか．

k-okada, 2015-04-14:
具体的にどうなりますか？
- エラーが帰ってくるなどなどpythonAPIレベルでの話
- HRPSYSのログの出力の話
などありそうですが．

k-okada, 2015-04-14:
https://github.com/fkanehiro/hrpsys-base/commit/d80205fb3aa6481628b172d4e444e777696fe16e#diff-a1140e67b345481d32aac3ffd6cb0ed9R541

snozawa, 2015-04-14:
rtcdのログに、サーボ直後で
```
servo limit over 
```
がでているか、でていれば関節名がなにになってるか、がわかれば教えてください

eisoku9618, 2015-04-14:
保存されているrtcdのlogをgrepで検索してみましたが，

> servo limit over 

は出ていませんでした．

```bash
[http://localhost:11311][192.168.97.48] leus@transvm:~$ cd ~/tmp/2015-04-12/
[http://localhost:11311][192.168.97.48] leus@transvm:~/tmp/2015-04-12$ grep "servo limit over" * -r
[http://localhost:11311][192.168.97.48] leus@transvm:~/tmp/2015-04-12$ 
```

※4/12はサガミハラの日で，この日は @YoheiKakiuchi さんも見ていて，確かにサーボONが出来なかった

k-okada, 2015-04-15:
再現する環境ができらた呼び出して下さい。

YoheiKakiuchi, 2015-04-15:
@eisoku9618 そのレベルのログも必要ですが、出来ればdataloggerでとったログを出してみてください。

snozawa, 2015-04-15:
@eisoku9618 さん

rtcdのログで，時系列的にサーボON前後の部分を教えてください．
おそらく`goActual`とでてる箇所の前後をみるとよさそうです．

eisoku9618, 2015-04-15:
申し訳ないですが，まだ実機で試せていないので，サガミハラのときのログになりますが，

https://gist.githubusercontent.com/eisoku9618/36754413e3700d7adb44/raw/2015-04-12_110349_hrpsys.log

に置きました．

（長すぎて行を指定したリンクを作れませんでしたので，goActualで検索してもらう必要がありますが，）
goActualの前後は

write_power_command -> joint Groupの remove /add -> ic / st / abc / co のdeactivate  -> goActual -> write_servo -> P gainを徐々に上げる

となっています．

snozawa, 2015-04-15:
ちなみに本件の症状は、サーボが入らない（servoOn操作をしても、status monitorなどでoff状態になってる）で良いでしょうか。
サーボはstatus 的にははいってるけど、初期姿勢などに遷移できない、というものではないというのであってますか？

k-okada, 2015-04-15:
そんなむつかしいはなしではなくて，
https://github.com/fkanehiro/hrpsys-base/commit/d80205fb3aa6481628b172d4e444e777696fe16e
がまるまる要らなくなった，というだけきがするんだけど違うかな．

YoheiKakiuchi, 2015-04-15:
症状としては、サーボは入って（全軸緑になって）、そのあと特定の軸が赤くなる（refとactualの差が大きい)です。
どこでどうなっているかdataloggerのログを見ないと（ひつようならsimple_shmのログも)正確にはわからないですが、
hrpsysでactualと同じrefを出力できていないのだと思います。

手でささえて、```send *ri* :angle-vector (send *ri* :state :potentio-vector)``` をすると、エラーにならずにサーボOnできます。
potentio vectorをsendしなくてもサーボOnできるようなきがしますが、ちゃんと試せていない気がします。

snozawa, 2015-04-15:
disableという関数がある以上、
> https://github.com/fkanehiro/hrpsys-base/commit/d80205fb3aa6481628b172d4e444e777696fe16e

は必要です。そもそもdisableいらなくない？となったらいらないと思います。
（ちなみにdisableはどういう用途で追加されてるものでしょうか。）

症状と状況を共有している段階だとおもったので
https://github.com/fkanehiro/hrpsys-base/issues/575#issuecomment-93287569
だったのですが、
jaxonのhrpsys.pyをみるとそういえばrobothardwareのset error limitを0に設定してdisableしています。
なので、エラーがでかいときの挙動としては
- https://github.com/fkanehiro/hrpsys-base/issues/575#issuecomment-92914454にあるように`servo limit over`はでない
- その下のレイヤでエラーにかかってサーボOFFになる（status monitorが赤くなる）（https://github.com/fkanehiro/hrpsys-base/issues/575#issuecomment-93338620にはoffになるとかいてないですが、OFFでよいでしょうか？）

なんだと思いました。

k-okada, 2015-04-15:
ああ，なるほど．
https://github.com/fkanehiro/hrpsys-base/blob/1af9f05f087c6b0976f0fd415da80f469dca1bcb/rtc/CollisionDetector/CollisionDetector.cpp#L440

if ( ! m_have_safe_posture && m_safe_posture ) {

みたいな条件でもm_q = m_qRefしないとだめだったりするかな．こっちで試せるので，とりあえず皆さんはペンディングでよいです．

hrpssyのbodystatみたいなのがあるといいですね． @fkanehiro

k-okada, 2015-04-15:
> d80205f
> #575 (comment)

enable/disable
も同じように，最初に電源をいれたときにcollisionになるから最初はdisableではじめて，途中でenableにしたい，という要望だったとおもいます．
で，いまはさいしょからcollisionをいれていいはずなので，enable/disableは必要なくて，それがあっても，servo
onを見る必要もないとおもうけどな．どうだろうか．

fkanehiro, 2015-04-15:
>
> hrpssyのbodystatみたいなのがあるといいですね． @fkanehiro
>

hrpsys-monitorが近いかもしれませんね。

eisoku9618, 2015-04-17:
もう遅いのかもしれませんが，

> 再現する環境ができらた呼び出して下さい。

電源が入るようになったので，いつでも再現できるようになりました．

> @eisoku9618 そのレベルのログも必要ですが、出来ればdataloggerでとったログを出してみてください。

JSK内部の人しか見えないきがしますが， https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/folderview?id=0B5qanGnXorOufmx6Z2lQdmVqblplOGpvQlp5REl6NkhNQzJKaWZ4dmFFMGVzR0Zvby1oVUE&usp=sharing に上げました．

> rtcdのログで，時系列的にサーボON前後の部分を教えてください．

https://drive.google.com/open?id=0B5qanGnXorOud2FHSWpRY2IyU0E&authuser=0 に上げました．

ご確認よろしくお願いします．

mmurooka, 2015-04-20:
今日もこのIssueの問題おきまして，
rtcdの出力で，コリジョンしている姿勢でも
サーボONできることできないことがありました．
なので，コリジョンしていたら絶対アウトというわけではなさそうです．

3回連続サーボON失敗して諦めたときのコリジョンしていたペア
```
12/154 pair: CHEST_JOINT2/LARM_JOINT2(3), distance = -0.000242433
126/154 pair: RLEG_JOINT2/RARM_JOINT6(13), distance = -1.83878e-05
```

1回でサーボONできたときのコリジョンしていたペア
```
1/154 pair: CHEST_JOINT1/LARM_JOINT3(5), distance = -0.00336285
12/154 pair: CHEST_JOINT2/LARM_JOINT2(3), distance = -0.0190919
18/154 pair: CHEST_JOINT2/RARM_JOINT2(3), distance = -0.0001201
```

```
CAUTION!! The robot is moving while collision detection!!!, since we do not get safe_posture yet
```
も出ていました．
rtcdのログは取り忘れてしまいました．