fkanehiro/hrpsys-base #577: Error occurs when rtmlaunch samplerobot-drc-testbed.launch

chiwunau, 2015-04-15:
It seems the problem is due to SampleRobot.DRCTestbed.xml under src/hrpsys/sample/SampleRobot was not copied to devel/share/hrpsys/samples/SampleRobot/

I have also found that SampleRobot.DRCTestbed.xml is not included in the following file. however, catkin build did not pass after I had tried adding
  ${CMAKE_CURRENT_BINARY_DIR}/SampleRobot.DRCTestbed.xml
to the file.
[CMakeLists.txt#L59](https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/CMakeLists.txt#L59)

how should I fix it?

output log 
```
au@au-ThinkPad-T440p:~$ rtmlaunch hrpsys samplerobot-drc-testbed.launch 
[rtmlaunch] Start omniNames at port 15005 

Wed Apr 15 23:07:46 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000e0000003139322e3136382e39372e3839009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000b2702e550100149f
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/au/.ros/log/cb372144-e378-11e4-8566-5c514f55c963/roslaunch-au-ThinkPad-T440p-5257.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://au-ThinkPad-T440p:46535/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion

NODES
  /
    hrpsys (hrpsys/hrpsys-simulator)
    modelloader (openhrp3/openhrp-model-loader)
    start_omninames (hrpsys/start_omninames.sh)

auto-starting new master
process[master]: started with pid [5565]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cb372144-e378-11e4-8566-5c514f55c963
process[rosout-1]: started with pid [5679]
started core service [/rosout]
process[start_omninames-2]: started with pid [5713]
omniORB: Failed to bind to address :: port 15005. Address in use?
omniORB: Error: Unable to create an endpoint of this description: giop:tcp::15005
Failed to initialise the ORB / POA: INITIALIZE_TransportError
Is omniNames already running?
[start_omninames-2] process has finished cleanly
log file: /home/au/.ros/log/cb372144-e378-11e4-8566-5c514f55c963/start_omninames-2*.log
process[modelloader-3]: started with pid [5730]
ready
process[hrpsys-4]: started with pid [5766]
I/O warning : failed to load external entity "/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.DRCTestbed.xml"
unable to parse file(/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.DRCTestbed.xml)
failed to parse /home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.DRCTestbed.xml
[hrpsys-4] process has died [pid 5766, exit code 1, cmd /home/au/ros/hydro_parent/devel/lib/hrpsys/hrpsys-simulator /home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.DRCTestbed.xml -o manager.is_master:YES -o naming.formats:%n.rtc -o exec_cxt.periodic.rate:1000000 -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/au/ros/hydro_parent/devel/share/hrpsys/lib -o manager.modules.preload:HGcontroller.so -o manager.components.precreate:HGcontroller -o exec_cxt.periodic.type:SynchExtTriggerEC -f /home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.RobotHardware.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.RobotHardware.200conf -o example.SequencePlayer.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.ForwardKinematics.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.ImpedanceController.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.AutoBalancer.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.StateHolder.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.DataLogger.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.log.conf -o example.TorqueFilter.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.TorqueController.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.ThermoEstimator.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.ThermoLimiter.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.VirtualForceSensor.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.AbsoluteForceSensor.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.RemoveForceSensorLinkOffset.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.KalmanFilter.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.Stabilizer.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.CollisionDetector.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.SoftErrorLimiter.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o example.HGcontroller.config_file:/home/au/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/SampleRobot.200.conf -o corba.nameservers:localhost:15005 -endless -realtime __name:=hrpsys __log:=/home/au/.ros/log/cb372144-e378-11e4-8566-5c514f55c963/hrpsys-4.log].
log file: /home/au/.ros/log/cb372144-e378-11e4-8566-5c514f55c963/hrpsys-4*.log
^C[modelloader-3] killing on exit
[rosout-1] killing on exit
^C[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rtmlaunch] terminate omniNames at port 15005 
```

k-okada, 2015-04-15:
On Wed, Apr 15, 2015 at 11:33 PM, Chi Wun Au <notifications@github.com>
wrote:

> how should I fix it?
>
>
add install line to cmake file

--
◉ Kei Okada

eisoku9618, 2015-04-15:
You can simply use ``insatll`` command like this https://github.com/fkanehiro/hrpsys-base/pull/579 because SampleRobot.DRCTestbed.xml is not a template file.

reference : http://qiita.com/osamu0329/items/edc66e2e1b6c96947771

chiwunau, 2015-04-16:
Thanks,
should we also install python scripts like samplerobot_auto_balancer.py?

k-okada, 2015-04-16:
which python script is not installed

chiwunau, 2015-04-16:
I think all the *.py [here](https://github.com/chiwunau/hrpsys-base/tree/master/sample/SampleRobot) are not installed.

k-okada, 2015-04-16:
i see, PR is welcome, use FILE(GROB to get all python code,
http://www.cmake.org/cmake/help/v3.0/command/file.html

k-okada, 2015-04-16:
PR -> https://github.com/fkanehiro/hrpsys-base/pull/581

closed 2015-04-22 by chiwunau