fkanehiro/hrpsys-base #655: ImpedanceControllerでstart-impedanceしていないlimbをstop-impedanceするとハングアップすることがある

orikuma, 2015-05-20:
ImpedanceControllerでstartImpedanceControllerしていないlimbをstopImpedanceControllerするとImpedanceControllerからstopImpedanceControllerのresposnseが返らずそのままハングすることがあります.
ImpedanceControllerが止まるだけで他のrtcは動いている(例えば関節角度を送ることは出来る)ようです.
再現性は100%ではありませんが高め(50%以上は起こる気がしている)です.

snozawa, 2015-05-20:
50%はかなりデバッグしやすい頻度ですね。
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_impedance_controller.py
的なかんじのバグを再現するコードを教えてください。

k-okada, 2015-05-23:
このときのstopImpedanceはpythonからでしょうか？eusからでしょうか？eusからだと，
https://github.com/start-jsk/rtmros_common/pull/732
で治ったかもしれないです．

元々はImepdanceControllerROSBridgeCompで，onInitializeでadvertiseServiceしていますが，
この時はRTCとしてはactivateしていないので，onExecuteがはしらず，spinOneceしないので，
serviceが呼ばれても処理されないとなっています．

https://github.com/start-jsk/rtmros_common/pull/732 のパッチでadvertiseServiceをonAcivateに
いれました．

もし，これが正しいとすると，onActivateするまえにstop-impedanceしていることになります．
なので，ちゃんとRosBridgeが上がり切ることを確認して，チョットsleepしてから
stop-impeadnceすればいいはずですので，６月上旬まではどちらの方針でもいいとおもいます．
無理にidl2srvをマージしなくてもよいかとおもいます．

k-okada, 2015-08-21:
start-jsk/rtmros_common#732 
で治っていないという噂がある．
ログ等あったら貼り付けて下さい． @garaemon @orikuma

garaemon, 2015-08-21:
JAXONでは地面に下ろした時に"Robot stands on the ground"と喋らないときは止まっている(stを常に叩いているので)