fkanehiro/hrpsys-base #684: hrpsys-simulatorで点群情報を取得する方法、サンプル

snozawa, 2015-06-14:
hrpsys-simulatorで点群情報を取得する方法、サンプルがございましたらご教示いただけますと幸いです。
よろしくお願いいたします。

fkanehiro, 2015-06-14:
点群を出すセンサとして、レンジセンサとデプスカメラがありますが、どちらもBodyRTCの出力ポートをプロジェクトファイルで設定することによってデータを取り出すことができます。
データポートの定義の仕方は他のデータタイプと同じく、OpenHRP3のコントローラブリッジの設定ファイルと同様です。
http://www.openrtp.jp/openhrp3/jp/controller_bridge.html

snozawa, 2015-06-14:
ありがとうございます。
ポートをかいてみましたが、
http://www.openrtp.jp/openhrp3/jp/controller_bridge.html
の`DEPTH_IMAGE`等はhrpsys-simulatorで対応しておらず、
`VISION_SENSOR`,`POINT_CLOUD`であればプロジェクトファイルで設定できるようでした。
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/util/BodyRTC.cpp#L356
また、データ型もopenhrp3のものでなくhrpsys-baes/idlのもののようでした。
これらは、upstreamのopenhrp3にも取り入れられますでしょうか。