fkanehiro/hrpsys-base #713: DataLoggerでの浮動少数点の書き出し

snozawa, 2015-07-07:
先日DatLoggerで保存した関節角度データを微分して微分してみたら、
ものすごくギザギザしてました。
よくよく考えるとテキストにAsciiでかいてるので、けた落ちがあったようです。

またファイルサイズがでかくなるときも時々あるので、
バイナリで読み書きもできる（か桁数をものすごくふやす）とよいかもと思いました。

ちなみに、OpenhRPかhrpsys-baseで、すでにどこかにバイナリでファイルをかいたりよんだりするものがありますでしょうか。

fkanehiro, 2015-07-07:
ロガーの出力をそのままgnuplotで見たりするのでアスキーにしています。
バイナリファイルの入出力をしているところはちょっと思い当たりません。

k-okada, 2015-07-07:
僕の資料によると小数点以下16桁ぐらい書くとdoubleの精度になるみたいです

今は6桁ぐらいでしょうか?

最近のeusのfloat-vectorはdouble だったきがしていますがreadしても大丈夫だったかどうか。。。

snozawa, 2015-07-07:
> ロガーの出力をそのままgnuplotで見たりするのでアスキーにしています。
> バイナリファイルの入出力をしているところはちょっと思い当たりません。

ありがとうございます、わかりました。
ちなみに、（挙動をかえないように）デフォルトをテキストにして、IDLで設定したらバイナリにもするのはいかがでしょうか。

また、gnuplotで見るプログラムはどちらかにありますでしょうか？
（もしくは、凡例や縦軸横軸の書き方的にそちらの制御系に則したものになっていて、hrpsys-base共通としては利用できなかったりしますでしょうか）

また、DataLogger.cppに
`        os << std::scientific << std::setprecision(16);`
をいれてみてテキストファイル＋桁をふやすようにしてみて、それっぽく書き出されてました。
ただ、`samplerobot_data_logger.py`のサンプルを試すと、
- もともとのログの容量：888kくらい
- 桁をふやしたもの：2.0Mくらい

になってしまいました。

> 今は6桁ぐらいでしょうか?

はい、固定小数点にデフォでなっているようです。

> 最近のeusのfloat-vectorはdouble だったきがしていますがreadしても大丈夫だったかどうか。。。

確認はこんなかんじであってるんでしたでしょうか。
```
;; 小数点16ケタくらいの数字で、16ケタ目をかえる
4.irteusgl$ (format nil "~30,20f" (- (read-from-string "-7.3343921916616499e-05") (read-from-string "-7.3343921916616490e-05")))
"        0.00000000000000000000" ;; 差がでない
;; 小数点16ケタくらいの数字で、15ケタ目をかえる
5.irteusgl$ (format nil "~30,20f" (- (read-from-string "-7.3343921916616499e-05") (read-from-string "-7.3343921916616400e-05")))
"       -0.00000000000000000011" ;; 差が出る
;; => Euslispはdoubleで15桁？
```

YoheiKakiuchi, 2015-07-07:
そうですね。Eusはそんな感じで、仮数部の2ビットは”tag ビットとして使用されている”ので、通常のdoubleの精度よりも低いです。
~~~
(dotimes (i 16) (print (list (1+ i) (= (sqrt 2) (read-from-string (format nil (format nil "~~~D,~DE" (1+ i) (1+ i )) (sqrt 2)))))))
(1 nil)
(2 nil)
(3 nil)
(4 nil)
(5 nil)
(6 nil)
(7 nil)
(8 nil)
(9 nil)
(10 nil)
(11 nil)
(12 nil)
(13 nil)
(14 nil)
(15 t)
(16 t)
~~~

fkanehiro, 2015-07-08:
>
> ちなみに、（挙動をかえないように）デフォルトをテキストにして、IDLで設定したらバイナリにもするのはいかがでしょうか。
>
はい、よろしいんじゃないでしょうか。

> また、gnuplotで見るプログラムはどちらかにありますでしょうか？
> （もしくは、凡例や縦軸横軸の書き方的にそちらの制御系に則したものになっていて、hrpsys-base共通としては利用できなかったりしますでしょうか）
>

いえ、特に整備してないです。
毎度手打ちでコマンドを入れています。

garaemon, 2015-07-16:
桁落ちが嫌でasciiしかゆるされていないなら、unsigned longとかにキャストして書き出すというのが漏れがないですね。