fkanehiro/hrpsys-base #733: TorqueControllerに関係しそうなprint文でどのRTCから出ているか表示に追加する

k-okada, 2015-07-23:
from:https://github.com/start-jsk/rtmros_tutorials/issues/356#issuecomment-123792342
```
@orikuma さん
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceController.cpp#L606
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceOutputGenerator.h#L93
みたいに、TorqueControllerなど関係しそうなprint文で、どのRTCからでてるか追加しておいてもらえるでしょうか。
```

assigned @orikuma

snozawa, 2015-07-23:
@orikumaさん
補足で説明すると，
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceController.cpp#L606
のようなRTCでは
```
std::cerr "[" << m_profile.instance_name << "] << ...
```
のようにして，各RTCの名前が追加できます．

また，RTCから更に呼ばれている各プログラム（今回はTwoDofController.[cpp,h]）でも
（やり方はいろいろあると思いますが）
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceOutputGenerator.h#L47
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceOutputGenerator.h#L93
のようにして引数でprint用stringをRTC側から渡してあげれば，各RTCの名前が出力にでてきます．

おそらくsedを２回と，引数をちょっと変える作業の２手間くらいの変更ですむと思うので，それでお願いします．
このあたりが最速で治ってくると，@eisoku9618さんもとても助かると思います．

snozawa, 2015-08-07:
こちら解決しましたのでcloseでお願いします。