fkanehiro/hrpsys-base #751: count_if.hppがboostバージョン1.34にない

mmurooka, 2015-08-06:
@eisoku9618 さん

HRP2実機の体内PCのboostのバージョンが1.34.1ですが
このバージョンだとcount_if.hpp がありませんでした．

https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/AutoBalancer/GaitGenerator.h#L11
でincludeしていますが，使わないように修正いただけますでしょうか．

k-okada, 2015-08-06:
+1 , PR　見たらわかりますが，qnxのコンパイルで落ちているのはコードがあたらしすぎるからだともいます＞ @eisoku9618

eisoku9618, 2015-08-06:
すみません．
https://github.com/fkanehiro/hrpsys-base/pull/752 で対応してみました．

今回は``#include <boost/range/algorithm/count_if.hpp>``と書いてあって，それがダメだったのですが，予め確認する方法としては，実機で確認する以外としては，

```bash
wget http://sourceforge.net/projects/boost/files/boost/1.34.1/boost_1_34_1.tar.bz2/download -O boost_1_34_1.tar.bz2
tar jxvf boost_1_34_1.tar.bz2
cd boost_1_34_1
ls range/algorithm/count_if.hpp
```

で存在を確認するで大丈夫そうでしょうか？

eisoku9618, 2015-08-06:
> qnxのコンパイルで落ちているのはコードがあたらしすぎるからだともいます

PRを見なおしてみたら，僕のPR以前はqnxのコンパイルが通っていて，僕のPRからqnxのコンパイルが通っていない気がしてきました．

k-okada, 2015-08-06:
https://github.com/fkanehiro/hrpsys-base/pull/752#issuecomment-128335159
からたどっていける
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/hrpsys-qnx/2230/console
で，同じようにふるい環境のQNXの結果がわかります．

どこかの時点で最近の黒岩くんの変更でエラーになるようになっています．
戻してもらえると嬉しい．

jenkinsをコンパイラとして使うのがいいと思います．

eisoku9618, 2015-08-06:
> どこかの時点で最近の黒岩くんの変更でエラーになるようになっています．

https://github.com/fkanehiro/hrpsys-base/pull/730 だと思います．

> jenkinsをコンパイラとして使うのがいいと思います．

これはPRを出すとjenkinsが走るので確認できる，という意味でしょうか？

k-okada, 2015-08-06:
2015-08-06 20:57 GMT+09:00 Eisoku Kuroiwa <notifications@github.com>:

> これはPRを出すとjenkinsが走るので確認できる，という意味でしょうか？
>

そうです．で，commitしなおすと．また走るので，てもとでソースをいじる，commit，結果を jenkinsで知る．がいいともいます．

--
◉ Kei Okada

k-okada, 2015-08-06:
で，全部動いたら最後にsquash，

--
◉ Kei Okada

2015-08-06 20:58 GMT+09:00 Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>:

>
> 2015-08-06 20:57 GMT+09:00 Eisoku Kuroiwa <notifications@github.com>:
>
>> これはPRを出すとjenkinsが走るので確認できる，という意味でしょうか？
>>
>
> そうです．で，commitしなおすと．また走るので，てもとでソースをいじる，commit，結果を jenkinsで知る．がいいともいます．
>
> --
> ◉ Kei Okada
>
>

eisoku9618, 2015-08-06:
分かりました．やってみます．

eisoku9618, 2015-08-06:
QNXも通りました．

ちなみに，QNXでコンパイルできたらhrp20xxcでもコンパイルできる，と思って差し支えないのでしょうか？それとも一概には言えない感じでしょうか？

k-okada, 2015-08-06:
> ちなみに，QNXでコンパイルできたらhrp20xxcでもコンパイルできる，と思って差し支えないのでしょうか？それとも一概には言えない感じでしょうか？

多分OK
--
◉ Kei Okada

k-okada, 2015-08-06:
boost マスターだね
--
◉ Kei Okada

2015-08-06 22:18 GMT+09:00 Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>:

snozawa, 2015-08-06:
:100:

eisoku9618, 2015-08-27:
16号機体内でビルドしてみたところ、

https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/AutoBalancer/testGaitGenerator.cpp#L72 の

```cpp
gg->get_support_leg_names() == boost::assign::list_of("rleg")
```

の比較のところ（この関数は``const std::vector<std::string> get_support_leg_names()``）でエラーがでまして、

```cpp
std::vector<std::string> hoge =　boost::assign::list_of("rleg");
```

と一旦おいてから、

```cpp
gg->get_footstep_front_leg_names() == hoge
```

とすることで解決したので、PR作ります。

jenkinsの方で通った -> HRP2体内で通る

というわけではなさそうです。

eisoku9618, 2015-08-27:
ちなみに以下のプログラムで確認したところ，16号機のcのboostのバージョンは1_34_1でした．

|  | boost | g++ |
| --- | --- | --- |
| hrp2016c | 1_34_1 | 4.2.4 |
| 手元のubuntu12.04の普通のPC | 1_46_1 | 4.6.3 |
| JAXON体内12.04 law latency | 1_46_1 | 4.6.3 |

```cpp
#include <iostream>
#include <boost/version.hpp>
int main()
{
  std::cout << BOOST_LIB_VERSION << "\n";
  return 0;
}
```

k-okada, 2015-08-27:
メモ：QNXは 1_42_0

mmurooka, 2015-09-18:
遅くなりましたが解決しました．ありがとうございました．

closed 2015-09-18 by mmurooka