fkanehiro/hrpsys-base #757: Release 315.6.1

130s, 2015-08-14:
そろそろいかがでしょうか．

yosuke, 2015-08-19:
@k-okada さんからこちらにも書いておくよう言われて、確かにgithubのほうがやりやすいな、とも思うので、別フォーラムの内容をコピペしておきます。

元の投稿はadd-apt-repository ppa:hrg/dailyで登録できるppaベースのレポジトリにhrpsys-baseパッケージがあり、そのバージョン管理はどのようにしているのか、という質問でその説明は以下になります。

--------------------------------------------------------------

ppaで提供されているhrpsys-baseのバージョン文字列は、「3.1.5+20150730+1155+9~ubuntu14.04.1」という感じになっていると思うのですが、それぞれ

3.1.5 : debian changelogのバージョン文字列(debian changelogの更新遅れで合っていない場合もあり)

20150730 : ビルドした日付

1150 : launchpadにインポートされたgithubレポジトリのコミット連番 http://bazaar.launchpad.net/~hrg/hrg-packaging/hrpsys-base/changes

9 : パッケージ化スクリプトのコミット連番

になっています。基本的に前日にコミットがあった内容で毎晩パッケージ化されるのですが、細かく見たい場合はコミット連番を見ていただくのが良いと思います。

--------------------------------------------------------------

ROSのバージョンは、315.日付になっていると思うのですが、それと同じような考え方です。

k-okada, 2015-08-19:
うーん．ここにも書かないといけないか．

```
なるほど．一応315.a.b でidlの変更があるとaが上がる，という風にしてあります．
https://github.com/fkanehiro/hrpsys-base/issues/113
また，プラグインから    hrpsys_version = hcf.seq.ref.get_component_profile().version
 でバージョン番号が取られてくるので，それが使われるといい気がします．

おっしゃるように /usr , /opt/ros で喧嘩はしないですが，よろしくはないですよね．
解決策は ubuntuのデフォルトのパッケージにしてしまうことだと思っているんですが
いかがでしょうか．
```
https://cybozulive.com/2_222079/gwBoard/view?bid=2%3A3226808&focusOn=follow#dummy=1439976166207

fkanehiro, 2015-08-19:
ちなみにROSのパッケージの更新頻度はどうなっているんでしょうか。
バージョン番号が変わった時のみ？
PPAの方は毎日更新なので、それはそれで存在意義があると思うのですが、共存できる方法はないでしょうか？

k-okada, 2015-08-19:
ROSのパッケージはバージョンが変わった時だけです．毎日バージョンを上げればいいとは思います．
今もあげたらいいんですが，一応上げる度にCHANGELOG.rstを作っていて，以下までは自動で作るんですが，せっかくなので毎回それなりに整理しているんですが，こう長くなると整理も面倒になってきますね．．．．

```
Forthcoming
-----------
* [GaitGenerator.cpp] fix bug: keep align the order of names and coordinates of foostep_nodes_list.front()
* [AutoBalancer.cpp, GaitGenerator.*] add multi_mid_coords function to calculate a midcoords of multi coordinates in fixLegToCoords, get_swing_support_mid_coords and stopWalking
* [AutoBalancer.cpp] use leg_names instead of "rleg" or "lleg"
* [GaitGenerator.*] use leg_type_map in order to convert between leg_type and name
* [AutoBalancer.cpp, GaitGenerator.h] move leg_type_map to gait_generator
* [GaitGenerator.*] rename get_support_leg_types_from_footstep_nodes to calc_counter_leg_types_from_footstep_nodes
* [GaitGenerator.*] move get_support_leg_types_from_footstep_nodes to gait_generator
* [doc] Remove old info from downstream pkg
* [AutoBalancer.cpp, GaitGenerator.h] fix return type of get_dst_foot_midcoords from std::vector<coordinates> to coordinates because we need the reference coordinates
* [GaitGenetarot.*] keep swing_legs at update_leg_steps
* [AutoBalancer.cpp] consider some variable life times
* [AutoBalancer.cpp, GaitGenerator.*, testGaitGenerator.cpp] use step_node instead of coords because we need to align the oder of names of legs and coords of legs
* [AutoBalancer.cpp, GaitGenerator.*, testGaitGenerator.cpp] fix variable names of legs and corresponding method names
* [GaitGenerator.h] add hints for the second template argument of boost::assign::list_of at a constructor initialization phase
* [GaitGenerator.*] use std::count_if instead of boost::count_if since HRP2 inside PC does not support boost::count_if
* [rtc/Stabilizer/CMakeLists.txt] Add subdirectory for qpoases linking
* [rtc/Stabilizer/CMakeLists.txt] Use qpOASES installed by 3rdparty directory.
* [CMakeLists.txt] Build 3rdparty directory
* [3rdparty] Add 3rdparty directory based on https://github.com/fkanehiro/hrpsys-base/pull/683 discussion (currently for qpOASES)
* removes an unused file
* adds a new component, VoxelGridFilter
* [GaitGenerator.*] use vector class functions to get errors : http://qiita.com/ota42y/items/f2067f6b81dd15bca95a
* [AutoBalancer.cpp] improve startWalking for multiple legs
* [AutoBalancer.cpp, GaitGenerator.*, testGaitGenerator.cpp] improve go_pos_param_2_footstep_nodes_list for multiple legs
* [AutoBalancer.cpp, GaitGenerator.*, testGaitGenerator.cpp] remove an unused argument of go_pos_param_2_footstep_nodes_list
* [AutoBalancer.cpp, GaitGenerator.h] fix indent
* [AutoBalancer.cpp, GaitGenerator.*, testGaitGenerator.cpp] add an argument of go_pos_param_2_footstep_nodes_list to set start_ref_coords no matter which gait we choose
* [GaitGenerator.*] improve get_swing_legs of leg_coords_generator for multiple legs
* [python/hrpsys_config.py] Enable thermolimiter and thermoestimator
* add outport for capture point
* [GaitGenerator.cpp/.h] extend append_go_pos_step_nodes to get an argument of multiple legs
* [AutoBalancer.cpp, GaitGenerator.cpp/h, testGaitGenerator.cpp] add a variable named all_limbs which stands for candidates of contact legs
* [GaitGenerator.cpp, .h] fix a function name to follow the naming rule
* [GaitGenerator.cpp] use count_if for multiple legs
* [GaitGenerator.cpp] use min max functions for the stride limits
* fix name : foot_midcoords -> ref_coords
* add mutex lock when writing is_emergency_mode variable
* add test-samplerobot-emergency.py and test EmergencyStopper in test-samplerobot.test
* add test code of EmergencyStopper function
* go to release_mode when deactivated in EmergencyStopper
* fix format specifier of size_t
* support multiple zmp offsets input to PreviewController
* [TorqueController] Fix typo, emergencyController -> normalController, in disable error message
* [TorqueController] Add enable/disable methods to MotorTorqueController
* [rtc/GraspController/GraspController.cpp] Add debug message to grasp controller start/stop grasp and add instance name for print message
* [rtc/SoftErrorLimiter/SoftErrorLimiter.cpp] Fix bug of Velocity limitation in https://github.com/fkanehiro/hrpsys-base/pull/726
* [TorqueController] Add enable flag to MotorTorqueController to manage activity of both normal and emergency controller
* [sample/SampleRobot/samplerobot_soft_error_limiter.py] Check hrpsys version
* [test/test-samplerobot-el.test] Increase rostest execution time
* [sample/SampleRobot/samplerobot_auto_balancer.py] Add attitude check to auto balancer test
* [sample/SampleRobot/samplerobot_stabilizer.py, SampleRobot.torque.xml.in] Add check for attitude to st test
* [rtc/SoftErrorLimiter/JointLimitTable., rtc/ImpedanceController/JointPathEx*, rtc/SoftErrorLimiter/CMakeLists.txt, rtc/SoftErrorLimiter/SoftErrorLimiter.h] Move limit table codes to separated file.
* [sample/SampleRobot/samplerobot_soft_error_limiter.py] Add position limit testing
* [test/test-samplerobot-el.*, sample/SampleRobot/samplerobot_soft_error_limiter.py] Add rostest for soft error limiter
* do not use boost::remove_erase_if() because it is too new for old systems
* remove undefined function
* I will squash this commit: Update variable names following the naming rule
* [TorqueController] Add instance name to error prefix
* [TwoDofController] Move error_prefix to TwoDofControllerInterface
* [TwoDofController] Add instnace_name to error message
* [TorqueFilter] Add instance name to error message of TorqueFilter
[... cut ...]
* [rtc/AutoBalancer/GaitGenertor.cpp] Use overwrite_footstep_node_list
* [samples/SampleRobot/samplerobot_auto_balancer.py] Add sample for footstep overwriting
* [rtc/AutoBalancer/AutoBalancer.cpp] Enable to pass footstep overwriting from outside of Autobalancer RTC.
* [rtc/AutoBalancer/GaitGenerator.*] Enable to overwrite current footsteps
* use switch instead of if to judge emergency_check_mode
* beep on emergency mode
* Contributors: Fumio KANEHIRO, Isaac IY Saito, Kei Okada, Masaki Murooka, Shunichi Nozawa, YutaKojio, eisoku9618, iori, mmurooka
```

fkanehiro, 2015-08-19:
逆にコロコロバージョンが変わって困る人はいないんでしょうか。
テストで互換性をチェックしているから大丈夫なのかな？IDLが変わっても？

k-okada, 2015-08-19:
https://github.com/fkanehiro/hrpsys-base/pull/762

k-okada, 2015-08-19:
２０分かぁ．面倒が先にきていただけかな．．．
```
catkin_generate_changelog --no-merges
```
してCHANGELOG.rst を直す．一旦ここでcommit．
https://github.com/k-okada/hrpsys-base/commit/3c79dd2434640caecb60451ca60a74eda4345c40
でコミット＆プッシュ `catkin_prepare_release --no-push (真ん中のバージョン番号を上げるときは`--bump minor`を付ける）, CMakeLists.txtのバージョン情報は手動でつける．

これでOK．

k-okada, 2015-08-19:
> 逆にコロコロバージョンが変わって困る人はいないんでしょうか。

基本は大丈夫だと思っていますが，どうでしょうか．
互換性としては 315.1.9 ととっているので，これを使っている（体内にインストールしている限り）
互換性のある形でつかえるはずです．
他に互換性を取っておいたらいいバージョンがあれば，それもテストしたらいいと思っています．というか，JVRCはなんかのバージョンに固定しないと，大変なことになるんじゃないかと思っているけど，大丈夫なんでしょうかね．

yosuke, 2015-08-19:
私もliveは便利とは思うのですが、github派なのでこちらに書いてしまいます（liveには後でまとめを書いておけば良いかな、、、）。

debianオフィシャルレポジトリへの登録は大賛成なのですが、スピード感的にオフィシャルレポジトリはこのスピードを受けつけてくれないので、いずれにせよdaily的なものと併用する感じになると思います。
PPAは心がないバージョンアップデートなので、それを嫌う人はROSパッケージを選べばよいし、選択肢があればよいのではないかな、、、と個人的には思います。

バージョンを固めたい場合はpinしてアップデートを止めてしまえばよかろうとは思うのですが、「昔のこのバージョンだと動いていたのに、、、」みたいなのを取り戻したいときに問題ですよね。
PPAに関しては最新のバージョンしかレポジトリに保存されていないようです。

PS. 今までhrpsys-baseのコミットコメントの書き方が気になっていたのですが、ようやく納得しました。以後、コミットすることがあったら倣います。

yosuke, 2015-08-19:
IDLの変更に関しては、今は @fkanehiro さんが優しい独裁者になって制御している状態だと思うのですが、民主的にやるのであればPRにLGTMコメントを集める感じでしょうか。

みんながすごく依存しているコンポーネントの大会直前アップデート、などあったら悪夢だというのは同意です。

k-okada, 2015-08-19:
> みんながすごく依存しているコンポーネントの大会直前アップデート、などあったら悪夢だというのは同意です。

https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L618
にあるコンポーネントはstable rtcと読んでいて，315.1.9のものと互換性があるようにすること
というルールにしています．なので，
https://github.com/fkanehiro/hrpsys-base/blob/master/.travis.sh#L92
でチェックしていて，この範囲であれば問題はないはずです．
普通に動かす分にはこれでよくてnextageなんかはほとんどの期待は315.1.9で
動いているんじゃないかと思っています．いろいろ考えた，というほどでもないですが，
結局マイクロソフトのAPIチックにXXMethod, XXMethod2 XXMethod3 みたいにして
あるバージョンのAPIは固定しておいて，新しいことがしたければ関数名をかえて，
後方互換性を持っていくのが安全安心ではないかというのが今の結論です

僕が心配することではないんでしょうけど，心配しているのはstable rtcに歩行系の物が含まれていないので，
歩行系のIDLはドンドン変わっています．一応今のところはJVRCでは標準の歩行プログラムは配らないと
聞いているので，全チーム自分たちで歩行プログラムを書いて頑張ることになっているようですが，
やっぱなんかリファレンスが必要だよねとなったら，stableとunstableの間で歩行を含んだ物（stable2みたいなもの）
を作って提供する必要があるんじゃないかと思っています．

> 
debianオフィシャルレポジトリへの登録は大賛成なのですが、スピード感的にオフィシャルレポジトリはこのスピードを受けつけてくれないので、いずれにせよdaily的なものと併用する感じになると思います。
> 
PPAは心がないバージョンアップデートなので、それを嫌う人はROSパッケージを選べばよいし、選択肢があればよいのではないかな、、、と個人的には思います。

ROSのパッケージは，中身はROSに異存していなくて，ROSのbuildシステムを間借りしているだけのものになっています．
なので，debianのオフシャルから提供されれば全く問題なくROSのパッケージは消去します．本当はバイナリを提供しよう
と思った瞬間にオフィシャルに持っていく方法を考えるべきだったとおもうのですが，僕の手間暇と知識だとROSからの
リリースが精一杯なので是非引き継いでオフィシャル側に持って行ってもらえると嬉しいです．
ppaでもわるくもないんですが，rosbridgeを使いたい以上，official 以外のパッケージはちょっと利用しづらいです．
普通のパッケージのように，バージョンが変わったらオフィシャルへ，毎日のものはPPAへ，となっていれば一番
使い勝手が良いです．

snozawa, 2015-08-19:
> 僕が心配することではないんでしょうけど，心配しているのはstable rtcに歩行系の物が含まれていないので， 歩行系のIDLはドンドン変わっています．一応今のところはJVRCでは標準の歩行プログラムは配らないと 聞いているので，全チーム自分たちで歩行プログラムを書いて頑張ることになっているようですが， やっぱなんかリファレンスが必要だよねとなったら，stableとunstableの間で歩行を含んだ物（stable2みたいなもの） を作って提供する必要があるんじゃないかと思っています．

こちらのネーミングの提案なのですが、例えばStabilizerというRTCがあったら、
- stableにする方をStabilizerStable
- unstableで開発続けるほうはStabilizerのまま

とすると、unstableを使いつづける人が移行作業が必要なく、
またStabilizerStableを使いたい人は概ね新規に使う人なので、その名前がはじめからStabilizerStableであっても問題ないのかなと思いました。

fkanehiro, 2015-08-19:
それで現時点での状況としては、ROSとPPAの両方を登録していると、意図しない方のパッケージが入ってしまうといった実害が生じている状況でしょうか。
それとも気持ち悪いので今後整理が必要ですね、という状況でしょうか。

k-okada, 2015-08-19:
後者です。基本的には平和に共存しています

k-okada, 2015-08-20:
と思ったんですが，collada-domはコンフリクトしないですかね？
http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/c/collada-dom2.4-dp/
https://launchpad.net/~hrg/+archive/ubuntu/daily

http://lists.ros.org/pipermail/ros-release/2015-June/004652.html
で議論されているみたいに元々はopenraveのppaから持ってきているので，
hrgのものに切り替えてもらうことは出来そうです．
（と同じものがいろいろなPPAから出てくるので，各プロジェクトでPPA持つのは良くないよねぇと思っているんですが）

で，質問ですが
１）package.ros.orgに入れるとなかなか変更されないので，dailyではなく，
https://launchpad.net/~hrg/+archive/ubuntu/stable
から取っていてもらうようにしたいのですがここにcollada-domが見つからないのはなぜでしょうか？
https://launchpad.net/~hrg/+archive/ubuntu/unstableから持っていけばいいでしょうか？

２）armhf のバイナリもあるといいと思うのですが，可能でしょうか？

yosuke, 2015-08-20:
collada-domですが、rosのパッケージを見たところ最新版が2月更新のもので、hrgの自動ビルドしているものも最終更新がそれぐらいだったと思うので、rosの側で特にパッチが当たっていなければABIレベルで一致しているはずです。パッケージ名も同じ（hrgのものもopenraveのppaのGPLライセンスのdebian定義ファイルから派生させたもの）なので、運よく喧嘩はしない状態ですね。

debian定義ファイルはGPLだったと思うので派生はOKではあると思うのですが、ROSに乗せるとなるとRosen君本人に一言言っておきたいところではあります。

ppa:hrg/stableは頃合いを見てhrg/unstableのものをコピーする、、、というような運用をしようという話だったように思っています。

PPAのビルドに使っているlaunchpadでは残念ながらarmhfのコンパイルサービスは提供されていません。

k-okada, 2015-08-20:
あ，そうですね．だとすると，https://github.com/rdiankov/collada-dom/pull/12
が含まれたものをリリースして貰いたいのが希望です．14.04以降だとこれでないと，時々変なことが起こります．（#7 参照）

Rosen くんにはhttps://github.com/rdiankov/collada-dom/issues/16 で言っては有るのと，そもそも
pakcage.ros.orgにあるのはopneraveのppaから持っていたものになります．

130s, 2015-08-20:
315.7.0 をリリース頂いているようで本題は解決したので閉じます．https://github.com/ros/rosdistro/pull/9243

closed 2015-08-20 by 130s

yosuke, 2015-08-21:
最後に @k-okada さんへの返答。collada-domのパッチへの対応、空き時間ができるまで時間がかかってしまうかもしれませんがやることリストに入れておきます。

k-okada, 2015-08-21:
ありがとうございます．是非JVRC中の作業としておねがいします．JVRCが開催されるのは大きな目的ですが，
それが終わった後に何がのこるか，と考えた時に，ヒューマノイドのシミュレータとコントローラとして
皆が情報交換共有できるプラットフォームとしてこういうのが資産として残って，本当にそれを真剣に使う
チームがこれだけ増えました，となって，日本発のロボットソフトの蓄積が加速してくれるのことを本当に期待しています．
今はJSKは相変わらずros版のhrpsys+openhrp3シミュレータを使っていて，
で，それとは別にchorenoidとhrpsysにパッチ当てて動くものがあって（
https://github.com/fkanehiro/hrpsys-base/issues/760)
でお互い情報交換共有出来る状況にないので，各人バラバラに歩行プログラムからつくっていって，，，
大会でたら３チームそれぞれ別の歩行プログラムで，いきなり転びまくる，という，かなり悲惨な状況なので，
JVRCはこれを変えていく，千載一遇のチャンスと思っています．
％ リソースが足りなければNEDOにお願いしたらいいとおもいます．

まずは
http://fkanehiro.github.io/hrpsys-tutorials/html/index.html
と
http://wiki.ros.org/hrpsys
http://wiki.ros.org/rtmros_common/Tutorials/GettingStart
http://wiki.ros.org/rtmros_nextage/Tutorials/Programming_Hiro_NEXTAGE_OPEN_RTM
あたりのドキュメントの２重化がなくなるようにするところからと思っていますが，そのためにもリリースプロセスを
整理するのが必要だと思って，この話題になりました．

もし，上のドキュメントやリリースプロセス等がJVRC向けで終わったらサポートもないしチョット面倒みれませんねぇ，
となるのであれば，私達は今の仕組みは今後もつかって行くつもりですから，説明文章等等はこちらでもているところに
統合してもらえれば助かりますし，いやいやJVRCおわってもちゃんとやっていきますよ（本来そうであるべきだと思っていますが）
となっているのであれば，ROSで使うhrpsysをubuntu版のものに変えていって，ドキュメント等もroswiki以外の所に集約して
行く方向でこちらが移行するのは是非やりたいと思っています．

yosuke, 2015-08-21:
JVRCには退路を断つためにも自身も参加登録しているので、追い詰められつつもきっと対応すると思います。 :)

例の大会はいよいよプラットフォームロボットとソフトウェアの時代が来たか、、、とある意味ワクワクして見させていただいていたのですがJVRCがソフト寄りの人たちの良いキックオフになると良いですね。