fkanehiro/hrpsys-base #761: PDcontrollerの使い方

YoheiKakiuchi, 2015-08-18:
PDcontrollerの使い方の例が以下になると思うのですが、
~~~
rtmlaunch hrpsys samplerobot.launch TORQUE_CONTROL:=true
rosrun hrpsys samplerobot_walk.py
~~~

この時に、ロボットの初期姿勢が0以外だと、PDControllerのangleRefが更新されずに、関節角度がすべて０の姿勢に戻ってしまいます。
seqやshをつないでからシミュレータを駆動した場合でも、shをアクティベートした瞬間にgoActualするようにしなければならないように思います。
トルクコントロールを用いた場合の初期姿勢を与えたシミュレーションはどのようにしたら良いか、知恵をお持ちの方はいらっしゃいますでしょうか？

snozawa, 2015-08-18:
PDControllerのonActivateの`m_angleRef`を０にセットしている箇所を
```
    for(int i=0; i < dof; ++i){
        m_angleRef.data[i]=m_angle.data[i];
    }
```
としたら解決しますか？
（初期化時の実角度にPDcontroller自体をgoActualさせておく、この値はm_angleRefがisNewになって入るまで固定値でいてくれる）

ちなみに、shははじめにgoActualされてると思います。
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/StateHolder/StateHolder.cpp#L244

YoheiKakiuchi, 2015-08-18:
> PDControllerのonActivateのm_angleRefを０にセットしている箇所を
>    for(int i=0; i < dof; ++i){
>        m_angleRef.data[i]=m_angle.data[i];
>    }
> としたら解決しますか？

とりあえずはこれでいい気がします。

seqやshをつないでからシミュレータを駆動した場合は、shのgoActualの問題でなくて、つながっているコンポーネントのactivateがトリガがくるまで始まらずに、PDcontrollerのangleRefにつながっているコンポーネントすべてがactivateされるまでにシミュレータが進んでしまうためのようです。

snozawa, 2015-08-19:
> とりあえずはこれでいい気がします。

これはとりあえずな解決法というわけでなく、そこそこちゃんとした解決法のつもりでした。
（問題がこれであってますか？という確認のつもりでした）

動いてない状況はchoronoidでしょうか、それとも何かしらの実機でしょうか。

> seqやshをつないでからシミュレータを駆動した場合は、shのgoActualの問題でなくて、つながっているコンポーネントのactivateがトリガがくるまで始まらずに、PDcontrollerのangleRefにつながっているコンポーネントすべてがactivateされるまでにシミュレータが進んでしまうためのようです。

この部分ですが、seq, shがつながってからシミュレータが起動する順番になってるということでしょうか。

hrpsys-simulator, gazebo, （こちらの）実機はすべて
　iobと通信する相手もしくはシミュレータが起動
　seq, shがcreateされ、connectされ、activateされる
という順番になっていると思います。

YoheiKakiuchi, 2015-08-19:
つかっているのはchoreonoidです。

> hrpsys-simulator, gazebo, （こちらの）実機はすべて
>　iobと通信する相手もしくはシミュレータが起動
>　seq, shがcreateされ、connectされ、activateされる
> という順番になっていると思います。

hrpsys-simulatorはちょっとわかりませんが、起動(RTCができる)とシミュレータ時間が動き出すのは別にできるので、
コネクトが全部終わるのを待ってからシミュレータ時間を動かし出すことができます。

> これはとりあえずな解決法というわけでなく、そこそこちゃんとした解決法のつもりでした。

@fkanehiro さん 解決法はこういう方法で良いでしょうか？ご意見があったらおねがいします。

snozawa, 2015-08-19:
> hrpsys-simulatorはちょっとわかりませんが、起動(RTCができる)とシミュレータ時間が動き出すのは別にできるので、
>コネクトが全部終わるのを待ってからシミュレータ時間を動かし出すことができます

これはそうできるというだけであって、しなくても良いんですよね？
だとすると、hrpsys-simulator, gazebo, 実機は
　実機・シミュレータが起動する＝＞rtcd起動＝＞create, connect, activate
の順だと思いますので、choreonoidもその順にして統一する方がよいと思います。

fkanehiro, 2015-08-20:
はい、その解決方法でよいと思います。

closed 2015-08-24 by YoheiKakiuchi