fkanehiro/hrpsys-base #768: SampleRobotのサンプル（テスト）がトルクコントロールのときに失敗する

YoheiKakiuchi, 2015-08-24:
~~~
rtmlaunch hrpsys samplerobot.launch TORQUE_CONTROL:=true
rosrun hrpsys samplerobot_auto_balancer.py
~~~
として立ち上げた時に、以下のgoVelocity and goStopで転倒してしまいます。
~~~
1. AutoBalancer mode by fixing feet
  actual base rpy =  True ( RTC.Orientation3D(r=-0.00010578233627515636, p=-0.006413175695184616, y=0.00016531024624490565) )
  Start and Stop AutoBalancer by fixing feet=>OK
2. AutoBalancer mode by fixing hands and feet
  actual base rpy =  True ( RTC.Orientation3D(r=-0.00010870906638997822, p=-0.0064144292008994856, y=0.00016512771023980107) )
  Start and Stop AutoBalancer by fixing hands and feet=>OK
3. getAutoBalancerParam
  getAutoBalancerParam() => OK
4. setAutoBalancerParam
  default_zmp_offsets setting check in start and stop
5. change base height, base rot x, base rot y, and upper body while AutoBalancer mode
  actual base rpy =  True ( RTC.Orientation3D(r=-0.00010787727843744909, p=-0.007591051551705694, y=0.00016432555573732348) )
6. start stop check
[hrpsys.py] saved data to /tmp/test-samplerobot-abc-startstop-0
[hrpsys.py] saved data to /tmp/test-samplerobot-abc-startstop-1
[hrpsys.py] saved data to /tmp/test-samplerobot-abc-startstop-2
[hrpsys.py] saved data to /tmp/test-samplerobot-abc-startstop-3
  actual base rpy =  True ( RTC.Orientation3D(r=-7.577781917609674e-05, p=-0.004960902418647722, y=0.00045573907719305103) )
7. balance against hand force
  actual base rpy =  True ( RTC.Orientation3D(r=-6.602334666792928e-05, p=-0.006673610562755411, y=0.0004493561062429655) )
1. goPos
  actual base rpy =  True ( RTC.Orientation3D(r=-0.0001103194498185246, p=-0.007441153845909162, y=0.33052766504328934) )
  goPos()=>OK
2. goVelocity and goStop
  actual base rpy =  False ( RTC.Orientation3D(r=0.5294567938720712, p=-0.012168259069332889, y=0.17531927854490995) )
Traceback (most recent call last):
  File "/home/lyouhei/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/samplerobot_auto_balancer.py", line 398, in <module>
    demo()
  File "/home/lyouhei/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/samplerobot_auto_balancer.py", line 382, in demo
    demoGaitGeneratorGoVelocity()
  File "/home/lyouhei/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/samplerobot_auto_balancer.py", line 144, in demoGaitGeneratorGoVelocity
    checkActualBaseAttitude()
  File "/home/lyouhei/ros/hydro_parent/devel/share/hrpsys/samples/SampleRobot/samplerobot_auto_balancer.py", line 44, in checkActualBaseAttitude
    assert (ret)
AssertionError
~~~

snozawa, 2015-08-24:
はい、torque controlだとstabilizerをいれないと歩行できないです。
samplerobot_auto_balancer.pyはHighGainモードでhrpsys-simulatorを利用することを想定しています。

snozawa, 2015-08-24:
auto balancerだけでなくほとんどのサンプルが各RTCの基本機能のチェック・例になっており、
ほとんどでHighGainモードを推奨しております。

Wiki的なもの
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/SampleRobot
http://wiki.ros.org/rtmros_common/Tutorials/WorkingWithEusLisp
もいずれもTORQUE:=trueをつけてません。

y-masutani, 2015-09-06:
Choreonoidとhrpsys-baseを組み合わせて使おうとしているのですが，HGcontrollerを使ってもsamplerobot_auto_balancer.py（choreonoid用に少し変更）の実行中に転倒してしまいます．見た目の印象ですが，hrpsys-simulatorと比べて，足裏と床の間の反発係数か剛性が大きな感じがします（各動作の最後で足裏と床が跳ねています）．どういったパラメータを調整すればいいのでしょうか？質問する場所が違うかもしれませんが，アドバイスをいただけると幸いです．

y-masutani, 2015-09-06:
Choreonoidとhrpsys-baseの組み合わせの件ですが，choreonoidの設定で「動力学モード」を「ハイゲイン動力学」にし，samplerobot_auto_balancer.pyのdemo()の中のdemoAutoBalancerStartStopCheck()をコメントアウトすれば，最後まで転倒せずに実行できることがわかりました．しかし，各動作の終了後にかなりぐらついています．

snozawa, 2015-09-07:
サンプルが動作したようでよかったです。
お試しいただいたようにsamplerobot_auto_baalncerはセンサフィードバックのない歩行軌道生成のサンプルとなりますので、
ハイゲインモードが推奨となっております。

センサフィードバックのサンプルは、
HGcontrollerのかわりにPDcontrollerをいれていただければ
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/SampleRobot
の
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_stabilizer.py
を実行することができます。
私のほうではhrpsys-simulator上での動作確認のみおこなっておりますが、
choreonoidの場合は動力学モードをハイゲインモードでなくする必要があるように思います。
こちらはhrpsys-baseとopenhrp3が最新にしていただくことがありますので、ご注意ください。

y-masutani, 2015-09-07:
ChoreonoidのAISTSimulatorの動力学モードの選択肢には「順動力学」「ハイゲイン動力学」「運動学」の3つがあります．「ハイゲイン動力学」ではなく「順動力学」を選ぶともっと転倒しやすくなります．「運動学」では腰が空間に固定されており，使えません．

それから，PDcontrollerとsamplerobot_stabilizer.pyの組み合わせは既に試しています．AutoBalancerの仕様が次々と変更になりますので，それを確認するために，HGcontrollerとsamplerobot_auto_balancer.pyの組み合わせも試そうとしている次第です．

snozawa, 2015-09-07:
色々と機能をお使いいただき、大変ありがとうございます。

hrpsys-base/rtc以下のRTC群のうち、StateHolder, SequencePlyaer, CollisionDetector, DataLogger, SoftErrorLimitter, ForwardKinematics
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L615
はStableRTCとよんでおり、互換性のためidlレベルの仕様変更を極力しないようにし、
またテストもしているものなります。

一方それ以外はidlレベルでも変更は頻繁に起こることが予想され、特にAutoBalancer, StabilizerはJVRCに向けてもどんどん新機能を追加しようとしてidlレベルから更新があることと思います。

現状動作確認がとれているバージョンがございましたら、そちらのバージョンで留めておくのがもっとも混乱ないかと思います。

closed 2015-11-28 by YoheiKakiuchi