fkanehiro/hrpsys-base #790: Use of dot in configuration variable

yosuke, 2015-09-03:
PDcontrollerのpdgains_sim.file_name値をrtm.pyのsetProperty関数で与えられるようにしようとしているのですが、configuration variableのキー値の中にドットが入っているとうまく外部から動的にセットできないようです。

キー値をpdgains_sim_file_nameと変えてonInitializeの中でbindParameterを使ってあげると動的に取ってこれるようなのですが、これを変えるとほうぼうで影響出ますでしょうか？

```cpp
  bindParameter("dt", dt, "0.005");
  bindParameter("pdgains_sim_file_name", gain_fname, "");
```

多分、OpenRTM-aistの実装がどこかで安易にsplit(".")してしまっているのだろうと思うので悪いのはhrpsysではないと思うのですが。

yosuke, 2015-09-03:
ref #789

snozawa, 2015-09-03:
変更に関しては、

> キー値をpdgains_sim_file_nameと変えてonInitializeの中でbindParameterを使ってあげると動的に取ってこれるようなのですが、これを変えるとほうぼうで影響出ますでしょうか？

はい、ROS関係のdownstreamの方でも変更が必要になりますが、私の知ってる箇所であれば
おっしゃっていただければ対応はできなくはない範囲かと思います。
また、こちらでは@YouheiKakiuchiのほうでchoreonoid対応もさせていただいており、
そちらにも影響がでるため、こちらも@YouheiKakiuhciのほうのタイミングの良いときを待っていただけますと助かります。

そもそも論としまして、この変更は、rtm.pyから動的にゲインをかえたいということでしょうか。
PDcontrolelrはプロジェクトファイルに記述しつつ、preload, precreateしてシミュレータと同時に
あげているのですが、onInitializeもそのタイミングでよばれます。
ですので、rtm.pyを後からよぼうとしたときにはすでにonInitializeが呼ばれているので
意図した変更にならないのではと理解しておりますが、違いますでしょうか。
実機との整合性としてはむしろ、RobotHardwareのsetGainPersentageのようなものが必要なのだろうとは思います。

> 多分、OpenRTM-aistの実装がどこかで安易にsplit(".")してしまっているのだろうと思うので悪いのはhrpsysではないと思うのですが。

":"以外にsplitしているのでしょうか。
こちらはOpenRTMのメーリングリストに聞いていただけますと大変たすかります。
もしくは、プロパティの名前に"."を使ってはいけない、という命名則が実はあったりしますでしょうか。
仕様として可能になっているかどうかは知っておきたいです。
（変更がdownstreamに影響があるのが問題で、"."のあるファイル名には特に理由がございませんでしたので）

yosuke, 2015-09-03:
コンフィギュレーションパラメータにはコンフィギュレーションセットという概念があり、OpenRTMの内部的にはconf.default.pdgains_sim.file_nameという名前で管理されているようです（この例だとdefaultがコンフィギュレーションセットでこれを複数持って切り替えられるらしい）。

おそらく、ですが、上記の管理のためのsplitが悪さしているのだと思います。本来であればsplit(".", 2)とするところだと思うのですが、これが仕様なのかバグなのかはよくわかっていません。

動的にセットしたい、の気持ちは、pdgainファイルをカレントフォルダのものを読んでほしいということで以下のようにしたいという用途です。

```python
pdc.setProperty("pdgains_sim_file_name", os.path.abspath("ROBOT.pdgain"))
```

変更についてはそんなに急いでいません。

snozawa, 2015-09-03:
コンフィギュレーションについてご説明いただきありがとうございます。

> 動的にセットしたい、の気持ちは、pdgainファイルをカレントフォルダのものを読んでほしいということで以下のようにしたいという用途です。

に関してなのですが、
https://github.com/fkanehiro/hrpsys-base/issues/790#issuecomment-137316629
の
> ですので、rtm.pyを後からよぼうとしたときにはすでにonInitializeが呼ばれているので
> 意図した変更にならないのではと理解しておりますが、違いますでしょうか。

の質問としましては、
1. hrpsys-simulator + PDcontroller
2. rtm.py

の順で起動したとしますと、1の段階ですでにPDcontrollerのonInitializeが呼ばれるため
２の段階でsetPropertyしてもゲインはかわらないのではないでしょうか。
また、2のようなタイミングで起動時より少したってパラメータをセットするのでしたら、
実機ではsetServoGainPercentageをサービスポートでよんでパラメータを変更しておりますので、
PDcontrollerも同様の機能で行うのが良いと思います。

yosuke, 2015-09-03:
今は、onInitlializeの中で
```cpp
  RTC::Properties& prop = getProperties();
  coil::stringTo(dt, prop["dt"].c_str());
  coil::stringTo(gain_fname, prop["pdgains_sim.file_name"].c_str());
```
としていると思うのですが、この部分を以下のように変えると、変数gain_fnameの中に任意のタイミングでセットした最新のコンフィギュレーションパラメータが格納されてくれます。
```cpp
  bindParameter("dt", dt, "0.005");
  bindParameter("pdgains_sim_file_name", gain_fname, "");
```
なので、こうするとonActivateで外部からセットしたgain_fnameを読んでもらえるので、都合が良いのです。

上記の変更で下記の流れでgain_fnameを読み込んでくれることは確認しています。
```python
pdc.setProperty("pdgains_sim_file_name", os.path.abspath("ROBOT.pdgain"))
pdc.start()
```

snozawa, 2015-09-03:
bindParameterとonActivateの利用はそうなるとは思いますが、
少し実際の起動手順でどういったものを想定されてますでしょうか。

https://github.com/fkanehiro/hrpsys-base/pull/781#issuecomment-136344330
で@fkanehiroさんのコメントにもあるように、ロボットがくずれ落ちないようになっており、
PDcontrollerがhrpsys-simulator起動時にパラメータセットとonInitizlieとonActivateが
よばれると思います。

1. そのため、
 1. hrpsys-simulator + PDcontroller (のonInitialize, onActivate)
 2. rtm.py でsetProperty+start (PDcontrollerのonActivateがよばれる)  
という手順でしょうか。
2. iとiiの間はどれくらいのタイムラグになりそうでしょうか。
3. iでonActivateされているので、rtm.pyではstopを一度してからstartはしなくて良いのでしょうか

yosuke, 2015-09-03:
こちらの起動手順ですが、hrpsysではなくChoreonoidとの接続を試しており、

i. rtm.pyでsetPropertyだけしてstartはしないで待機

ii. Choreonoid側で以下の設定ファイル＋BodyRTCを作ってシミュレーションスタート（するとPDControllerをActivateしてくれる）

```
in-port = u_in:JOINT_TORQUE
out-port = q:JOINT_VALUE
out-port = lhsensor:lhsensor:FORCE_SENSOR
out-port = rhsensor:rhsensor:FORCE_SENSOR
out-port = gyrometer:gyrometer:RATE_GYRO_SENSOR2
out-port = gsensor:gsensor:ACCELERATION_SENSOR2
connection = q:PDcontroller0:angle
connection = u_in:PDcontroller0:torque
```

という流れで使っています。

これが正しい使い方なのかどうかというとかなり自信が無いです。

yosuke, 2015-09-03:
あと、場合によって「Gain file [....] opened」と出てくれる場合とそうでない場合があり、、、うまく読んでくれると制御が効くのですが、高確率で強すぎるトルクがかかってロボットが吹っ飛びます。

トルクのリミット処理を入れたほうが良いかもしれません。

yosuke, 2015-09-03:
「Gain file [....] opened」が出てくれない理由については
```cpp
  if(m_angleIn.isNew()){
```
となっていてdofを読んでいるのでangleに事前に少なくとも1回はメッセージが来ていないといけないのだと理解しました。

yosuke, 2015-09-03:
上記のangleに事前にメッセージが必要な件なのですが、hrpsys-simulatorだと安定して動いているということでよいでしょうか？

Choreonoidで上記の形で使うと、ほとんどの場合、メッセージが先に来てくれません。

yosuke, 2015-09-03:
あれこれコミットをしていますが、ご意向伺いのような形です。

より良い形があったら、そちらが良いと思います。

snozawa, 2015-09-03:
https://github.com/fkanehiro/hrpsys-base/issues/790#issuecomment-137343360
ご説明いただきありがとうございます、手順がわかりました。

（以下、こちらの質問項目が多くなっておりますが）

choreonoidで行う場合、https://github.com/fkanehiro/hrpsys-base/issues/790#issuecomment-137343360
の手順iでPDゲインファイルのパスを指定することが、confファイルでは不可能で、
コンフィギュレーションの機能以外にないということなのでしょうか。

こちらでは、現状
- RTC起動時にはいってて欲しいパラメータはconfファイルに記述し、指定する
- 動的にかえたいパラメータはサービスポートで設定

としております。
前者に関しては、こちらのhrpsys-abseを使わせていただく前よりございました
HRP４Cのサンプルが
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/HRP4C/HRP4C.conf.in
を読み込む形になっており、SequencePlayerなども
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/SequencePlayer/SequencePlayer.cpp#L112
それを使うかたちになっておりました。
当方ではそれに基づいて、基本的には起動時にセットされててほしいパラメータは
起動時（onInitializeなど）にスタティックにきめる方法に統一しております。

今回のissueとhttps://github.com/fkanehiro/hrpsys-base/pull/789 のご提案にはpdgain fileとdtに関してプログラム変更案が記載されておりますが、
- PDゲインのパスを適当なところに設定したい
- https://github.com/fkanehiro/hrpsys-base/issues/790#issue-104600629　のように、confファイルで設定していたパラメータをsetPropertyで設定できるようにしたい

とがあるように見受けられます。

おそらくご要望とされていることは前者はとは思いますが、
もしbindParameter + setPropertyで行う場合は、pdゲインファイルのみ設定としていただくのがよろしいかと思いますが、いかがでしょうか。

後者の場合(dtも変更する)、PDcontrollerおよび`pdgains_sim.file_name`だけに止まらない大きな変更のように思います。
また、どうしてもchoreonoidではconfファイルのパスを指定できないという場合でなければ、
パラメータ設定の仕方が多岐に渡ると、それはそれでユーザにとっては混乱になるかとは思いますし、
上記のようにSequencePlayerをはじめほとんどのRTCで変更が必要になり、
またonInitializeでconfファイルから読み込んだ`model`, `dt`を利用しているため
ライフサイクルも見直す必要があります。

snozawa, 2015-09-03:
angleの件ですが、
> 上記のangleに事前にメッセージが必要な件なのですが、hrpsys-simulatorだと安定して動いているということでよいでしょうか？

hrpsys-simulatorでは動いておりますが、おっしゃるとおり確かに問題があるように思いました。

元々はangleの長さを取得しつつチェックもすることを糸しておりましたが、
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/HGcontroller/HGcontroller.cpp#L125
のようにonExecuteでangleがisNewになったときにangleの長さを取得して対処する、という方式が
hrpsys-base, choreonoidのどちらでも動くため、ベストなように感じました。
しばしお待ちいただくので構いませんでしたら、こちらで修正はできます。

yosuke, 2015-09-03:
bindとsetPropertyの件ですが、排他的に利用する（confが使えなくなってしまう）というわけではなく、confだけでなくsetPropertyでも使えるようになる、という変更ですので、影響範囲はそこまで大きくはないのではないかな、とは思っています。ただ、キー値にドットがあるとsetProperty側が使えないので、簡単に修正できるのであれば変えてほしい、、、という要望です。

bindParameter方式はhrpsysの他のコンポーネントでも使われているのですが、利用方針は開発者によるということでしょうか？

https://github.com/fkanehiro/hrpsys-base/search?q=bindParameter

yosuke, 2015-09-03:
動的にセットしたい、、、と書いてしまったので誤解がありそうな気がしているのですが、onInitializeのあとで、onActivateされる前にパラメータをセットしたい、というだけで、onActivate後の変更については私については必要ありません（その部分はおっしゃるとおりサービスポートでの実現が良いと思います）。

rtc.confの設定はとにかく不便なので別の方法でもできるようになっているとありがたい、というのが意図です。

snozawa, 2015-09-03:
こちらまだ議論させていただきたい点がございますが、
お待たせするのもアレですので、取り急ぎpdgainファイルのみbindParameter+setPropetyの方式にして
PRしていただけますと幸です。

以下は、confでかいてあることを全て（排他的でなく）bindParameterサポートしていくかという点についてです。

> bindとsetPropertyの件ですが、排他的に利用する（confが使えなくなってしまう）というわけではなく、confだけでなくsetPropertyでも使えるようになる、という変更ですので、影響範囲はそこまで大きくはないのではないかな、とは思っています。ただ、キー値にドットがあるとsetProperty側が使えないので、簡単に修正できるのであれば変えてほしい、、、という要望です。

もちろんconfファイルがdeprecatedになるとは思いませんでしたが、
同じことをする方法が複数種類に増えると少し混乱が生じると思っております。
（値がsetPropertyでセットされていることをうっかりわすれて、confファイルを探して値を設定するけど反映されない、など）

おっしゃるようにconfファイル自体はそれほど便利ではないですし、
そもそもOpenRTMでconfもコンフィギュレーションもどちらも利用可能な仕様ですので、
bindParameter自体に問題があるとも思っておりません。

ただ現状confファイルを前提にしているRTCも多く変更も多そう、というのを懸念しております。

> bindParameter方式はhrpsysの他のコンポーネントでも使われているのですが、利用方針は開発者によるということでしょうか？

質問に質問でかえす形で恐縮ですが、こちらに関しては、おそらく同じ方（グループ）で開発されてる部分でbindParameterとconfを使いわけておられる理由を逆にお伺いできればと思います。
開発元が同じ
https://github.com/fkanehiro/hrpsys-base/search?q=bindParameter
はパラメータ群はbindParameterを利用し、
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/SequencePlayer/SequencePlayer.cpp#L124
はconfを使っております。
ポート設定(InPort)やModelLoader利用など、onInitializeでやっておきたいことはコンフィギュレーションで可能なのでしょうか。

また、そもそもこちらでコミットさせていただいておりますRTC群も、
元からございましたSequencePlayerやHRP4Cサンプルなどを参照しつつ
見よう見まねで作成しているところがございます。
hrpsys-base(openhrp3)レベルの例でいうと、
- 初期化的なもの、基礎的なパラメータのセット方法等、onInitialize、onActivatedでそれぞれ何をするべきで
- conf、コンフィギュレーション、サービスポートではそれぞれ、どういった種類のパラメータを設定するようにするのが普通で

といった指針がございましたらご教示いただけますと幸いです。

最後にこちらの使い方で申しますと、動的にパラメータを設定する用途でサービスポートを利用し、
これはROSと相互利用できるようになっております。
スタティックに設定、もしくは起動次に設定したいパラメータはconfファイルにしつつ、パス解決でROSを利用しておりますので、こちらもROS相互利用可能といえる状況にございます。
一方で、コンフィギュレーションはまだROS相互利用できてないので、こちらとしてはまだconfを利用することになるとは思います。

yosuke, 2015-09-03:
経緯については @fkanehiro さんに聞くのが良いと思うのですが、私が見た感じbindParameterはOpenRTM-aist 1.1からの新機能なので、新しく作られたコンポーネントはなるべくこれを使うように実装してある。それ以前からあるコンポーネント（SequencePlayerなどはずいぶん歴史があるコンポーネントだと思います）は、bindParameter登場以前の実装が残っている、という感じなのではないかなと思っています。

ただ、間違っている可能性は十分あるので @fkanehiro さんにコメントいただくのが一番よいです。

snozawa, 2015-09-04:
pd_gainファイルの指定は現状のconfファイルにパスがかいてあり、さらにconfファイル自体のパス指定も必要でダブルで不便とは思います。
ですので、取り急ぎはdtは今のままで、pd_gainファイルの指定はbindParameterとなるようにさせていただければと思います。

> 私が見た感じbindParameterはOpenRTM-aist 1.1からの新機能なので、新しく作られたコンポーネントはなるべくこれを使うように実装してある。それ以前からあるコンポーネント（SequencePlayerなどはずいぶん歴史があるコンポーネントだと思います）は、bindParameter登場以前の実装が残っている、という感じなのではないかなと思っています。

はい、bindParameterは標準的機能ですし、confと両立はできるとは思いますのでお使いいただくので問題ないとは思いますが、

- confファイルだけでなくbindParameterで値を設定する場合、現状dt, modelなどはonInitializeの中でメンバ変数初期化に使っているものはbindParameterを理容可能でしょうか。具体的には、onInitializeのbindParameter記述以降でコンフィギュレーションが設定できるので、その後外のrtm.pyからすかさず値をsetPropertyしてあげる必要があると思いますが、タイミングよくできるものなのでしょうか。 > @yosuke様
- bindParameterとconfで使い分けはどのようにされてますでしょうか。指針等ございましたらお教えいただけると助かります。> @fkanehiro様

yosuke, 2015-09-04:
onInilializeはコンポーネントをcreateした時に呼ばれます、onActivateはstartした時に呼ばれるので、こちらでは以下の形で問題なく使えています。

```python
mgr = rtm.findRTCmanager()
mgr.load('PDcontroller')
pdc = mgr.create('PDcontroller')
pdc.setProperty("dt", "0.001")
pdc.setProperty("pdgains_sim_file_name", os.path.abspath("ROBOT.pdgain"))
pdc.start()
```

実際には最後のpdc.start()は書かれてなく、ChoreonoidにActivateしてもらう形です。

少なくともPDcontrollerについてはonActivateでしかパラメータを使っていないと思うのですが、onInitializeでパラメータを使う処理を入れたい、みたいな話でなければ上の形でうまくいくはずです。

snozawa, 2015-09-04:
はい、PDcontrollerのdtは実際値が使われるのはonExecute内が最初ですので、
bindParameterでセットしても問題なく動くと思います。

SequencePlayerなど既存のdt, model, ...をconfからよんでいるものはbindParameterを
```
mgr = rtm.findRTCmanager()
mgr.load('SequencePlayer')
seq = mgr.create('SequencePlayer')
seq.setProperty("dt", "0.001")
```
のようにセットすると、
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/SequencePlayer/SequencePlayer.cpp#L163
`mgr.create`がよばれた際のonInitializeでdt, modelの値を使い
`seq.setProperty`でセットされる前に値を使ってしまうきがしますが、動作させることは可能なのでしょうか。

PDcontrollerの`dt`をbindParameterも対応させるのであれば、
今`dt`がはいってるすべてのRTCでも対応させる必要があります。
bindparameterで適切にセットできれば問題ないと思います。

yosuke, 2015-09-04:
> PDcontrollerのdtをbindParameterも対応させるのであれば、
> 今dtがはいってるすべてのRTCでも対応させる必要があります。

これ、本当でしょうか？

RTMの仕様上できない、という話なのか統一感がないのでここだけ許可したくない、という話なのか明確になると良いと思います。

snozawa, 2015-09-04:
> RTMの仕様上できない、という話なのか統一感がないのでここだけ許可したくない、という話なのか明確になると良いと思います。

こちらは後者です。
PDcontrollerの`dt`と他のRTCの`dt`、および`model`などその他のパラメータをセットするものは
やりかたを統一したいです。

yosuke, 2015-09-04:
なるほど、理解しました。

coil::stringTo(dt, prop["dt"].c_str());としているコンポーネントもかなりの数がありますし、それぞれ理由もありそうですね。

PDcontrollerなどはよく使うコンポーネントですし単純な構造をしているので両対応してほしい、という気持ちもありますが、この部分は方針に従います。

rtc.confはむやみに複雑なファイル構造だったり、間違えやすキーの定義方法なのでできるだけ使用箇所を減らしたい、、、という気持ちがあるので、自分が作るものに関しては両対応で作ります。

よろしくお願いします。

snozawa, 2015-09-04:
もちろんbindParameterでできるのであれば対応可能とは思いますが、
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/SequencePlayer/SequencePlayer.cpp#L163
のようなonInitializeでセットされているものをbindParameter的なかんじでセットする方法は、本当にOpenRTMではないのでしょうか。
bindParameterでもよいのですが、できるだけ各RTCでやりかたを統一したいというのもあり、confファイルで記述できてbindParameterで記述できないことがある以上、少し対応が難しい点ご理解いただけますと幸いです。

よろしくお願いします。

yosuke, 2015-09-04:
coil::stringTo(dt, prop["dt"].c_str());の箇所が思いのほか多かったので、すぐに変えられない理由は理解しました。

パラメータをセットする方法はrtc.confから読むかCORBA IDLで定義された関数から外部でセットするか（ただこれはonInitializeの後になってしまう）、あとはコマンドラインからセットする方法もあったような、、、しか私の知っている限りではないと思います。config情報をセットして共有しておける別プロセスのサーバなどあれば良いと思うのですが、それがないのですよね。

汚いworkaroundですが、pythonでrtc.confを生成して、、、みたいなのはできると思うのですが負けた感じがするのでやりたくないです。

yosuke, 2015-09-04:
managerの実装を見るとcreate_component関数がコンポーネント名のあとにパラメータを設定できるようになっているようで、ここでもセットは可能かもしれませんね。これであれば少しは綺麗かもしれません。

http://svn.openrtm.org/OpenRTM-aist/branches/RELENG_1_1/OpenRTM-aist/src/lib/rtm/ManagerServant.cpp

snozawa, 2015-09-04:
> coil::stringTo(dt, prop["dt"].c_str());の箇所が思いのほか多かったので、すぐに変えられない理由は理解しました。

確かに変更箇所がおおいのはアレですが、
（本issueで長々と書いてしまいましたが）最終的な懸念事項は１点で、confで記述できるけどbindParameterで記述できないものがあるため両方対応が難しい、という点です。

> パラメータをセットする方法はrtc.confから読むかCORBA IDLで定義された関数から外部でセットするか（ただこれはonInitializeの後になってしまう）、あとはコマンドラインからセットする方法もあったような、、、しか私の知っている限りではないと思います。config情報をセットして共有しておける別プロセスのサーバなどあれば良いと思うのですが、それがないのですよね。

やはり、onInitializeでのパラメータセット方法は限られているのですね。

> 汚いworkaroundですが、pythonでrtc.confを生成して、、、みたいなのはできると思うのですが負けた感じがするのでやりたくないです。

こちらでは
https://github.com/fkanehiro/openhrp3/blob/master/server/ModelLoader/projectGenerator.cpp
を利用しております。
https://github.com/fkanehiro/openhrp3/blob/master/server/ModelLoader/projectGenerator.cpp#L621
ここで、プロジェクトファイルを生成している他、confファイルも生成しており、
各人のPCのモデルファイルパスが入るようにしております。
また
https://github.com/fkanehiro/openhrp3/issues/55
で環境変数のような形が議論されておりまして、パス解決方法自体は色々手段が選べるように思います。

yosuke, 2015-09-04:
コマンドラインからセットする方法も超複雑ですね、、、。同じようにcreate_componentしてみるも有効にならず。

http://www.openrtm.org/openrtm/ja/content/openrtm-users-01723-rtcconf-%E3%81%AB%E3%81%A4%E3%81%84%E3%81%A6

負けを認めたくないと思いつつrtc.confを生成する方法でいくかもしれません（projectGeneratorの参照情報もありがとうございます）。

fkanehiro, 2015-09-04:
すみません、だいぶ出遅れて今更ってかんじですが、こちらでは、
confにはモデルのパスやサンプリング時間等多数のRTCが共有し、かつシステムが稼働中には変わることがない情報を書き、
setProperty()はシステムが稼働中に変えたり、アプリごとに変えたりするようなパラメータ設定に使う、
という使い分けになっています。ただこれも、こちらでは試行錯誤してるうちにそういう風になってきた、という程度のもので、OpenRTMの公式ガイドでも無いですし、紆余曲折があった関係でいまあるRTCでもその方針通りになっていないものもあったりします。

yosuke, 2015-09-04:
@fkanehiro さん

ついでなので質問なのですが、onInitializeに初期化処理が書かれているコンポーネントはシビアな制御系が多かったので気になったのですが、onActivateで大きな初期化処理を書くとActivationが遅れてロボットが倒れてしまう、、、みたいなのはあるでしょうか？

fkanehiro, 2015-09-04:
リアルタイムOSでは動かしている場合は、onActivate()は周期実行しているリアルタイムスレッドから呼ばれるので、そこにModelLoaderの呼び出しとかを書いてデッドラインをミスする、というのは避けたいというのは有ります。
リアルタイムで使うコンポーネントのアクティベートはシステムの初期化の段階でロボットが静止している時に行う場合がほとんどなので実害はほとんどないのでしょうけれど。

YoheiKakiuchi, 2015-09-04:
> i. rtm.pyでsetPropertyだけしてstartはしないで待機
> ii. Choreonoid側で以下の設定ファイル＋BodyRTCを作ってシミュレーションスタート（するとPDControllerをActivateしてくれる）

サンプルと言うほどまとまっていませんが、以下にJVRCに参加するためのプログラムを置いてあり、
https://github.com/start-jsk/rtmros_choreonoid

以下が、choreonoidでPDコントローラを使っている例になります。
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_ros_bridge_jvrc/config/jaxon_pdO1.cnoid

当方では、BodyRTCにPDControllerを指定しているのですが、その場合は、onActivateのときに
```if(m_angleIn.isNew()){```になっており、失敗したことはないように思います。
デバッグメッセージを見ると、同じタイムスタンプのm_anglesがonActivateとonExecuteの1回目に使われているようです。

> onActivateで大きな初期化処理を書くとActivationが遅れてロボットが倒れてしまう、、、みたいなのはあるでしょうか？

choreonoidの上記の設定限定の話ですが、その場合はsleepを入れてみたことがあるのですが、onActivateがブロックしてchoreonoidのシミュレーション周期も止まり、そのあとで、同じタイムスタンプのm_anglesがonActivateとonExecuteの1回目に使われてシミュレーションが回り始めてるようです。

yosuke, 2015-09-04:
完全に必勝法になっているレポジトリ、紹介ありがとうございます。 :)

ChoreonoidにPDcontrollerを生成してもらうとonActivateの時にメッセージが来てくれるのですね。私の方ではrtm.pyの方でコンポーネントを作った関係でmoduleNameのところが空欄になっていました。

ただタイムスタンプの件については同じものがonActivateとonExecuteに二度来るというのは若干違和感があります。実際を考えるとonExecuteは次のサイクルなのでないかな、という違和感なので全く直す必要はないのですが。

onActivateが大きくなるのはchoreonoidに関しては問題なさそうですね。

snozawa, 2015-09-04:
@fkanehiroさん

https://github.com/fkanehiro/hrpsys-base/issues/790#issuecomment-137653772
コメントありがとうございます．参考になります．
こちらでもシステム稼働時に決定して変化しないのはconfと思い，こちらでコミットさせていただいているＲＴＣ群もその方法になっております．

snozawa, 2015-09-05:
@yosuke 様
https://github.com/fkanehiro/hrpsys-base/pull/793
で
- onExecuteで関節角度値が初期化されるようにしました。
- bindPaameterでPDゲインファイルをセットできるようにしました(pdgains_sim_file_name)
- トルクリミット値をVRMLのclimitなどにかくと設定できるようにしました。

を修正しましたので、ご確認いただけますと幸いです。

yosuke, 2015-09-07:
確認しました。良いのではないかと思います。

#796 は関連するissueですね。