fkanehiro/hrpsys-base #796: PDcontrollerが原因でChoreonoidがコアダンプ

y-masutani, 2015-09-06:
当方では，choreonoidのBodtRTC用のコントローラとしてPDcontrollerを使っていますが，hrpsys-baseを更新したところ，シミュレーション実行時にコアダンプするようになりました．他のRTCを何も起動しない場合でも，問題は発生しますので，PDcontrollerが原因ではないかと思います．choreonoidを起動した端末には以下のようなメッセージが表示されています．

PDcontroller0: onInitialize() 
omniORB: ERROR -- the application attempted to invoke an operation
 on a nil reference.
CORBA::SystemException raised by ModelLoader: IDL:omg.org/CORBA/INV_OBJREF:1.0
[PDcontroller0] failed to load model[file:///usr/local/share/OpenHRP-3.1/sample/model/sample1.wrl]
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
PDcontroller0: on Activated 
[PDcontroller0] Gain file [SampleRobot.PDgain.dat] opened
Segmentation fault (コアダンプ)

どのように対処すればいいでしょうか？

yosuke, 2015-09-07:
#793 の変更が影響していると思います。

起動時に別のウィンドウでopenhrp-model-loaderを起動しておくことと、rtc.confにモデルファイルの設定をしておく必要があるのだと思うのですが、 @snozawa さんからフォローアップがあると思います。

k-okada, 2015-09-07:
運営の側でどのバージョンを使うとかそういうのはないのでしょうか？
あるいは運営の側でtravisに何かテストを追加いただくみたいなものでもいいかもしれないです．

yosuke, 2015-09-07:
多分ここにヌルチェックが必要なのだと思います。

https://github.com/fkanehiro/hrpsys-base/pull/793/files#diff-2179ca2958cd872d18336e0b7feb0497R152

モデルのロードに失敗した場合（存在しないファイルのURLが指定されていた場合）のテストとかはあったほうが良いですかね。

y-masutani, 2015-09-07:
当方では， http://fkanehiro.github.io/hrpsys-tutorials/html-ja/choreonoid/toChoreonoid.html にしたがって，choreonoidの中からopenhrp-model-loaderを起動していました．これをやめて，別のシェルからopenhrp-model-loaderを起動するよと，コアダンプしなくなりました．ただ，on Initializeとon Activatedのメッセージの間に以下のような表示は残ったままです（以前はなかったと思います）．

PDcontroller0: onInitialize() 
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
PDcontroller0: on Activated

y-masutani, 2015-09-07:
「運営の側のバージョン」とか「travis」というのは何のことかわかりません．

snozawa, 2015-09-07:
@y-masutani様
遅くなりました。
https://github.com/fkanehiro/hrpsys-base/issues/796#issuecomment-138152457
でコメントいただいているように、
openhrp-model-loaderを起動していただく必要があります。

> choreonoidの中からopenhrp-model-loaderを起動していました．

@fkanehiro様、@s-nakaoka様
choreonoidでの openhrp-model-loaderの起動順序は、PDcontrollerよりも前ではないのでしょうか。

> 「運営の側のバージョン」とか「travis」というのは何のことかわかりません．

こちらは、@fkanehiro様、@s-nakaoka様、@yosuke様向けのコメントと思います。
hrpsys-baseとの連携は（公式サポートではないとは存じますが）どのあたりまで想定されることになりますでしょうか。
おそらくPDcontrollerまではあるのかなとかんがえております。

y-masutani, 2015-09-07:
なんどもスミマセン．「rtc.confにモデルファイルの設定」とはどういうことでしょうか？当方では，rtc.confの中に
example.PDcontroller.config_file: XXX.conf
と書き，XXX.confの中でwrlファイルのURLを指定しています．

yosuke, 2015-09-07:
> 「rtc.confにモデルファイルの設定」とはどういうことでしょうか？当方では，rtc.confの中に
> example.PDcontroller.config_file: XXX.conf
> と書き，XXX.confの中でwrlファイルのURLを指定しています．

私もこのやり方がよくわかっておらず、、、なところで拙い説明をしています。
XXX.confの中まで設定方法がわかるとうれしいです。

y-masutani, 2015-09-07:
XXX.confの中には以下のように書いています．
model: file:///usr/local/share/OpenHRP-3.1/sample/model/sample1.wrl

yosuke, 2015-09-07:
情報有り難うございます。設定通りのファイルを読み込もうとしているようで、良さそうですね。

今回の変更はモデルファイルに以下のようなclimit属性があることを前提にしているようなのですが、sample1.wrlにはそれがなさそうなのが気になっています。
https://github.com/start-jsk/rtmros_choreonoid/blob/master/jvrc_models/JAXON_JVRC/JAXON_JVRCmain.wrl#L342

ここでヌルチェックっぽい処理は入っていそうなのですが、、、。
https://github.com/fkanehiro/openhrp3/blob/ae340ee3ec31ed48888fcc75bfaec5678288be34/server/ModelLoader/BodyInfo_impl.cpp#L369

yosuke, 2015-09-07:
エラーメッセージの出処はここみたいですね。
シミュレーションスタートのタイミングが変わるとでる、、、ということなのか、、、よくわかりませんね。

https://github.com/s-nakaoka/choreonoid/blob/34a389d7cb8c74d094e3613bf7e72d251428aff4/src/BodyPlugin/GLVisionSimulatorItem.cpp#L343

yosuke, 2015-09-07:
cloneMap.setNonNodeCloning(false);のエラーメッセージについてはVisionSimulatorが出すようで、ちょうど2個出ていると思うのですが、カメラ2つ分のVisionSimulatorを作成したということはないでしょうか？

snozawa, 2015-09-07:
confファイルに関しては、rtc.confに
```
example.PDcontroller.config_file: XXX.conf
```
のように記述していただくと、PDcontrollerが`XXX.conf`を読み込むようになります。
各RTCが中で使う設定はこちらの`XXX.conf`のファイルに記述しておきます。

> XXX.confの中には以下のように書いています．
> model: file:///usr/local/share/OpenHRP-3.1/sample/model/sample1.wrl

とあり、また
https://github.com/fkanehiro/hrpsys-base/issues/796#issue-105090889
を拝見しますと
```
 [PDcontroller0] failed to load model[file:///usr/local/share/OpenHRP-3.1/sample/model/sample1.wrl]
```
とありますので、`XXX.conf`の記述方法およびその読み込ませ方は問題なく、
openhrp-model-loaderがなかったことが問題ということと思います。
https://github.com/fkanehiro/hrpsys-base/issues/768
のサンプルが動いていることからもconfの設定方法は問題ないと思います
（#768のサンプルは`XXX.conf`が設定されている必要があります）

confの内容に関しては、hrpsys-baseのトップページ
https://github.com/fkanehiro/hrpsys-base
のAPI DOCのところからたどると、Doxygenのドキュメントページがでてきます。
各RTCで
http://fkanehiro.github.io/hrpsys-base/dd/d91/SequencePlayer.html
の"Configuration file"のようにどのプロパティ情報がどう使われているかが記載してあります。

また、全体的にドキュメントは充実させきれておりませんが、
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/HRP4C
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/SampleRobot
などのサンプル実行もご覧いただいていると思いますので、こちらの方も今後ともご参照いただけますと助かります。

ちなみにですが、ドキュメントは人手で記述しているため誤りや最新内容からの遅れはございますが、
上記のサンプル（主にSampleRobot）はPullRequest時にテストも行っており、
最新内容にそれなりに追いついている使用例となっております。

y-masutani, 2015-09-07:
皆様，色々とご教示いただき有り難うございます．
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
の件ですが，choreonoidの中でGLVisionSimulationItemを削除すると出なくなりました．PDcontrollerのメッセージの間に表示されていたので，関係あるのだろうと思い込んでいました．申し訳ありません．

fkanehiro, 2015-09-07:
hrpsys-baseの使用を義務付けているわけではないので、バージョンの指定はありません。
choreonoidのシミュレーション機能については、問題が生じた場合に修正をする程度ですので、最新版をお使い頂ければ大丈夫かと思います。

y-masutani, 2015-09-27:
Choreonoidの中でopenhrp-model-loaderを起動するとコアダンプする件の続きです．
JVRCでは，競技PCではChoreonoidを実行することしか許されないので（審判が操作するのだと思います），openhrp-model-loaderはChoreonoidから起動しないといけないのですが，何かやり方がないでしょうか？

そもそも，なぜopenhrp-model-loaderを予め起動しておかないと，Choreonoidがコアダンプするのかわかっていなので，私には対処のしようがありません．ご助言をいただけると幸いです．よろしくお願いいたします．

snozawa, 2015-09-27:
https://github.com/fkanehiro/hrpsys-base/pull/827
にてmodel-loaderがない場合でも落ちないようにしつつ、
ただ動作としてはmodel-loaderを前提にしている箇所がこれ以外（他RTC）にございますので、
ワーニングを出すようにPDcontrollerを修正しました。

またchoreonoid環境に関しては私自身でも分からない点がございましたので、
openhrp-model-loaderが起動できない点の報告はchoreonoid本家の
https://github.com/s-nakaoka/choreonoid/issues/37
に作成いたしました。

よろしくお願いいたします。

y-masutani, 2015-09-27:
対応ありがとうございます．最新版をpullして試しました．openhrp-model-loaderをchoreonoidの外部コマンドとして登録し，choreonoidから起動するようにしても，シミュレーションでコアダンプしなくなりました．シミュレーションも正しく動いているように見えます．choreonodのメッセージ欄にはopenhrp-model-loaderが出力する「loading file:...」や「cache found for file:...」が表示されています．しかし，choreonoidを起動した端末には，
[PDcontroller0] m_robot is not set properly!! Maybe ModelLoader is missing?
が延々表示されるのが気になります．openhrp-model-loaderが起動されてからもずっと出ています．他のメッセージを見逃してしまいかねません．なんとかならないでしょうか？

snozawa, 2015-09-28:
その場合は
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/PDcontroller/PDcontroller.cpp#L165
のあたりをコメントアウトしてビルドしなおすというのをお試しいただけますと幸いです。
こちらの作業で出力がでなくなると思います。

ただ、PDcontrollerをはじめとしてopenhrp-model-loaderなしでは動作ができないものが
ほとんどですので、この辺をお使いの場合はやはりopenhrp-model-loaderと併用できる必要があり、
（RTC側が）意図した状況でない場合にはあえて気になるようなワーニングメッセージをだすようにしております。
ちなみにですが、CollisionDetectorという自己干渉回避のためのフェイルセーフRTCも、同様に
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/CollisionDetector/CollisionDetector.cpp#L328
のようなワーニングメッセージを干渉ペアを定義してない場合に出力するようにしております。

@s-nakaoka様、@fkanehiro様
https://github.com/s-nakaoka/choreonoid/issues/37
の方もよろしくお願いいたします。

fkanehiro, 2015-09-28:
> s-nakaoka/choreonoid#37

@y-masutani様、

Choreonoidの中でopenhrp-model-loaderを起動するとコアダンプする、は
openhrp-model-loaderを上げておかないとChoreonoidがコアダンプする、
の書き間違えではないでしょうか。

y-masutani, 2015-09-28:
わかりにくくて申し訳ありません．
Choreonoidの中でopenhrp-model-loaderを起動して，シミュレーションを実行すると，そのタイミングでコアダンプしていました（#827以前）．Choreonoidを起動するより前に，openhrp-model-loaderを実行しておくと問題はありません．

y-masutani, 2015-09-28:
@snozawa 様，
メッセージが全てダメということではなく，openhrp-model-loaderが起動した後もなぜ「Maybe ModelLoader is missing?」と言い続けるのかということです．ロジックを理解していませんので，見当違いなコメントならば申し訳ありません．

fkanehiro, 2015-09-28:
そのメッセージがでていることと合わせて考えると、Choreonoidからのopenhrp-model-loaderの起動ができていないのだと思います。
起動できていないために以前はPDcontrollerが原因でChoreonoidごと落ちていた、修正が入った現在は...Missing?というメッセージがでるということではないかと思います。
まずは起動できているかどうかをご確認いただければと思います。

snozawa, 2015-09-28:
@y-masutani 様

> メッセージが全てダメということではなく，openhrp-model-loaderが起動した後もなぜ「Maybe ModelLoader is missing?」と言い続けるのかということです．ロジックを理解していませんので，見当違いなコメントならば申し訳ありません．

こちらは、こちらの意図してない挙動のようです。
もしかしましたら私の方のコミットミスの可能性がありますので、何点かご確認させていただきたいのですが、
上記のMaybe ModeLoader is missing?メッセージはopenhrp-model-loaderを正常に起動している状況でも
でておりますでしょうか。
また、その場合
```
[PDcontroller0] failed to load model[file:///usr/local/share/OpenHRP-3.1/sample/model/sample1.wrl]
```
がでていないかをご確認いただけますでしょうか。
この出力がでている場合、PDcontrollerがopenhrp-model-loaderからモデル情報を取得できておらず、その場合は`Maybe ModelLoader is missing?`が出力されます。

y-masutani, 2015-09-28:
openhrp-model-loaderを単体で起動した場合に端末に表示されているメッセージが，Choreonoidのメッセージ欄に表示されていますので，openhrp-model-loaderは正常に起動しているのではないかと思います．
現在，試す環境が手元にありませんので，後ほど再度確認します．

y-masutani, 2015-09-28:
確認しました．Choreonoidでは，利用するアイテムを並べるリストがあるのですが，今までopenhrp-model-loaderを起動する外部コマンドアイテムはリストの下の方にありました．それを一番上に持ってくると，[PDcontroller0] failed to load model も Maybe ModelLoader is missing? も出なくなりました．choreonoidのメッセージ欄には loading file: wrlのパス に続いてノードの情報がズラズラ表示されます．

一方，リストの下の方においた場合も再度確認しました．choreonoidのメッセージ欄にはreadyしか表示されていません．openhrp-model-loaderはプロセスとして存在するのは確認しました．choreonoidの起動時に[PDcontroller0] failed to load model と表示され，シミュレーションを実行するとMaybe ModelLoader is missing? が繰り返し表示されます．

実行の順番が重要なようですね．
なお，choreonoidのリポジトリの方でも対応していただいたようですが，そちらはまだ試していません．