fkanehiro/hrpsys-base #815: GraspControllerのconf設定のnameについて

snozawa, 2015-09-17:
@orikuma さん

https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/GraspController/GraspController.cpp#L112
の指定方法が
```
 // <name> : <joint1>, <direction1>, <joint2>, <direction2>, <joint1>, <direction2>, <name> : <joint1>, <direction1>
```
で、例えば
```
RHAND : RARM_JOINT7, -1
```
のようになってますが、このname(RHAND)は必要でしょうか。
RHANDのようなラベルづけがなくてもRARM_JOINT7で一意に指定できるように思いました。
また、ここの記述がへるとdownstreamでもコードで書かなければいけない部分がへるので、
減らせるのであればnameを減らしたいと思いました。

よろしくお願いします。

orikuma, 2015-09-18:
だいぶ前のことなので歴史的経緯が定かではないのですが,
arm_controller的に特定の指セットだけ動かすような需要に対応するための仕様だったような気がします.
@wesleypchan さんが直近のユーザなのでより詳しい情報をご存知かもしれません.

snozawa, 2015-09-18:
> arm_controller的に特定の指セットだけ動かすような需要に対応するための仕様だったような気がします.

なるほど、確かにコードをよくみると、HRP4Rのように２指あるときに、２指両方閉じるといったやりかた使ったり、
１指ずつ動かしたりするためのもののようですね。
ありがとうございました。

closed 2015-09-18 by snozawa