fkanehiro/hrpsys-base #820: setTargetPoseRelative()がうまく動作しない

y-masutani, 2015-09-23:
SampleRobotでsetTargetPoseRelative()がうまく動作せずに困っています．
相対的な変化量を全て0にして，
hcf.setTargetPoseRelative('rarm', 'RARM_WRIST_P', 0,0,0,0,0,0,1)
のように書いても，
setTargetPose failed. Maybe SequencePlayer failed to solve IK.
Currently, returning IK result error
と言われてしまいます．何が悪いのでしょうか？
ご教示いただけると幸いです．

snozawa, 2015-09-23:
hrpsys-simulatorの端末には何か出力されておりますでしょうか。

もし
`joint group is not installed`
のようなワーニングメッセージがでていた場合、
```
    hcf.seq_svc.addJointGroup('rarm', ['RARM_SHOULDER_P', 'RARM_SHOULDER_R', 'RARM_SHOULDER_Y', 'RARM_ELBOW', 'RARM_WRIST_Y', 'RARM_WRIST_P', 'RARM_WRIST_R'])
```
でJointGroupを設定してから同じコマンドをお試しいただけますでしょうか。

y-masutani, 2015-09-24:
回答ありがとうございます．当方では，choreonoidを使っていますが，起動した端末に，
[getJointGroup] group name RARM is not installed
というメッセージが出ていました．そして，
hcf.seq_svc.addJointGroup('rarm', ['RARM_SHOULDER_P', 'RARM_SHOULDER_R', 'RARM_SHOULDER_Y', 'RARM_ELBOW', 'RARM_WRIST_Y', 'RARM_WRIST_P', 'RARM_WRIST_R'])
を追加すると，setTargetPoseRelative()が使えるようになりました．ありがとうございます！
追加で質問なのですが，
・このjoint groupの設定は，confファイルの「end_effectors:」の設定とは別物ですか？
・setTargetPoseRelative()の後に，setJointAngles()を使うと，rarmの関節が動かなくなりました．これはそのような仕様でしょうか？setJointAngles()をrarmにも効くようにするにはどのようにすればいいですか？
・setTargetPose()の場合は，gname='rarm'を与えるとして，eenameを与える場所がありません．また，setTargetPoseRelative()の場合は，frame_nameを与える場所がありません．引数の仕様の違いに混乱しています．
・以上のようなことを解説したドキュメントはありますか？現在は，hrpsys_config.pyを見ながら手探りでやっています．
ご教示いただけると幸いです．よろしくお願いいたします．

snozawa, 2015-09-24:
動作したようでよかったです。
取り急ぎ答えやすいものから失礼いたします。

>  ・このjoint groupの設定は，confファイルの「end_effectors:」の設定とは別物ですか？

はい、別物です。

前者joint Groupは、rarmのように関節のリストにリムの名前をラベル付けするためのものです。
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_sequence_player.py#L223
のようにrarmと関節のリストが対になっております。

後者end_effectorsは、エンドエフェクタ（およびリンク系列）を設定するためのものです。
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/SampleRobot.conf.in#L5
のように、rarmなどのエフェクタ名とエンドリンク・根元リンク・エンドリンクからのエフェクタの相対位置姿勢を記述します。

両方の情報は一本化できそうで、かつVRMLに直接記述する方向も検討しておりますが、現状まだ対応しておりません。
https://github.com/fkanehiro/hrpsys-base/issues/232#issuecomment-46523993

> ・setTargetPoseRelative()の後に，setJointAngles()を使うと，rarmの関節が動かなくなりました．これはそのような仕様でしょうか？setJointAngles()をrarmにも効くようにするにはどのようにすればいいですか？

`hcf.removeJointGroup("rarm")`をお試しいただけますでしょうか。
SequencePlayerの補間・逆運動学機能などは、
- 全身版
- JointGroupを指定したリム毎版

との２つに概ね対応しているものが多いです。
通常の使用方法(setJointAngles)では関節角度を動かすことになりますが、
右腕と左腕を独立に動かしたい、全身とは別に頭部を動かしたいといった要望に対応するため、
後者のJointGroupを採用しています。

この場合、`addJointGroup`されており、かつ`setJointAnglesOfGroup("rarm", ...)`のように指定すると、
全身版の関節角度をrarmのリム版関節角度がうわがくようになります。
これ以降は、右腕関節に関してのみ全身版の指令値`setJointAngles`などは無視され、rarmのリムの指令値が反映されます。
`removeJointGroup("rarm")`をすると、上書きモードから何もしないモードへ遷移し、
右腕関節も全身版関節角度が反映されます。

先日お試しいただいた`setTargetPoseRelative`でも、右腕上書きモードになりますので、
- 上書きモードとして使いつづける場合は`setJointAnglesOfGroup("rarm", ...)`
- 全身版として使いたい場合は`removeJointGroup("rarm")`してから全身版指令値コマンド

のどちらかで意図した挙動となります。

> ・以上のようなことを解説したドキュメントはありますか？現在は，hrpsys_config.pyを見ながら手探りでやっています．

申し訳ございませんが、現状まだこちらは充実しきっていない状況ですので、
samples/SampleRobot/samplerobot_sequence_player.pyなど動くサンプルを
ご参照いただけますと幸いです。
もちろんご質問等ございましたら、お手数ですがissueをきっていただければ対応可能です。

現状のドキュメントはトップページからたどれる
http://fkanehiro.github.io/hrpsys-base/
の
fkanehiro.github.io/hrpsys-base/dd/d91/SequencePlayer.html
のような各RTC毎の説明、
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/SampleRobot
のようなサンプル説明等がございます。

snozawa, 2015-09-25:
> ・setTargetPose()の場合は，gname='rarm'を与えるとして，eenameを与える場所がありません．また，setTargetPoseRelative()の場合は，frame_nameを与える場所がありません．引数の仕様の違いに混乱しています．

setTargetPoseのframe_nameは、どこ座標からみたtargetPose(位置姿勢)に向かって
逆運動学を求めるかを示しています。
別リポジトリではございますが
https://github.com/start-jsk/rtmros_hironx/blob/hydro-devel/hironx_ros_bridge/test/test_hironx_ik.py
が使い方のサンプルで、
```
 ret = self.robot.setTargetPose("larm", [0.3255627368715471, 0.1823638733778268, 0.07462449717662004+0.2], [-3.0732189053889805, -1.5690225912054285, 3.0730289207320203], 5, "CHEST_JOINT0")
```
のようにした場合、CHEST_JOINT0座標からみた値を指定しています。

setTargetPoseRelativのeenameに関しては、不要なのではと今ふと思いました。

@130sさん
実は我々の方ではsetTargetPose関係は使っておりませんので、何かフォローしていただけますと助かります。

130s, 2015-09-25:
https://github.com/start-jsk/rtmros_hironx/issues/61 (とりわけ https://github.com/start-jsk/rtmros_hironx/issues/61#issuecomment-37535993) にあるように，`eename` は EEF のことではなく，参照するフレーム名のはずです．従って，

> ・setTargetPose()の場合は，gname='rarm'を与えるとして，eenameを与える場所がありません．
> また，setTargetPoseRelative()の場合は，frame_nameを与える場所がありません．引数の仕様の違いに混乱しています．

Relative の場合は eename が frame_name に相当する，というのが回答かなと思います．

eename というのが直感的でないので変えたいところですが API break になるので，#823 はドキュメント追加に留めています．

y-masutani, 2015-09-27:
右腕が動かなくなる問題は，hcf.seq_svc.removeJointGroup("rarm") で解決しました．
ありがとうございました．