fkanehiro/hrpsys-base #822: ThermoEstimatorあたりで、温度にしきい値をもうけてサーボOFFかサーボゲインを落とす

snozawa, 2015-09-25:
@orikumaさん
ThermoEstimatorあたりで、温度にしきい値をもうけてサーボOFFかサーボゲインを落とす機能が
あると良いと思いました。
JAXONなどのフェイルセーフをHRP２などで行うためのものと思ってもらうと良いかと思います。

k-okada, 2015-09-27:
それがトルクコントローラなんだったんじゃなかったっけ？
電流の積分から温度を推定して，
温度がリミットをこないようにトルクのリミットを作っている，
という話だった記憶が．．．

orikuma, 2015-09-30:
返信が遅くなりまして申し訳ありません.

ゲインを下げる機能はThermoLimiterが上限温度を超えないトルク上限を計算し, TorqueControllerがトルクを満たすよう関節を動かす, ということになっていますが, 
- ThermoLimiter, ToruqeControllerのパラメータ調整が難しい
- TorqueControllerがImpedanceControllerやStabilizerと干渉する
という問題があってHRP2などで正しく動かせていない, というのが現状です.
前者は真面目にやればいいのですが, 後者の解決はTorqueControllerの抜本的な変更が必要です.
変更はするとして時間がかかりそうなので, 急ぐのであれば温度上限に対してサーボを落とす機能を付けるようにします.

YoheiKakiuchi, 2015-09-30:
> TorqueControllerがImpedanceControllerやStabilizerと干渉する

これは具体的にどんな話だっけ？

snozawa, 2015-09-30:
> という問題があってHRP2などで正しく動かせていない, というのが現状です.

そもそもTorqueControllerが今動いているロボットはいますか？

orikuma, 2015-09-30:
TorqueControllerはそれが関節角度を書き換えた後でImpedanceControllerなどの他のコンポーネントが角度を更に書き換えることを想定できていません.
また例えばImpedanceControllerの手先位置, Stabilzerの重心位置等を保存することができません.

TorqueControllerはHRP4RとSTAROでは動いていましたが, 今実用されているロボットには適用できていないです.

YoheiKakiuchi, 2015-09-30:
> TorqueControllerはそれが関節角度を書き換えた後でImpedanceControllerなどの他のコンポーネントが角度を更に書き換えることを想定できていません.
また例えばImpedanceControllerの手先位置, Stabilzerの重心位置等を保存することができません.

代替手段が、温度上昇でサーボを落とす（ImpedanceControllerの手先位置, Stabilzerの重心位置等を保存できないだろう）なら、使わない理由にはならないんじゃないの？

orikuma, 2015-09-30:
前者の重複した角度書き換えに関する問題があるので, 現状で併用するとTorqueControllerがうまく動かないです.

YoheiKakiuchi, 2015-09-30:
具体的に何が問題か分からない。
torquecontrollerはstやicの先にあるの後にあるの？

orikuma, 2015-09-30:
HRP4Rではic->st->tcとなっていましたが, 同時には使っていませんでした.
具体的な問題はトルク制御系で使っている関節モデル内においてomega (正確にはq, dq, ddq)とtauの間の関係式が静的なことです. 他のコンポーネントによって関節角度を書き換えられることがこの関係式の中で想定されていないため, おそらくうまく動かないと思っています.

YoheiKakiuchi, 2015-09-30:
> 具体的な問題はトルク制御系で使っている関節モデル内においてomega (正確にはq, dq, ddq)とtauの間の関係式が静的なことです. 他のコンポーネントによって関節角度を書き換えられることがこの関係式の中で想定されていないため, おそらくうまく動かないと思っています.

stなどが書き換えることと、上位からの指示で目標角度が変わることに差がないように思うけど違うのかな？

orikuma, 2015-09-30:
元論文とコードを再確認していましたが, tcをstなど関節角度を書き換えるコンポーネントの最後に置いてtcが受け取るqRefが実機のqRefと一致するようにすれば動くような気がしてきました.
(この条件下では関節モデルに与えるomegaとtauの関係が多分変わらない)