fkanehiro/hrpsys-base #825: 別のPCのchoreonoidでシミュレーションをする方法

mitani-syun, 2015-09-25:
JVRCの競技会と同じような環境を用意するために，二つのPCを用意しました．
PC1はchoreonoidを起動しシミュレーションを開始しました．
PC２は端末上で./walk.py (SampleRobot_walk.pyをHRP2DRC用にしたものです)　としたのですが，以下のようなところで実行が止まってしまいます．
どのようにすれば別のPCのchoreonoidでシミュレーションをできるのでしょうか？

PC1の端末
player@odens-W65-W670SH:~/HRP2DRC/choreonoid$ choreonoid HRP2DRC.cnoid
Loading body customizer "/usr/lib/choreonoid-1.5/customizer/CabinetBoxCustomizer.so" for CABINETBOX
Loading body customizer "/usr/lib/choreonoid-1.5/customizer/HoseCustomizer.so" for hose
PDcontroller0: onInitialize() 
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
PDcontroller0: on Activated 
[PDcontroller0] Gain file [/jvrc/HRP2DRC/choreonoid/HRP2DRC.PDgain.dat] opened
[PDcontroller0] No pdcontrol_tlimit_ratio found.
[PDcontroller0] All tlimit_ratio are set to 1.0.
omniORB: Failed to bind to address :: port 2810. Address in use?
omniORB: Error: Unable to create an endpoint of this description: giop:tcp::2810
omniORB: Failed to bind to address :: port 2810. Address in use?
omniORB: Error: Unable to create an endpoint of this description: giop:tcp::2810

PC2の端末
hao123@hao123:~/HRP2DRC/choreonoid$ ./walk.py
configuration ORB with odens-W65-W670SH:2809
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for HRP2DRC : <hrpsys.rtm.RTcomponent instance at 0x7f9f3188b908> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f9f3188b908> isActive? = True 
[hrpsys.py] simulation_mode : True
[hrpsys.py] creating components
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")

y-masutani, 2015-09-26:
JVRCでmitani-syuと同じチームの升谷です．別のissueでもアドバイスをいただきありがとうございます．そちらにまだ応えておらず申し訳ありませんが，JVRCの競技用の環境の準備に手間取っており，こちらを優先しています．

JVRCでは，シミュレーション用のPCと操縦用のPCが分かれています．私達のチームでは，シミュレーション用のPCでchoreonoidとPDcontorllerを動かし，残りのコンポーネントを操縦用PCで動かすつもりです．このようなやり方で大丈夫でしょうか？

そのようにする場合，操縦用PCのコンポーネントを登録するネームサーバ（rtm.nshost）は，シミュレーション用PCでしょうか？上記のmitani-syuの質問ではそれを試そうとしています．rtm.nshost='odens-W65-W670SH'と設定しました．

あるいは，操縦用PCのコンポーネントはlocalhostのネームサーバに登録し，シミュレーション用PCのネームサーバに登録されているChoreonoidのBodyRTCやPDcontorollerに接続するのでしょうか？その場合は，どのように接続相手を指定すればいいでしょうか？waitForRobotHardware("シミュレーション用PC名/HRP2DRC")というようなことはできますか？少し試したところNGでした．書式が違いますか？

RTミドルウェアでは，異なるPC間のコンポーネントを接続できるというのは理屈ではわかっていますし，RT System Editorではやったことがありますが，hrpsys-baseを使う場合にどのようにすればいいかわかりません．

JVRCに依存した質問で恐縮ですが，アドバイスやヒントをいただけると幸いです．
よろしくお願いいたします．

y-masutani, 2015-09-26:
お騒がせしております．再びy-masutaniです．

hrpsys_config.pyを少しだけ書き換え，ネームサーバ（nshost）はChoreonoidのリモートPCを指定し，RTCマネージャ（managerhost）はローカルのPCを指定しました，ローカルのPCでは，Choreonoidの代わりにｒｔｃｄコマンドを実行しRTCマネージャにしました．これによって，リモートのChoreonoid+PDControllerとローカルのｈｒｐｓｙｓ-baseのRTC群を組み合わせてシミュレーションができるようになりました．

しかしながら，実行速度が極端に遅く，JVRCの競技に使えそうにありません．Choreonoidでは周期1msでシミュレーションを実行していますが，ほぼ実時間で動いていたところが，hrpsys-baseのRTCが接続されると，実時間の13倍に低下してしまいます．2台のPCでRTC間の通信を1ms周期で行うのは無理ですか？あるいは，何か設定を間違っているでしょうか？

アドバイスやヒントをいただけると幸いです．
よろしくお願いいたします．

fkanehiro, 2015-09-26:
hrpsys_configのことはよくわからないのですが、ネットワーク通信を挟んで1msで回すのは無理だと思います。

産総研では、実際のロボットではロボット上で動く実時間制御を行うRTCに関してはシミュレータが走っているPC上で、操作用PC上で動くものは操作用PC上で動かしています。

またhrpsys-baseのrtm.pyをポート接続に使われる場合に注意が必要なのは、接続タイプのデフォルトがflush（同期型通信）になっていることです。シミュレータ用PCから操作用PCへの接続があり、その接続のタイプがflushになっていると通信が終わらないと処理が進まなくなってしまうので、newなどの非同期型通信のタイプに変える必要があります。

y-masutani, 2015-09-27:
返信ありがとうございました．
当方では，RoboCupの3Dシミュレーションリーグに参加しているため，そのシステム構成に囚われていました．
リモート側（Choreonoidを実行しているPC）にhrpsys-baseをインストールし，ローカル側のPythonのスクリプトからそれらを起動させることにします．