fkanehiro/hrpsys-base #840: GoVelocity()でうまく歩けない

y-masutani, 2015-10-02:
GoPos()では問題ないのですが，GoVelocity()でうまく歩けない場合は，何をチェックすればいいでしょうか？
SampleRobotではどちらもうまく行っていたのですが，HRP2DRCで難航しています．
曖昧な質問で申し訳ありませんが，何かヒントをいただけないでしょうか．
よろしくお願いいたします．

snozawa, 2015-10-06:
こちら遅くなりました。
おそらくchoreonoidでかつPDcontrollerと併用してシミュレーションされていると思いますが、
Stabilzierはお使いでしょうか。

y-masutani, 2015-10-06:
はい．Stabilzierはもちろん使っております．

snozawa, 2015-10-06:
> GoVelocity()でうまく歩けない場合は，

これはどのような挙動となりますでしょうか。

y-masutani, 2015-10-06:
前進ではそれほど転倒しません．横歩きやその場旋回では，体の横方向の振れ幅が大きすぎるような感じで，横方向の振幅が徐々に大きくなり転倒します．なお，アルゴリズムはTPCCを使っています．EEFMQPは，SampleRobotの時点でうまくいっておらず，深追いせぬままです．

snozawa, 2015-10-06:
(A)STの設定について

> EEFMQPは，SampleRobotの時点でうまくいっておらず，深追いせぬままです．

こちらは、足部に粘弾性要素がないと力の値がギザギザし、発振してしまうそうです。

githubのhrpsys-baseのトップページからたどれます
https://github.com/fkanehiro/hrpsys-base/tree/master/sample/SampleRobot
にStabilizerのサンプルがございます。
これは少し前からございましたが、ちょっと前に起動方法がかわりまして
```
rtmlaunch hrpsys samplerobot.launch TORQUE_CONTROL:=true
```
でご使用いただくと足首ゴムブッシュをシミュレートする機能がはいり、EEFMQPもしようできるようになります。

(B)autoBalanerの設定について
またそれとは別に、
```
        abcp=hcf.abc_svc.getAutoBalancerParam()[1]
        abcp.default_zmp_offsets = [[0.0, -0.01, 0], [0.0, 0.01, 0], [0, 0, 0], [0, 0, 0]]
        hcf.abc_svc.setAutoBalancerParam(abcp)
```
をお試しいただくと効果があるかもしれません。
このパラメータは、各手足の目標COP位置のエンドエフェクタ位置からのオフセット量です。
http://fkanehiro.github.io/hrpsys-base/d5/d3a/structOpenHRP_1_1AutoBalancerService_1_1AutoBalancerParam.html#ad8705383503c3dce0d753f753f647a14

上記では、samplerobotの場合左足、右足、左手、右手の順ですので、
左足をーY方向に0.01[m]ずらして右足を＋Y方向に0.01[m]ずらすようにする（手は使ってないので０）
というパラメータになっております。
これにより、右足左足ともにどちらの支持脚相でも内股側によることになるので、
横揺れが改善させることができるかもしれません。

(A),(B)とございますが、(A)はchoreonoid用に設定をかかないといけない部分がございますので、
(B)のほうをまずお試しいただくと効果があるかもしれません。

y-masutani, 2015-10-06:
丁寧な説明ありがとうございます．
そもそも，ZMPの指定が適切にできているか確信がありません．モデルのwrlファイルからだけでは，足裏の中心位置（それが，上述のエンドエフェクタ一でしょうか？）が決まらないですよね？さらに基本的な質問で申し訳ありません．
それから，GoPos()では横歩きもその場旋回もうまくいっているのですが，GoVelocity()では転倒してしまうという現象はどのように解釈すればいいのでしょうか？GoPos()の歩幅の計画がよくできているということでしょうか？

snozawa, 2015-10-06:
> そもそも，ZMPの指定が適切にできているか確信がありません．モデルのwrlファイルからだけでは，足裏の中心位置（それが，上述のエンドエフェクタ一でしょうか？）が決まらないですよね？さらに基本的な質問で申し訳ありません．

エフェクタはopenhrp3およびchoreonoidでは公式な設定方法がないようですので、
今はconfファイルにかいてAutoBalancer, Stabilizer等でしようしております。
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/SampleRobot.conf.in#L5
たとえば
`lleg,LLEG_ANKLE_R,WAIST,0.0,0.0,-0.07,0.0,0.0,0.0,0.0,`
の部分は、
- `lleg`：エフェクタ名
- `LLEG_ANKLE_R`:エンドリンク名
- `WAIST`:（逆運動学で使う場合）リンク系列のルート
- `0.0,0.0,-0.07`:エンドリンク相対のエフェクタまでの位置オフセット（この場合、足首下に0.07[m]）
- `0.0,0.0,0.0,0.0,``:エンドリンク相対のエフェクタ姿勢オフセット（この場合単位行列）

です。
これがconfに設定されている場合
```
[st] End Effector rleg
....
```
のようにprintメッセージがでます。

> それから，GoPos()では横歩きもその場旋回もうまくいっているのですが，GoVelocity()では転倒してしまうという現象はどのように解釈すればいいのでしょうか？GoPos()の歩幅の計画がよくできているということでしょうか？

こちらの実機およびシミュエータでは倒れることはなかったのですが、
goVelocityではオンラインで速度をかえられるようにするため
歩行中の１歩ないし２歩先の情報をかきかえています。
したがって速度の与え方によっては実ZMPが目標ZMPにAutoBalancerから出六されるパターんお時点で
追従しないことはかんがえられます。

goPosが動いている場合、goPosをご使用されることをお勧めいたします。
goPos歩行中は基本的には現状歩行パターンがかわりませんが、
```
hcf.emergencyStop()
```
で、（すべての状況で安定に停止はできませんが）goPos歩行中に１歩ないし２歩程度で停止できます。

y-masutani, 2015-10-06:
再びありがとうございます．
実は，end_effectorsの設定がよくわからずにいました．SampleRobotのエンドリンクの座標系原点は，足関節のピッチとロールの軸の交わる点ですよね？そこからz軸方向にしかオフセットを与えていないということは，足裏の中心でない点に設定しているということですか？それから，姿勢のオフセットの四つの数値は四元数ですか？しかし，|q|=1でないので違いますか…

snozawa, 2015-10-06:
> 実は，end_effectorsの設定がよくわからずにいました．SampleRobotのエンドリンクの座標系原点は，足関節のピッチとロールの軸の交わる点ですよね？そこからz軸方向にしかオフセットを与えていないということは，足裏の中心でない点に設定しているということですか？

はい、足裏中心でなくZ方向のみ足裏平面高さになるようにオフセットをあたえています。

角度指定はVRMLやプロジェクトファイルでのrotation指定方法同様、回転軸と回転角度にしています。
４つの値はx,y,zと回転角度です。
http://fkanehiro.github.io/openhrp3-doc/jp/create_model.html#rotation
(ほんとうは0,0,0,0でなく1,0,0,0とするほうが誤解なかったかもしれません)

y-masutani, 2015-10-06:
足裏の中心にしていない意図は何でしょうか？中心がベストとの思い込みのせいで混乱していました．
VRMLのrotationは四元数ではないのですか！これまた四元数だと思い込んでいました．

snozawa, 2015-10-06:
> 足裏の中心にしていない意図は何でしょうか？中心がベストとの思い込みのせいで混乱していました．

samplerobotの場合は特に理由はなく設定していないという形になっております。
実機の場合は、総合的な挙動をみてデフォルトなオフセットを決めておりますが、その場合は必ずしも足裏中心でない場合がおおいです。

y-masutani, 2015-10-06:
特に理由なくとのこと，了解です．
頑張ってチューニングします．色々と解説してくださり，ありがとうございました．