fkanehiro/hrpsys-base #844: 実機（HRP2）でsetFootStep時に落ちる

snozawa, 2015-10-07:
@eisoku9618 
実機でsetFootStep時に落ちます．
補間器のところでした．
```
[abc] setFootStepsList                                                                                                                                                                                                                                         
[abc] print footsteps                                                                                                                                                                                                                                          
[abc]  Set normal footsteps                                                                                                                                                                                                                                    
[0] footstep                                                                                                                                                                                                                                                   
  name = [rleg]                                                                                                                                                                                                                                                
  pos = [1.35701,  -0.188662,  0.177294]                                                                                                                                                                                                                       
  rot = [           1, -2.44956e-05,  1.53008e-20] [ 2.44956e-05,            1, -1.22925e-19] [-1.52978e-20,  1.22926e-19,            1]                                                                                                                       
  step_height = 0[m], step_time = 1.75[s], toe_angle = 0[deg], heel_angle = 0[deg]                                                                                                                                                                             
[1] footstep                                                                                                                                                                                                                                                   
  name = [lleg]                                                                                                                                                                                                                                                
  pos = [1.62701,  0.0213455,  0.327294]                                                                                                                                                                                                                       
  rot = [           1, -2.45736e-05, -4.51512e-12] [ 2.45736e-05,            1,  1.90488e-07] [-1.65852e-13, -1.90488e-07,            1]                                                                                                                       
  step_height = 0.05[m], step_time = 1.75[s], toe_angle = 30[deg], heel_angle = 5[deg]                                                                                                                                                                         
[2] footstep                                                                                                                                                                                                                                                   
  name = [rleg]                                                                                                                                                                                                                                                
  pos = [1.62701,  -0.188655,  0.327294]                                                                                                                                                                                                                       
  rot = [           1, -2.44956e-05,  1.53008e-20] [ 2.44956e-05,            1, -1.22925e-19] [-1.52978e-20,  1.22926e-19,            1]                                                                                                                       
  step_height = 0.05[m], step_time = 1.75[s], toe_angle = 30[deg], heel_angle = 5[deg]                                                                                                                                                                         
[3] footstep                                                                                                                                                                                                                                                   
  name = [lleg]                                                                                                                                                                                                                                                
  pos = [1.75201,  0.0213486,  0.327294]                                                                                                                                                                                                                       
  rot = [           1, -2.45736e-05, -4.51512e-12] [ 2.45736e-05,            1,  1.90488e-07] [-1.65852e-13, -1.90488e-07,            1]                                                                                                                       
  step_height = 0.05[m], step_time = 1.75[s], toe_angle = 30[deg], heel_angle = 5[deg]                                                                                                                                                                         
[4] footstep                                                                                                                                                                                                                                                   
  name = [rleg]                                                                                                                                                                                                                                                
  pos = [1.75201,  -0.188652,  0.327294]                                                                                                                                                                                                                       
  rot = [           1, -2.44956e-05,  1.53008e-20] [ 2.44956e-05,            1, -1.22925e-19] [-1.52978e-20,  1.22926e-19,            1]                                                                                                                       
  step_height = 0.05[m], step_time = 1.75[s], toe_angle = 30[deg], heel_angle = 5[deg]                                                                                                                                                                         
[5] footstep                                                                                                                                                                                                                                                   
  name = [lleg]                                                                                                                                                                                                                                                
  pos = [2.02201,  0.0213553,  0.477294]                                                                                                                                                                                                                       
  rot = [           1, -2.45736e-05, -4.51512e-12] [ 2.45736e-05,            1,  1.90488e-07] [-1.65852e-13, -1.90488e-07,            1]                                                                                                                       
  step_height = 0.05[m], step_time = 1.75[s], toe_angle = 30[deg], heel_angle = 5[deg]                                                                                                                                                                         
[... cut ...]                                  
RobotHardware0(0.19), seq(0.02), sh(0.16), fk(0.04), kf(0.05), rmfo(0.10), es(0.06), ic(0.04), abc(1.86), st(0.79), co(0.35), te(0.35), tl(0.04), hes(0.03), el(0.06), log(0.12), hc(0.04),                                                                    
interpolator::get dvs = NULL, dq.size() = 0, length = 1                                                                                                                                                       
```

eisoku9618, 2015-10-11:
実機でしか再現しないケースでしょうか？

``roslaunch hrpsys samplerobot.launch``して，
``ipython -i ~/ros/hydro_parent/src/hrpsys/sample/SampleRobot/samplerobot_data_logger.py``に以下を貼り付けても再現しませんでした．

```python
hcf.abc_svc.startAutoBalancer(['rleg','lleg'])
ggp = hcf.abc_svc.getGaitGeneratorParam()[1]
ggp.default_orbit_type = OpenHRP.AutoBalancerService.STAIR
hcf.abc_svc.setGaitGeneratorParam(ggp)
hcf.setFootStepsWithParam([OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([1.35701,  -0.188662,  0.177294], [1,0,0,0], "rleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([1.62701,  0.0213455,  0.327294], [1,0,0,0], "lleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([1.62701,  -0.188655,  0.327294], [1,0,0,0], "rleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([1.75201,  0.0213486,  0.327294], [1,0,0,0], "lleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([1.75201,  -0.188652,  0.327294], [1,0,0,0], "rleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([2.02201,  0.0213553,  0.477294], [1,0,0,0], "lleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([2.02201,  -0.188646,  0.477294], [1,0,0,0], "rleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([2.14701,  0.0213584,  0.477294], [1,0,0,0], "lleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([2.14701,  -0.188643,  0.477294], [1,0,0,0], "rleg")]),
                           OpenHRP.AutoBalancerService.Footsteps([OpenHRP.AutoBalancerService.Footstep([2.14701,  0.0213584,  0.477294], [1,0,0,0], "lleg")])],
                          [OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.00, step_time=1.75, toe_angle=0.00, heel_angle=0.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.05, step_time=1.75, toe_angle=30.0, heel_angle=5.0)]),
                           OpenHRP.AutoBalancerService.StepParams([OpenHRP.AutoBalancerService.StepParam(step_height=0.00, step_time=1.75, toe_angle=0.00, heel_angle=0.0)])])
```

snozawa, 2015-10-11:
今のところ実機でしかみたことがなく，
また実機でも時々しか再現してないようにみえます．
https://github.com/fkanehiro/hrpsys-base/pull/848
でどのinterpolatorが落ちてるかがわかるようにしています．

eisoku9618, 2015-10-11:
わかりました．ありがとうございます．

snozawa, 2016-08-16:
https://github.com/fkanehiro/hrpsys-base/pull/973
以降再現してないので治ったかもしれません。
また問題があれば再度reopenします。ありがとうございました。

closed 2016-08-16 by snozawa