fkanehiro/hrpsys-base #891: Autobalancerで時間がかかることがある

mmurooka, 2015-11-26:
JAXONでの物体押し実験の結果です．

```
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
[abc] Too large IK error (vel_p) = [0.00516131 -0.0217202 -0.0300846][m], count = 300
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error [st] getParameterRARM_JOINT7

...

[1448538862.411800] Timeover: processing time = 829.65[ms]
RobotHardware0(0.17), seq(0.01), sh(0.06), fk(0.01), kf(0.02), rmfo(0.04), es(0.03), ic(0.10), abc(827.98\
), st(0.42), co(0.62), hes(0.04), log(0.16),
Lower joint limit error RARM_JOINT7
[1448538862.414424] Timeover: processing time = 2.53[ms]
RobotHardware0(0.22), seq(0.03), sh(0.22), fk(0.05), kf(0.05), rmfo(0.12), es(0.10), ic(0.27), abc(0.47),\
 st(0.35), co(0.51), hes(0.05), log(0.10),
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7
Lower joint limit error RARM_JOINT7

Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
[1448538863.877006] Timeover: processing time = 92.85[ms]
RobotHardware0(0.22), seq(0.02), sh(0.07), fk(0.02), kf(0.03), rmfo(0.05), es(0.04), ic(0.11), abc(91.76)\
, st(0.14), co(0.32), hes(0.02), log(0.06),
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7

...

Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
[abc] Too large IK error (vel_r) = [-0.00494771 -0.00451907 -0.00166506][rad], count = 500
Lower joint limit error LARM_JOINT7
Lower joint limit error LARM_JOINT7
[st] getParameter
[1448538871.293719] Timeover: processing time = 2311.56[ms]
RobotHardware0(0.22), seq(0.01), sh(0.09), fk(0.03), kf(0.04), rmfo(0.06), es(0.06), ic(0.13), abc(2310.2\
0), st(0.16), co(0.44), hes(0.03), log(0.09),
Lower joint limit error LARM_JOINT7
```

garaemon, 2015-11-27:
`startWalking`のmutexが怪しい?

snozawa, 2015-11-27:
たぶんこのあたりは歩いている最中で、startWalkingがよばれたしばらく後になると思うので、
今回の件とは関係ないと思ってます。
startWalkingは歩き始めに１かいよばれるだけ

snozawa, 2015-11-27:
https://github.com/jsk-ros-pkg/jsk_control/pull/523のgetParameterなのかなあと思ってます。

garaemon, 2015-11-27:
https://github.com/jsk-ros-pkg/jsk_control/pull/523 のあとでも8msecぐらいかかる時があるのは昨日の晩に見ました