fkanehiro/hrpsys-base #905: readDataPortのスレッドセーフバージョンはありますでしょうか。

snozawa, 2015-12-13:
** 症状
readDataPortをずっと回していると、rtcd側で
```
omniORB: ERROR -- A servant has been deleted that is still activated.
      id: root<285> (deactivating)
omniORB: Assertion failed.  This indicates a bug in the application
using omniORB, or maybe in omniORB itself.
 file: ../../../../../src/lib/omniORB/orbcore/omniServant.cc
 line: 223
 info: activation_found
```
のようなエラーがでてrtcdが落ちます。
gdbをかけると以下のようになります（btしわすれました）
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd9287700 (LWP 10857)]
0x00007ffff6eba15d in omni::omniOrbPOA::lastInvocationHasCompleted(omniLocalIdentity*) () from /usr/lib/libomniORB4.so.1
(gdb)
```

** 確認方法（取り急ぎROSからの方法で失礼します）
ある端末で
```
rtmlaunch hrpsys samplerobot.launch
```
をあげて、別端末で
```
python -i `rospack find hrpsys`/samples/SampleRobot/samplerobot_data_logger.py
while True:        
   a=rtm.readDataPort(hcf.el.port("q")).data
```
でreadDataPortをずっと回すとでます。

** その他
readDataPortはrtm.pyで"Pull"型で定義されてると思いますが、
"Pull"型だとsubscription typeが指定できないので、スレッドセーフにならないという理解であってますでしょうか。
また、readDataPortでやっていることを"Push"型でやる方法などございますでしょうか。
やりたいことは、(できればhrpsys-base/python以下でできる範囲で)pythonからデータポートの値をスレッドセーフに取得する、です。

よろしくお願いします。

fkanehiro, 2015-12-14:
これはPythonからの呼び出し方の問題というよりは、OpenRTM-aistのC++版の問題のような気がします。スタックトレースはどのようになっていますでしょうか。
@n-ando さん、いかがでしょうか。

snozawa, 2015-12-14:
gdbでbtをした結果をお送りします。
```
omniORB: ERROR -- A servant has been deleted that is still activated.
      id: root<2201> (deactivating)

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd9287700 (LWP 5434)]
0x00007ffff6eba15d in omni::omniOrbPOA::lastInvocationHasCompleted(omniLocalIdentity*) () from /usr/lib/libomniORB4.so.1
(gdb) (gdb) bt
#0  0x00007ffff6eba15d in omni::omniOrbPOA::lastInvocationHasCompleted(omniLocalIdentity*) () from /usr/lib/libomniORB4.so.1
#1  0x00007ffff6e9e2ba in omniLocalIdentity::dispatch(omniCallHandle&) ()
   from /usr/lib/libomniORB4.so.1
#2  0x00007ffff6ee1b92 in omni::GIOP_S::handleRequest() ()
   from /usr/lib/libomniORB4.so.1
#3  0x00007ffff6ee29d0 in omni::GIOP_S::dispatcher() ()
   from /usr/lib/libomniORB4.so.1
#4  0x00007ffff6edf3b3 in omni::giopWorker::real_execute() ()
   from /usr/lib/libomniORB4.so.1
#5  0x00007ffff6edfa2f in omni::giopWorker::execute() ()
   from /usr/lib/libomniORB4.so.1
#6  0x00007ffff6e91425 in omniAsyncWorkerInfo::run() ()
   from /usr/lib/libomniORB4.so.1
#7  0x00007ffff6e91b4f in omniAsyncWorker::run(void*) ()
   from /usr/lib/libomniORB4.so.1
#8  0x00007ffff6bcc4d3 in omni_thread_wrapper ()
   from /usr/lib/libomnithread.so.3
#9  0x00007ffff7182e9a in start_thread ()
   from /lib/x86_64-linux-gnu/libpthread.so.0
#10 0x00007ffff576e38d in clone () from /lib/x86_64-linux-gnu/libc.so.6
#11 0x0000000000000000 in ?? ()
(gdb)
```

> これはPythonからの呼び出し方の問題というよりは、OpenRTM-aistのC++版の問題のような気がします。スタックトレースはどのようになっていますでしょうか。

本来、pullだとスレッドセーフになるはずなものでしょうか。

snozawa, 2015-12-17:
最近気づいたこととしましては、この現象はtravis上では最近とくに頻繁に見かけるようになりました。

fkanehiro, 2015-12-17:
pullであろうとなかろうと、外からCORBA IFを読んでいるだけなので、落ちないようにするべき、だと思うのですがどうでしょうか-> @n-ando
さん。
ちょっとこのスタックトレースだとなんとも追えないとは思いますが。

n-ando, 2015-12-17:
金広様

安藤です

すみません、具体的にはどこで落ちているんでしょうか？
pullはおそらく使用している人が少ないので枯れてない可能性が大きいです。
ともかく落ちるのはおかしいので、状況をもう少し詳しく教えていただければ対応いたします。

-- 
--
安藤慶昭＠国立研究開発法人 産業技術総合研究所
ロボットイノベーション研究センター
ロボットソフトウェアプラットフォーム研究チーム長
(兼務)首都大学東京 客員教授,  博士(工学)
〒305-8568 つくば市梅園1-1-1 中央第2
e-mail: n-ando@aist.go.jp, web: http://staff.aist.go.jp/n-ando
OpenRTM-aist: http://www.openrtm.org

Noriaki Ando, Ph.D.
    Team Leader, Robot Software Platform Research Team,
    Robot Innovation Research Center, AIST
    Visiting Prof., Tokyo Metropolitan University
    AIST Tsukuba Central 2, Tsukuba, Ibaraki 305-8568 JAPAN
    e-mail: n-ando@aist.go.jp, web: http://staff.aist.go.jp/n-ando
    OpenRTM-aist: http://www.openrtm.org

snozawa, 2015-12-17:
@n-ando様

hrpsys-baseで定義されているpythonのreadDataPort関数を使っており、
その時data port通信設定は
https://github.com/fkanehiro/hrpsys-base/blob/master/python/rtm.py#L637
のようになっております。

fkanehiro, 2015-12-17:
上に@snozawa さんにスタックトレースをつけてもらっていますが、あまり参考にならないですね。

fkanehiro, 2016-01-12:
僕も今日同様の現象に遭遇しました。
呼び出し側のPythonは以下のようなトレースになっています。
```
---------------------------------------------------------------------------
TRANSIENT                                 Traceback (most recent call last)
<ipython-input-31-25c7c38e11ce> in <module>()
----> 1 rtm.readDataPort(stnd.port("rpyRef"))

/home/kanehiro/openrtp/lib/python2.7/dist-packages/hrpsys/rtm.pyc in readDataPort(port, timeout)
    670                 tm = tm + 0.1
    671 
--> 672     port.disconnect(prof.connector_id)
    673     return None
    674 

/usr/lib/python2.7/dist-packages/RTC_idl.py in disconnect(self, *args)
   1215 
   1216     def disconnect(self, *args):
-> 1217         return _omnipy.invoke(self, "disconnect", _0_RTC.PortService._d_disconnect, args)
   1218 
   1219     def disconnect_all(self, *args):

TRANSIENT: CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
```

n-ando, 2016-01-28:
放置していてすみません。試しに以下のようなコードで、SeqOutに対してreadDataPortをやってみたのですが、簡単には落ちませんでした。どのくらいの頻度で落ちますでしょうか？
ちなみに、このreadDataPortの呼び出しって、100ms以上はかかってますよね？ポートの接続は結構時間がかかるので、このオペレーションのコストはけっこう高いと思いますが、大丈夫でしょうか？

`
#!/usr/bin/env python

import rtm
import RTC
import OpenRTM_aist

fd = open("ior.txt")
ior = fd.read()
fd.close()

rtm.initCORBA()

obj = rtm.orb.string_to_object(ior)
rtc = rtm.RTcomponent(obj)

print rtc
print dir(rtc)
print rtc.getProperty("data_type")
rtc.start()
rtc = obj
port_octet = rtm.findPort(rtc, "Octet")
port_short = rtm.findPort(rtc, "Short")
port_long = rtm.findPort(rtc, "Long")
port_float = rtm.findPort(rtc, "Float")
port_double = rtm.findPort(rtc, "Double")

while (True):
    print rtm.readDataPort(port_octet)
    print rtm.readDataPort(port_short)
    print rtm.readDataPort(port_long)
    print rtm.readDataPort(port_float)
    print rtm.readDataPort(port_double)
`

n-ando, 2016-01-28:
放置していてすみません。試しに以下のようなコードで、SeqOutに対してreadDataPortをやってみたのですが、簡単には落ちませんでした。どのくらいの頻度で落ちますでしょうか？
ちなみに、このreadDataPortの呼び出しって、100ms以上はかかってますよね？ポートの接続は結構時間がかかるので、このオペレーションのコストはけっこう高いと思いますが、大丈夫でしょうか？

```
#!/usr/bin/env python

import rtm
import RTC
import OpenRTM_aist

fd = open("ior.txt")
ior = fd.read()
fd.close()

rtm.initCORBA()

obj = rtm.orb.string_to_object(ior)
rtc = rtm.RTcomponent(obj)

print rtc
print dir(rtc)
print rtc.getProperty("data_type")
rtc.start()
rtc = obj
port_octet = rtm.findPort(rtc, "Octet")
port_short = rtm.findPort(rtc, "Short")
port_long = rtm.findPort(rtc, "Long")
port_float = rtm.findPort(rtc, "Float")
port_double = rtm.findPort(rtc, "Double")

while (True):
    print rtm.readDataPort(port_octet)
    print rtm.readDataPort(port_short)
    print rtm.readDataPort(port_long)
    print rtm.readDataPort(port_float)
    print rtm.readDataPort(port_double)
```

n-ando, 2016-01-29:
現象を再現して、OutPortCorbaCdrProvider のデストラクタ付近でのCORBAオブジェクトの deactivate の前後が怪しいことは突き止めました。経緯は以下の通りです。
http://redmine.openrtm.org/issues/3438

ちなみに、omniORB 4.2 以降では、この現象に関係のありそうな部分の実装が若干変わっているのですが、お使いのomniORB のバージョンはいくつでしょうか？＞ @snozawa @fkanehiro

fkanehiro, 2016-01-29:
4.1.6です(Ubuntu14.04LTS)

n-ando, 2016-01-29:
了解です。＞4.1.6
4.2.0でやってみましたが、状況はあまり変わらず。引き続き調査します。

snozawa, 2016-01-29:
手元で落ちてる方は今すぐには調べられないのですが、
hrpsys-baseのtravis上で同様に落ちています。
https://s3.amazonaws.com/archive.travis-ci.org/jobs/102526586/log.txt

このときはubuntu は
```
Description:	Ubuntu 12.04.5 LTS
```
で、omniORBは4.1.6でした。

fkanehiro, 2016-05-11:
@n-ando さん、相変わらず数十回の一度の頻度で
```
PortBase.cpp:191: virtual RTC::ReturnCode_t RTC::PortBase::connect(RTC::ConnectorProfile&): アサーション `!isExistingConnId(connector_profile.connector_id)' に失敗しました。
```
というメッセージと共にRTCが落ちてしまうのですが、その後進展ないでしょうか。

n-ando, 2016-05-12:
すみません、追いかけていたらCORBAの中の方まで追跡する必要がありそうなため止まっています。
問題は頻繁にCORBAオブジェクトを生成・削除するためアクティブオブジェクトマップに不整合が置いているためのようです。
また、rtm.pyではデータを1回取得するごとにポート接続を生成していますのでオーバーヘッドが大きいので、rtm.py側とOpenRTM側両方の修正も含めてワンショットでデータを取得する方法を考えます。結果は、とりあえずパッチという形で提供するのでよろしいでしょうか？

fkanehiro, 2016-05-13:
@n-ando さん、そもそもデータポートからデータを読むのって今のやり方でいいのでしょうか。
データポートからデータを読むにはコネクタを作る必要があり、ポート側からwriteしてコネクタにデータが入ってこないとデータが読めないと思っているのですが、正しいでしょうか。writeのタイミングに関わらずすぐにデータを取り出すことは可能でしょうか。

n-ando, 2016-05-13:
試しに、もっと効率の良いやり方で実装してみました。データポートのデータをPortProfile.properties['dataport.data_value'] に代入しておいて、rtm.pyのreadDataPortではget_port_profile()でデータを取得する方法です。これだと170倍くらい速くなりました。

500個のデータを取得するためにかかった時間
- 変更前 time:  46.7247030735
- 変更後 time:  0.272583007812

ただし、いくつか問題、というか考慮すべき事項があります。

- データが一度も生成されていない段階で、初期値が取得できてしまう。（これは、実装の都合上そうなっています、変更可能）
- 以前はRTCがActive状態かつwrite()が呼ばれたときしかデータが来なかったが、今回は最後の値がRTCの状態にかかわらず取得できる。（いいか悪いかは別として仕様が変わっています。）
- OutPortが常に値をAnyに変更するオーバーヘッドがある。（オプションでprofileにデータ値をexportするかどうか選択できるようにすることも可能。下に示したrtm.pyの変更だと、exportしていなければ従来の方法で取得するので、値が取れないということはない。）
- この方法で落ちなくなったという保証はない。（がCORBAオブジェクトの生成頻度・速度が全く異なるため、これで落ちるようなら、普段の運用でももっと落ちるはず。）ただいまテスト中。

ご意見ください。

OutPort.h (trunkにはコミット済み)
```
===================================================================
--- OutPort.h   (リビジョン 2718)
+++ OutPort.h   (作業コピー)
@@ -138,6 +138,8 @@
 #endif
         m_value(value), m_onWrite(0), m_onWriteConvert(0)
     {
+      addProperty("dataport.data_value", m_value);
+      m_propValueIndex = NVUtil::find_index(m_profile.properties, "dataport.data_value");
     }

     /*!
@@ -209,6 +211,7 @@
           (*m_onWrite)(value);
           RTC_TRACE(("OnWrite called"));
         }
+      m_profile.properties[m_index].value <<= value;

       bool result(true);
       std::vector<const char *> disconnect_ids;
@@ -490,6 +493,8 @@
     coil::TimeMeasure m_cdrtime;

     DataPortStatusList m_status;
+
+    CORBA::Long m_propValueIndex;
   };
 }; // namespace RTC
```

rtm.py
```
--- rtm.py.org  2016-05-13 10:12:34.892000000 +0900
+++ rtm.py      2016-05-13 10:12:57.004000000 +0900
@@ -635,11 +635,19 @@
 # \return data
 #
 def readDataPort(port, timeout=1.0):
-    pprof = port.get_port_profile()
+    try:
+        pprof = port.get_port_profile()
+    except:
+        return None
+    classname = None
+    datavalue = None
     for prop in pprof.properties:
         if prop.name == "dataport.data_type":
             classname = any.from_any(prop.value)
-            break
+        if prop.name == "dataport.data_value":
+            return any._to_tc_value(prop.value)[1]
     nv1 = SDOPackage.NameValue("dataport.interface_type", any.to_any("corba_cdr"))
     nv2 = SDOPackage.NameValue("dataport.dataflow_type", any.to_any("Pull"))
     nv3 = SDOPackage.NameValue("dataport.subscription_type", any.to_any("flush"))
```

fkanehiro, 2016-05-13:
個人的には簡単にかつすぐにデータが取れるようになってありがたいです。