fkanehiro/hrpsys-base #913: setJointAnglesSequenceFullのclearのテストが通らない

mmurooka, 2015-12-17:
https://github.com/fkanehiro/hrpsys-base/pull/908#issuecomment-165356170 での話のように，
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_sequence_player.py#L352-L367
のテストが，手元では通るのですが，travisのテストでは失敗します．

travisのテスト失敗ログ: https://s3.amazonaws.com/archive.travis-ci.org/jobs/92443624/log.txt

snozawa, 2015-12-17:
手元で通ってます？
こちらの手元では通らなかった気がします。

mmurooka, 2015-12-18:
１． `demoSetJointAnglesSequenceOfGroup()` （ https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_sequence_player.py#L383 ）を呼んだ後だと，
`demoSetJointAnglesSequenceFull()` （ https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_sequence_player.py#L384 ）でclearのテストが通らないこと．

２． `demoSetJointAnglesSequenceFull()`だけを呼ぶとclearのテストが通ること

が分かりました．

仕様通りなのかバグなのかはわかりませんが，
`demoSetJointAnglesSequenceOfGroup()`の最後でjoint groupを削除したところ
１．のケースでも通るようになりましたので，
https://github.com/fkanehiro/hrpsys-base/pull/914 でPRを切りました．
自分のbranchのtravisのテストで，このPRの変更で通ることを確認した（ https://travis-ci.org/mmurooka/hrpsys-base/jobs/97588239 ）ので，PRのtravisでも通るはずです．

mmurooka, 2015-12-18:
https://github.com/fkanehiro/hrpsys-base/pull/908で解決しましたのでcloseします．

closed 2015-12-18 by mmurooka