fkanehiro/hrpsys-base #959: VirtualForceまわりでエンバグしている

iory, 2016-03-01:
```
rtmlaunch hrpsys_ros_bridge_tutorials hrp4r.launch
```
を実行すると以下の様なエラーが出ます。
rtmros_tutorials/hrpsys_ros_bridge_tutorials/model以下のHRP4R.confのvirtual_force_sensorの部分で、
wristをコメントアウトすると落ちなくなります。
https://github.com/fkanehiro/hrpsys-base/pull/467

```
iori@welsh: rtmlaunch hrpsys_ros_bridge_tutorials hrp4r.launch

[rtmlaunch] Start omniNames at port 15005

Tue Mar  1 12:44:29 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000e0000003139322e3136382e39372e3837009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000001d10d556010055b3
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iori/.ros/log/e73c9340-df5f-11e5-8e01-5c514f549692/roslaunch-welsh-21933.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/iori/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.97.87:34477/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path:
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    AutoBalancerServiceROSBridge (hrpsys_ros_bridge/AutoBalancerServiceROSBridgeComp)
    CollisionDetectorServiceROSBridge (hrpsys_ros_bridge/CollisionDetectorServiceROSBridgeComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    EmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
[... 603 lines cut ...]
[ic]   localPos =     [0,  0,  -0.1][m]
[ic]   localR =     [-3.67321e-06,  5.40683e-23,            1]
    [ 2.94393e-17,            1,  5.40683e-23]
    [          -1,  2.94393e-17, -3.67321e-06]
[ic] End Effector [rleg]R_ANKLE_R WAIST
[ic]   target = R_ANKLE_R, base = WAIST
[ic]   localPos =     [0,  0,  -0.091849][m]
[ic]   localR =     [1, 0, 0]
    [0, 1, 0]
    [0, 0, 1]
[ic] End Effector [lleg]L_ANKLE_R WAIST
[ic]   target = L_ANKLE_R, base = WAIST
[ic]   localPos =     [0,  0,  -0.091849][m]
[ic]   localR =     [1, 0, 0]
    [0, 1, 0]
    [0, 0, 1]
[ic] Add impedance params
[ic]   sensor = rfsensor, sensor-link = R_ANKLE_R, ee_name = rleg, ee-link = R_ANKLE_R
[ic]   sensor = lfsensor, sensor-link = L_ANKLE_R, ee_name = lleg, ee-link = L_ANKLE_R

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc5ffb700 (LWP 22792)]
0x00007fffac2850f5 in ImpedanceController::onInitialize() ()
   from /home/iori/catkin_ws/hrp4/devel/share/hrpsys/lib/ImpedanceController.so
```

k-okada, 2016-03-01:
@snozawa もう少しちゃんと言うと#467
でvirual_force_sensorのリンク名かセンサ名が既に実機にないと落ちるように仕様変更されていませんか？ということです．
テストで試しているSampleRobotはモデル上では既に四肢に力センサがあって，それにVirtualセンサをつけている．
一方HRP4Rは手にはモデル上では力センサが無いけど，Virutalセンサをつけている，という違いな気がします．

@iory がもう少し詳しく解説してくれるはずです．

snozawa, 2016-03-01:
> @snozawa もう少しちゃんと言うと#467 でvirual_force_sensorのリンク名かセンサ名が既に実機にないと落ちるように仕様変更されていませんか？ということです．

HRP4Rの落ちの原因自体は、
正規のVRMLのForceSensorのPROTOを使って定義されてる
force sensorでないと動かないコードになってしまってて、
virtual_force_sensorだと落ちるじゃないかと思っています。

コードを修正するときにvirtual_force_sensorまで気が配れてなかった心あたりが各所にあります

今hrpsys-baseでvirtual_force_sensorがさしてそうなものは２つあり、
- VirtualForceSensor RTC
- Confにかいてあるvirtual_force_sensor設定

があります。

> テストで試しているSampleRobotはモデル上では既に四肢に力センサがあって，それにVirtualセンサをつけている

SampleRobotは前者のVirtualForceSensor RTCのテストはしてますが、
Confにかいてあるvirtual_force_sensor設定とImpedanceControllerの組み合わせはチェックしていません。
もう少し掘り下げますと、後者のconfの設定自体は書くだけですが、
実際の各RTCの挙動の実装としては各RTC内でvirtual_force_sensorに対応するコードを書かなければいけなくなってますので、
例えばHRP２とそこそこ同じことをHRP4Rでやろうとすると、
IMpedanceControllerの他にAutoBalancer, Stabilzier等も修正がいりそうです。

また、多分このvirtual_force_sensor+impedanceControllerの組み合わせが動いていたころからの
「仕様変更」としてあげられるのは（変更したのはだいぶ前ですが）、
- 旧：Impedanceの力センサエンドエフェクタ設定をidl越しに指定、setImpedaneControlParamでimpedanceのエフェクタ情報を指定
- 新：Impedanceの力センサエンドエフェクタ設定をidl越しに指定しない、confのエンドエフェクタとForceSensor情報を使う

です(関連：https://github.com/fkanehiro/hrpsys-base/issues/232)
その時に「ForceSensor情報を使う」部分でvirtual_force_sensorに対応できてない気がします。
また、エンドエフェクタとforceSensorの対応をとってる箇所で
「いっこのlimbにはImpedanceControllerとかで使いたいようなForceSensor・virtual_force_sensorは一個しかない」
ことを仮定してます。
ForceSensorがあるlimbではviratul_force_sensorを使うことはこれまでなかったためです。

なので、virtual_force_sensor+impedanceContorllerの組み合わせをテストするとしたら、
力センサのないロボットを使ってテストを書くと良いです。
ぱっとみでは、PA10がそれっぽいです。

snozawa, 2016-03-01:
@iory に試してもらうとよさそうなこととしては、

- ImpedanceControllerは、正規のForceSensorは動いていてvirtual_force_sensorが動かない。
なので、ImpedanceContorller.[cpp,h]中のForceSensorとかいてありそうな部分をさらってみて、
具体的にどこの箇所で落ちていそうか調べる
- 上記の箇所（何箇所かあるはず）をvirtual_force_sensorで動くように対応するPRをつくってみる。HRP４Rで動作確認
- HRP4Rを使わないtravisでテストできるテストコードのPRをつくってみる。例えばPA10とかを使って、virtual_force_sensorの設定をconfにかいてみて、virtual_force_sensorのテストプログラム、virtual_force_sensor+ImpedanceControllerの組み合わせのテストプログラムをかいてみる（ホントはPA１０でなくて手に力センサのないHRP４Rライクな脚腕ヒューマノイドがいいですが、、、）

でしょうか。

なお一個前のcommentのとおり、根本原因としてはirtual_force_sensorに対応するコードを
各RTCに書かないといけなくなっています。
具体的には
```
if (ForceSensor) {
 ...
} else { // VirtualForceSensor
 ...
}
```
を全コードのForceSensorを使ってることろに書かないと
virtual_force_sensor動かなくなってます。

これの他の方法としては、virtual_force_sensorもVRMLのForceSensorとして記述
利用できるようにしておけば、上記のif文がなくなって
RTC側のバグもなく、コード記述もフツウなかんじになります。
（issueがどこかにありましたが）ただ、これはローレベルに
virtual_force_sensor対応のコードがいきそうで、
より触れる人が少ないんじゃないかということで採用されてきてません。
他には、ForceSensorにアクセスするコードを
別ForceSensor+virtual_Force_sensor対応したコードでwrapするとかでしょうか（どうやるんでしょう）

k-okada, 2016-03-02:
https://github.com/fkanehiro/hrpsys-base/pull/467/files#discussion_r54674992

で落ちているらしいです by @iory

snozawa, 2016-03-02:
了解です。
https://github.com/fkanehiro/hrpsys-base/issues/959#issuecomment-190729279
の一個目が判明してきたようですので、その次のForceSensorに対応しているコードをvirtual_force_sensorに対応する変更をいれてみて落ちないか確認していただけると助かります。

方法は、
https://github.com/fkanehiro/hrpsys-base/issues/959#issuecomment-190729279
の後半部分の文章のようにいい感じの方法が今のところなさそうですが、
現状（と元々のコード）のやり方を踏まえて
```
if (ForceSensor) {
 ...
} else { // VirtualForceSensor
 ...
}
```
のようなif文で対応してみることになると思います。

iory, 2016-03-02:
>これの他の方法としては、virtual_force_sensorもVRMLのForceSensorとして記述
>利用できるようにしておけば、上記のif文がなくなって
この方法を探していましたが、難しいのですか。

もっと良いやり方（書き方）がありそうですが、以下のような変更で立ち上がるようになります。
https://github.com/iory/hrpsys-base/commit/0f27893695b47a8af892d89f05a24fe6e424c89c

これで立ち上がるのですが、以下のエラーが出ています。これは、
どこで対処すればいいでしょうか？

以下抜粋です。
```
[rtm.py]    Connect abc.limbCOPOffset_lhsensor - st.limbCOPOffset_lhsensor
[rtm.py]    Failed to connect abc.vlhsensor to None([None])
[rtm.py]    Connect sh.vlhsensorOut - es.vlhsensorIn
[rtm.py]    Connect es.vlhsensorOut - ic.ref_vlhsensorIn
[rtm.py]    Connect es.vlhsensorOut - abc.ref_vlhsensor
[rtm.py]    Failed to connect abc.limbCOPOffset_vlhsensor to None([None])
[rtm.py]    Failed to connect abc.vrhsensor to None([None])
[rtm.py]    Connect sh.vrhsensorOut - es.vrhsensorIn
[rtm.py]    Connect es.vrhsensorOut - ic.ref_vrhsensorIn
[rtm.py]    Connect es.vrhsensorOut - abc.ref_vrhsensor
[rtm.py]    Failed to connect abc.limbCOPOffset_vrhsen
```

iory, 2016-03-02:
```
rostopic echo /vlhsensor
```
で値は読めています。

snozawa, 2016-03-03:
この部分は、データポートを接続しようとしてできないというメッセージです。
`[RTC名].[ポート名]`になっていて、出ているエラーはAutoBalancer (abc)のものです。
現状のimpedanceControllerの挙動を確認するのには問題はなさそうなので、ImpedanceControllerがvirtual force sensorの出力で使えているかのデバッグを引き続き行ってもらえるとたすかります。

auto-blancerの他にstabilizerなども同様にvirtual force sensorに対応していないメッセージがでるかもしれず、
本当は対応がいるかもしれないです。

> この方法を探していましたが、難しいのですか。

以前かいたissueを探してて見つかってないですが、
virtual force sensorもVRMLにForceSensor PROTOで定義します。
そして、ForceSensor PROTOをかいたセンサの値はRobotHardware RTCからでるのですが、
virtual force sensorに対応する箇所のみVirtualForceSensorRTCからの出力にデータポートをつなぎ変えるイメージです。
@k-okadaとの議論して現状なしの方向に、以前はなりました。
他によい方法が思いついたら教えてもらえるとたすかります。

iory, 2016-03-03:
なるほど。
試しにhrp4rで足のvirtual forceを出力してみたら、値が０となっていました。

これはうまく行っていないと思うのですが、
virtual forceの計算を有効にするトリガがあるのか、
そもそも設定が間違っていて計算できていないのでしょうか？

```
 iori@welsh: rostopic echo /rfsensor -n1
header: 
  seq: 51696
  stamp: 
    secs: 262
    nsecs: 889999999
  frame_id: rfsensor
wrench: 
  force: 
    x: -26.8188709269
    y: -2.09600054733
    z: 163.374527846
  torque: 
    x: -7.4608621825
    y: 3.1149973348
    z: 3.71985411649
---
 iori@welsh: rostopic echo /vrfsensor -n1
header: 
  seq: 52241
  stamp: 
    secs: 1456980583
    nsecs: 380542000
  frame_id: vrfsensor
wrench: 
  force: 
    x: 0.0
    y: 0.0
    z: 0.0
  torque: 
    x: 0.0
    y: 0.0
    z: 0.0
---
```

iory, 2016-03-03:
こちらのissueにも関係あるかと思います。
https://github.com/fkanehiro/hrpsys-base/issues/48

snozawa, 2016-03-03:
>  試しにhrp4rで足のvirtual forceを出力してみたら、値が０となっていました。

そういえばHRP４Rは足に設定がかいてあったきもしますが、
足のvirtual forceはその値自体もimpedance controllerとの接続でも使われてきてないです。
理由はそもそも(HRP4Rなら)実在力センサが ついてるので（少なくても足先力計測・力制御には）必要ないからです。
https://github.com/fkanehiro/hrpsys-base/issues/48
のissueも色々オフラインで話しているうちになんだかいらない、
もしくはやろうとすることを突き詰めるとvirtual force sensor対応だけでなくものすごく色々必要そうになってきたかんじになってきていました
そのため、(当時はgithubでなくてgooglecodeでしたが)issueには返信がついておらず進んでませんでした。

おそらく@ioryもこのissueをきったときに、元々何かやりたいこと・やるべきことがあって、
HRP4Rを起動もしくは、impedanceを起動してアレ動かない、となってissueを作成してたと思うので、
- そもそも何をやろうとしていて
- ここのデバッグではどこまでできたらOKとするか

などを列挙してみてください。

その時に、足のvirtual force sensorがやはり必要であれば再度ニーズ・どういう使い方をしているかなどをかいてみてください。
例えば、足先での力計測は実在力センサがあるので必要ないけど、膝や（腕なら）肘とかで計測がいるっぽい、といったニーズはあると思います

iory, 2016-03-03:
・そもそも何をやろうとしていて
hrp4rで物を持つということと、台車を手で押すということを考えています。
・ここのデバッグはどこまでできたらOKとするか
物を持つ、台車を押すという目的のために手に力センサがついていないhrp4rではvirtual forceが機能していないと、実現が難しいのではないかと考えておりvirtual forceの出力が正常にできることが必要な条件です。

そのデバッグ過程で手のvirtual forceも０となってしまっており、
そもそも力センサから測定している足でvirtual forceの出力が正常に出ているのかという判断をするために
>試しにhrp4rで足のvirtual forceを出力してみたら、値が０となっていました。

といったことを行っていました。

```
 iori@welsh: rostopic echo /vrhsensor -n1
header: 
  seq: 522077
  stamp: 
    secs: 1456982935
    nsecs: 749746000
  frame_id: vrhsensor
wrench: 
  force: 
    x: 0.0
    y: 0.0
    z: 0.0
  torque: 
    x: 0.0
    y: 0.0
    z: 0.0
---
```

snozawa, 2016-03-03:
> そのデバッグ過程で手のvirtual forceも０となってしまっており、
> そもそも力センサから測定している足でvirtual forceの出力が正常に出ているのかという判断をするために

これは今度はvirtual force sensor + impedance controlの組み合わせでなく
VirtualForceSensorそのものの問題なので、
https://github.com/fkanehiro/hrpsys-base/issues/959#issue-137461097
のようにlaunch起動後の出力をおくってください。
また、シミュレータ起動後しばらくして
```
rtconf localhost:15005/vs.rtc set debugLevel 1
```
とするとVirtualForceSensorがデバッグ情報をprintします。
それも送ってください。

また、起動後しばらくして
```
rtcat -ll localhost:15005/vs.rtc
```
とするとポートの接続関係も見えるので、それも送ってください。

hrp4r.launchが（多分セグフォで）起動しないのは以前まで動いているはずが動かない系のバグですが、
以前までの正常な状態で（少なくても@orikumaが実機実験してる時期に）
hrpsys-simulator上でVirtualForceSensorが値を出力するかはだれかに聞きましたか？
実は僕はこれを確認したことがないです。
VirtualForceSensor（センサ自体）+Impedanceの組み合わせ自体は、目標力をセットできれば確認できるので、シミュレータ上でそれを確認したことはあります。
https://github.com/fkanehiro/hrpsys-base/blob/master/sample/SampleRobot/samplerobot_virtual_force_sensor.py

ちなみに@orikumaさん
シミュレータ上で値がでることを確認したことあります？

> ・そもそも何をやろうとしていて
> hrp4rで物を持つということと、台車を手で押すということを考えています。
> ・ここのデバッグはどこまでできたらOKとするか
> 物を持つ、台車を押すという目的のために手に力センサがついていないhrp4rではvirtual forceが機能していないと、実現が難しいのではないかと考えておりvirtual forceの出力が正常にできることが必要な条件です。

なるほど。
そうであれば、AutoBalancer (場合によってはST)の部分も同様にVirtualForce対応の修正が何かいりそうです。
ちなみにHRP2、JAXON、SampleRobotなどの力センサがついてるロボットで
- 物をもつ
- 台車をおす

のがどのプログラムがどのようにして行っていて、hrpsys-baseのサンプルとしてはどれに相当するかはきいてますか？

また、これは具体的な対象物や、実機でできてる必要がある、シミュレータも同様にできる必要がある、など何か目標はありますか？
シミュレータである程度は確認できて、このissueの起動しない系のバグはtravisチェックできるとうれしいですが、
実際手でマニピュレーションする実験をシミュレータ上でやるのはだいぶ限界があるので、
どこかで実機実験にシフトするか、シミュレータでもできる必要があればけっこうがんばる必要があるかも。
具体的には、
- 力の値をシミュレータ上でうまく扱うのはけっこう骨がおれてVirtualFOrceSensorが動くか微妙
- HRP４Rのハンドはだいぶ簡略化されたものなので、そもそもものを把持できないか、対象物とつかみ方を（シミュレータ特有に）うまく工夫する必要がある

iory, 2016-03-03:
launch起動後の出力を送ります。
 
```
iori@welsh$ rtmlaunch hrpsys_ros_bridge_tutorials hrp4r.launch
[rtmlaunch] Start omniNames at port 15005

Thu Mar  3 16:07:41 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003133332e31312e3231362e31333400009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000bde2d756010016e4
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iori/.ros/log/9ee548fc-e10e-11e5-8ba8-28d244359c2b/roslaunch-welsh-5854.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/iori/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://133.11.216.134:55324/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path:
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    AutoBalancerServiceROSBridge (hrpsys_ros_bridge/AutoBalancerServiceROSBridgeComp)
    CollisionDetectorServiceROSBridge (hrpsys_ros_bridge/CollisionDetectorServiceROSBridgeComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    EmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HardEmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    ImpedanceControllerServiceROSBridge (hrpsys_ros_bridge/ImpedanceControllerServiceROSBridgeComp)
    KalmanFilterServiceROSBridge (hrpsys_ros_bridge/KalmanFilterServiceROSBridgeComp)
    RemoveForceSensorLinkOffsetServiceROSBridge (hrpsys_ros_bridge/RemoveForceSensorLinkOffsetServiceROSBridgeComp)
    SequencePlayerServiceROSBridge (hrpsys_ros_bridge/SequencePlayerServiceROSBridgeComp)
[... 1391 lines cut ...]
[ INFO] [1456988871.725095993, 3.659999999]:     JOINT: L_HIP_Y L_HIP_R L_HIP_P L_KNEE_P L_ANKLE_P L_ANKLE_R
[addJointGroup] group name = lleg
[rtm.py]    Connect rmfo.off_lfsensor - log.rmfo_off_lfsensor
[ INFO] [1456988871.748355395, 3.684999999]: ADD_GROUP: rleg (/rleg_controller)
[ INFO] [1456988871.748394397, 3.684999999]:     JOINT: R_HIP_Y R_HIP_R R_HIP_P R_KNEE_P R_ANKLE_P R_ANKLE_R
[addJointGroup] group name = rleg
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = off_rhsensor to rmfo_off_rhsensor
[ INFO] [1456988871.773718593, 3.709999999]: ADD_GROUP: torso (/torso_controller)
[ INFO] [1456988871.773755056, 3.709999999]:     JOINT: CHEST_P CHEST_Y
[addJointGroup] group name = torso
[rtm.py]    Connect rmfo.off_rhsensor - log.rmfo_off_rhsensor
[ INFO] [1456988871.800624588, 3.734999999]: ADD_GROUP: head (/head_controller)
[ INFO] [1456988871.800657791, 3.734999999]:     JOINT: NECK_Y NECK_P
[addJointGroup] group name = head
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = off_lhsensor to rmfo_off_lhsensor
[rtm.py]    Connect rmfo.off_lhsensor - log.rmfo_off_lhsensor
[log] Log cleared
[hrpsys.py] setup joint groups
[hrpsys.py] initialized successfully
[hrpsys_py-4] process has finished cleanly
log file: /home/iori/.ros/log/9ee548fc-e10e-11e5-8ba8-28d244359c2b/hrpsys_py-4*.log
[ INFO] [1456988872.468165340, 4.409999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 208[Hz] (exec 1777 Hz, dropped 1)
[INFO] [WallTime: 1456988873.071682] [5.010000] check 0 collision status, 0.000000 Hz
[ INFO] [1456988873.470301270, 5.414999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 201[Hz] (exec 1795 Hz, dropped 0)
```

iory, 2016-03-03:
以下が
```
rtconf localhost:15005/vs.rtc set debugLevel 1
```
を送った後の出力です。
```
[ INFO] [1456989397.252636562, 82.959999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 202[Hz] (exec 1811 Hz, dropped 0)
^T  sensor name  : vlhsensor
sensor torque  :  0 0 0 0 0 0 0
    raw force  :  0 0 0 0 0 0
  output force  :  0 0 0 0 0 0
  sensor name  : vrhsensor
sensor torque  :  0 0 0 0 0 0 0
    raw force  :  0 0 0 0 0 0
  output force  :  0 0 0 0 0 0
[ INFO] [1456989398.261499334, 83.969999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 202[Hz] (exec 1809 Hz, dropped 0)
  sensor name  : vlhsensor
sensor torque  :  0 0 0 0 0 0 0
    raw force  :  0 0 0 0 0 0
  output force  :  0 0 0 0 0 0
  sensor name  : vrhsensor
sensor torque  :  0 0 0 0 0 0 0
    raw force  :  0 0 0 0 0 0
  output force  :  0 0 0 0 0 0
``` 

次は
```
rtcat -ll localhost:15005/vs.rtc
```
を送った後のポート接続の出力です。

```
 iori@welsh$ rtcat -ll localhost:15005/vs.rtc              
vs.rtc  Active
  Category       example
  Description    null component
  Instance name  vs
  Type name      VirtualForceSensor
  Vendor         AIST
  Version        315.7.0
  Parent         
  Type           Monolithic
  Extra properties:
    abc_leg_offset                           0.0, 0.079919, 0.0
    activity_type                            DataFlowComponent
    alarm_ratio                              0.75
    category                                 example
    conf.default.debugLevel                  0
    config.version                           1.1.0
    config_file                              
    corba.endpoint                           
    corba.id                                 omniORB
    description                              null component
    dt                                       0.005
    end_effectors                            rarm,R_WRIST_R,CHEST_Y,0.0,0.0,-0.1,-1.471962e-17,1.0,-1.471962e-17,1.5708, larm,L_WRIST_R,CHEST_Y,0.0,0.0,-0.1,1.471962e-17,1.0,1.471962e-17,1.5708, rleg,R_ANKLE_R,WAIST,0.0,0.0,-0.091849,0.0,0.0,0.0,0.0, lleg,L_ANKLE_R,WAIST,0.0,0.0,-0.091849,0.0,0.0,0.0,0.0,
    exec_cxt.evdriven.type                   EventDrivenExecutionContext
    exec_cxt.periodic.rate                   1000000
    exec_cxt.periodic.type                   SynchExtTriggerEC
    implementation_id                        VirtualForceSensor
    instance_name                            vs
    lang_type                                compile
    language                                 C++
    logger.enable                            YES
    logger.log_level                         INFO
    max_instance                             10
    model                                    file:///home/iori/catkin_ws/hrp4/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP4R/HRP4Rmain.wrl
    motor_heat_params                        0.0075,0.5, 0.003,0.5, 0.003,0.5, 0.003,0.5, 0.003,0.5, 0.0075,0.5, 0.0075,0.5, 0.003,0.5, 0.003,0.5, 0.003,0.5, 0.003,0.5, 0.0075,0.5, 0.003,0.5, 0.003,0.5, 0.2,0.3, 0.2,0.3, 0.003,0.5, 0.003,0.5, 0.02,0.3, 0.003,0.5, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.003,0.5, 0.003,0.5, 0.02,0.3, 0.003,0.5, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.2,0.3, 0.2,0.3
    motor_temperature_limit                  60.0, 60.0, 60.0, 60.0, 45.0, 60.0, 60.0, 60.0, 60.0, 60.0, 45.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0, 60.0
    naming.enable                            YES
[... 135 lines cut ...]
> そうであれば、AutoBalancer (場合によってはST)の部分も同様にVirtualForce対応の修正が何かいりそうです。

他の方がどういった風にhrpsysを学んでいくかがわかっていませんが、今までhrpsysにあまり触れてくることがなかったので、中のシステムと実機の制御（研究会などでAutoBalancerやSTなどの単語が出てくると思いますが）を理解するためにもやる必要があるのかと考えています。

>ちなみにHRP2、JAXON、SampleRobotなどの力センサがついてるロボットで
>物をもつ
>台車をおす
>のがどのプログラムがどのようにして行っていて、hrpsys-baseのサンプルとしてはどれに相当するかはきいてますか？

http://wiki.ros.org/rtmros_common/Tutorials/WorkingWithEusLisp
サンプルロボットの例ならばtutorialにあり、そこにeusコードがあるのでそれを少し見た程度なので、
深く理解しているわけではないです。

>また、これは具体的な対象物や、実機でできてる必要がある、シミュレータも同様にできる必要がある、など何か目標はありますか？

物を持つということに関しての具体的な対象物は決まっていませんが、
最終的な目標としては障害物のあるような場所でも台車を押して目的地まで到達できることを目指していてます。物を持つ（道具を持つ）ということの延長として、台車を考えています。

>シミュレータである程度は確認できて、
>このissueの起動しない系のバグはtravisチェックできるとうれしいですが、
>実際手でマニピュレーションする実験をシミュレータ上でやるのはだいぶ限界があるので、
>どこかで実機実験にシフトするか、シミュレータでもできる必要があればけっこうがんばる必要があるかも。

実機実験にシフトするということは、実機だけで行う（シミュレータなし）で行うということでしょうか。
学習の試行回数を増やすことを考えると、シミュレータ上でもできる方が良いと思います。

orikuma, 2016-03-03:
> ちなみに@orikumaさん
> シミュレータ上で値がでることを確認したことあります？
だいぶ前なので当時の記憶があいまいで申し訳ありませんが,
実機のみで試していてシミュレーション上では確認していなかったような気がします.

@iory さんのログをちらっと見たところだとvs.rtc:tauInに値が正しく来ているか怪しそうですが
tf.rtc:tauOutはトルクらしき値を出してきているでしょうか?

iory, 2016-03-03:
@orikuma さん
```
rtconf localhost:15005/tf.rtc set debugLevel 1
```
の出力が以下です。
```
[ INFO] [1456993114.620554126, 3809.315000059]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 202[Hz] (exec 1801 Hz, dropped 0)
[tf]raw torque:  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[tf]gravity compensation:  0 -0.932961 -0.11741 -0.198635 0.0302325 0.064589 0 0.719988 -0.117556 -0.198635 0.0302325 -0.0645891 -5.48932 0 0 0.0116432 -0.124583 0.0359371 0 -0.102242 0 0.00492444 0.025745 -0.00121044 1.09462e-05 -0.132598 -0.0203242 0 -0.110257 0 -0.0030875 -0.0258537 0.00121044 -0.000122617
[tf]filtered torque:  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[ INFO] [1456993115.621582329, 3810.325000059]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 202[Hz] (exec 1791 Hz, dropped 0)
[tf]raw torque:  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[tf]gravity compensation:  0 -0.932961 -0.11741 -0.198635 0.0302325 0.064589 0 0.719988 -0.117556 -0.198635 0.0302325 -0.0645891 -5.48932 0 0 0.0116432 -0.124583 0.0359371 0 -0.102242 0 0.00492444 0.025745 -0.00121044 1.09462e-05 -0.132598 -0.0203242 0 -0.110257 0 -0.0030875 -0.0258537 0.00121
```

raw torqueがすべて０になっているのでtf.rtc:tauOutに正しくトルクが出てきていないということでしょうか。

orikuma, 2016-03-03:
vs.rtc:tauInに来ているtorqueがすべて0ならvs.rtcの出力はすべて0になります.
tf.rtcの出力でraw torqueが出ていないということはtf.rtc:tauInにすべて0の入力が来ていそうです.
tf.rtc:tauInからtorqueのパイプラインをたどっていって, すべて0の出力を出しているrtcが原因かと思います.

iory, 2016-03-03:
@orikuma さん
ありがとうございます。
まだrtcのデバッグ方法がわかっていないのですが、
具体的にどういう風に（コマンド）でたどっていけばいいでしょうか。

orikuma, 2016-03-03:
基本的には`rtcat -lv hostname:portname/rtcname` などで接続先を確認して, `rtprint hostname:portname/rtcname:portname` あたりで出力を見ればよいかと思います.

iory, 2016-03-04:
@snozawa さん
@orikuma さん
```
rtprint localhost:15005/HRP4R\(Robot\)0.rtc:tau   
```
の出力が
```
[305.629999999] [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
[305.634999999] [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
[305.639999999] [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
常に０となっているのですが、この原因はなんでしょうか？
私の理解だと、rtprint localhost:15005/HRP4R\(Robot\)0.rtc:tau でロボットの関節トルクが出力されると思うのですが。

snozawa, 2016-03-04:
あ、これはでないかもしれないですね。

hrpsys-simulatorはHighGainモードというのとtorqueControlモードという２つの関節駆動モードがあって、前者の場合だとトルクが計算されず出力もされなかった気がします。

ということで、impedancecontroller + virtual force sensorのデバッグとしては、
`(send *ri* :start-impedance :rarm)`
をおこなって、落ちなければOKそうです。

iory, 2016-03-04:
```
roseus hrp4r-interface.l
```
で、
```
7.E2-irteusgl$ send *ri* :start-impedance :rarm
#<hrpsys_ros_bridge::openhrp_impedancecontrollerservice_startimpedancecontrollerresponse #X8e4dcb0>
8.E2-irteusgl$ send *ri* :start-impedance :larm
#<hrpsys_ros_bridge::openhrp_impedancecontrollerservice_startimpedancecontrollerresponse #X8f4a430>
```
を行って落ちませんが、
```
rtconf localhost:15005/vs.rtc set debugLevel 1
```
で出力を見てみると
```
[ INFO] [1457114606.220624550, 245.539999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 202[Hz] (exec 1800 Hz, dropped 0)
  sensor name  : vlhsensor
sensor torque  :  0 0 0 0 0 0 0
    raw force  :  0 0 0 0 0 0
  output force  :  0 0 0 0 0 0
  sensor name  : vrhsensor
sensor torque  :  0 0 0 0 0 0 0
    raw force  :  0 0 0 0 0 0
```
となっていてやはり出力が出ていないですね。

snozawa, 2016-03-06:
はい、HighGainモードではトルクがでないので、vsのトルクおよびエンドエフェクタの力・モーメントは０になってると思います。

それで、確認としてはstart-impedance後に
```
(send *ri* :set-ref-force #f(10 20 30) :rarm)
```
してみて、ref-force値（目標手先力）におうじて右腕が動けばimpedance+virtual force sensorの組み合わせ自体はOKです。

iory, 2016-04-11:
またこの話にもどってしまうのですが，
これを適用した上で，
rtmlaunch hrpsys_ros_bridge_tutorials hrp4r_startup.launch
を行うとSoftErrorLimitter is not workingが出てしまいます．
```
 rtmlaunch hrpsys_ros_bridge_tutorials hrp4r_startup.launch
[rtmlaunch] Start omniNames at port 15005 

Mon Apr 11 19:41:42 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003133332e31312e3231362e31373400009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000667f0b5701001b8c
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iory/.ros/log/fae4deb0-ffd1-11e5-9be2-507b9d9efada/roslaunch-crab-7046.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.174:41901/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    hrpsys (hrpsys/hrpsys-simulator)
    hrpsys_py (hrpsys_tools/hrpsys_tools_config.py)
    modelloader (openhrp3/openhrp-model-loader)

auto-starting new master
process[master]: started with pid [7061]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fae4deb0-ffd1-11e5-9be2-507b9d9efada
process[rosout-1]: started with pid [7074]
started core service [/rosout]
process[modelloader-2]: started with pid [7077]
ready
process[hrpsys-3]: started with pid [7103]
process[hrpsys_py-4]: started with pid [7104]
loading file:///home/iory/catkin_ws/hrp4/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP4R/HRP4Rmain.wrl
The "equivalentInertia" field of the Joint node is obsolete.
Humanoid node
Joint node WAIST
  Segment node BODY
    AccelerationSensorgsensor
    Gyrogyrometer
  Joint node R_HIP_Y
      Segment node R_HIP_Y_LINK
      Joint node R_HIP_R
          Segment node R_HIP_R_LINK
          Joint node R_HIP_P
              Segment node R_HIP_P_LINK
              Joint node R_KNEE_P
                  Segment node R_KNEE_P_LINK
                  Joint node R_ANKLE_P
                      Segment node R_ANKLE_P_LINK
                      Joint node R_ANKLE_R
                          Segment node R_ANKLE_R_LINK
                            ForceSensorrfsensor
                          Joint node R_FOOT
[... 807 lines cut ...]
[hrpsys_py-4] process has finished cleanly
log file: /home/iory/.ros/log/fae4deb0-ffd1-11e5-9be2-507b9d9efada/hrpsys_py-4*.log
SoftErrorLimiter is not working...
         m_qRef 0
     m_qCurrent 34
   m_servoState 34
[co] beepCommand data port connection found! Use BeeperRTC.
[el] beepCommand data port connection found! Use BeeperRTC.
SoftErrorLimiter is not working...
         m_qRef 0
     m_qCurrent 34
   m_servoState 34
SoftErrorLimiter is not working...
         m_qRef 0
     m_qCurrent 34
   m_servoState 34
SoftErrorLimiter is not working...
         m_qRef 0
     m_qCurrent 34
   m_servoState 34
SoftErrorLimiter is not working...
         m_qRef 0
     m_qCurrent 34
   m_servoState 34
```

snozawa, 2016-04-11:
elまでqRefデータポートが届いてないですね。
ログをみると、activateCompsの中で各RTCをstart (Activate) しているところで、
onActivate とだけ出力されるのが正しいところ、
```
[st] onActivated(1000)
[st] onDeactivated(1000)
```
となってるので、今度はSTがonExecuteなどでエラってる気がします。
同様にSTのどこで落ちているかしらべてみてください。

iory, 2016-04-11:
```
https://github.com/fkanehiro/hrpsys-base/blob/d06269ebefbfa418535809e0f7f0122e8a133188/rtc/Stabilizer/Stabilizer.cpp#L529
```
getActualParameters()という関数で落ちているようです．

iory, 2016-04-11:
さらに調べるとcalcZMPの中の以下の部分で落ちていました．
https://github.com/fkanehiro/hrpsys-base/blob/d06269ebefbfa418535809e0f7f0122e8a133188/rtc/Stabilizer/Stabilizer.cpp#L1088
このときのstikp[i].sensor_nameの値はrfsensorでした．

iory, 2016-04-11:
https://github.com/fkanehiro/hrpsys-base/blob/d06269ebefbfa418535809e0f7f0122e8a133188/rtc/Stabilizer/Stabilizer.cpp#L172
npforceの値が2になっていて,
stikp.size()が4になっているため，
メモリ確保をしていないm_wrenches[2~3]にアクセスしようとしています．
このstikp (stikparameter) もしくはnpforceはどこの値を参照しているでしょうか．

snozawa, 2016-04-12:
おそらくここもvirtualForceSensor対応できてない箇所があるため、
このようになっていると思います。
stikpはHRP4C.confのend_effectorを、
npforceは現状は実在力センサを参照しています
後者がvirtualForceSensor対応になれば、センサの数が4になり数にあうと思います。

iory, 2016-04-12:
以下のPRのように修正した結果，
https://github.com/fkanehiro/hrpsys-base/pull/976
```
rtmlaunch hrpsys_ros_bridge_tutorials hrp4r_startup.launch
```
を起動中に落ちることはなくなりました．
そこで
```
rtmlaunch hrpsys_ros_bridge_tutorials hrp4r.launch
```
を実行したところ，下記のエラーが発生しました．
ros_bridgeで落ちているのですが，これは
```
Warning:  Joint ID is not given to joint L_F33 of model HRP4R.
```
などのWarningが出ていることからwrlファイルが原因でしょうか？

```
 iory@crab:~$ rtmlaunch hrpsys_ros_bridge_tutorials hrp4r.launch        

[rtmlaunch] Start omniNames at port 15005 

Wed Apr 13 01:27:47 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003133332e31312e3231362e31373400009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c00000001000000010001000100000001000105090101000100000009010100035454410800000003220d5701000e7d
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iory/.ros/log/7e795a2c-00cb-11e6-a54b-507b9d9efada/roslaunch-crab-3702.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.174:35145/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True
[... 1312 lines cut ...]
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   24 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   25 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[co] CAUTION!! The robot is moving without checking self collision detection!!! please define collision_pair in configuration file
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   26 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   27 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   28 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle   29 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[co] CAUTION!! The robot is moving without checking self collision detection!!! please define collision_pair in configuration file
[rtmlaunch] [ERROR] Could not Connect ( /localhost:15005/HRP4R(Robot)0.rtc:q , /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle ):  No such object: /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle 
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc   0 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc   1 /30
[sensor_ros_bridge_connect.py]   wait for  HrpsysSeqStateROSBridge0  :  None
[rtmlaunch] Wait for  /localhost:15005/HrpsysSeqStateROSBridge0.rtc   2 /30
```

snozawa, 2016-04-13:
GDBしてみてくれるかな？
hrpsys_ros_bridge/launch/hrpsys_ros_bridge.launchの中の
```
  <node name = "HrpsysSeqStateROSBridge"
```
のROSBridgeを起動している箇所に
```
        launch-prefix='xterm -e gdb -ex run --args'
```
を追加してROSBridgeがxtermの別端末でGDBつきで起動します。
その結果をコピペしてもらえるかな。
また、それでどこでおちてるか多少調べられます。

iory, 2016-04-13:
gdbした結果，
hrpsys_ros_bridgeないで同じようにvirtual forceに未対応の部分があったので，PRを送りました．
https://github.com/start-jsk/rtmros_common/pull/923

iory, 2016-04-13:
HRP4Rのlaunchは落ちることなく，立ち上がるようになりました

k-okada, 2016-04-14:
おお，ナイス．close しても大丈夫かな？ @iory

iory, 2016-04-14:
はい，大丈夫です．

k-okada, 2016-04-14:
自分でクローズできるかな？

iory, 2016-04-14:
クローズしました

closed 2016-04-14 by iory