fkanehiro/hrpsys-base #969: interprocess_semaphoreの利用について

snozawa, 2016-04-06:
rtc以下の
```
rtc/StateHolder/StateHolder.h:  boost::interprocess::interprocess_semaphore m_waitSem, m_timeSem;
rtc/RobotHardware/robot.h:    boost::interprocess::interprocess_semaphore wait_sem;
rtc/SequencePlayer/SequencePlayer.h:  boost::interprocess::interprocess_semaphore m_waitSem;
```
の主に３箇所くらいでboostのinterprocess_semaphoreが使われているのですが、
boostでないsemaphore.hをお使いでなくこちらをお使いの理由はなにになりますでしょうか。
ちなみに、boost versionが1.35.0以前だとboostのinterprocess_semaphoreが使えない状況のようです。

よろしくお願いいたします。

fkanehiro, 2016-04-06:
え～っと、正直記憶にありません。
普通のセマフォでいい気がします。

snozawa, 2016-04-07:
ありがとうございます

> 普通のセマフォでいい気がします。

boost versionが1.35.0以前でも（パッチなく）動作するようにしようと思っているのですが、
- interprocess_semaphore=>semaphoreに変更する、多分大丈夫そうですが挙動はかわりそうです
- boost versionを参照して、新しければ現状と同じ挙動、古ければsemaphoreになるようにifdefなどで対応する

のどちらがよろしいでしょうか。

fkanehiro, 2016-04-07:
スッキリと統一する、でいいんじゃないでしょうか。

snozawa, 2016-04-09:
https://github.com/fkanehiro/hrpsys-base/issues/970
でsemaphore.hを使うように統一いたさいmした。

ありがとうございました。

closed 2016-04-09 by snozawa