fkanehiro/hrpsys-base #979: ポートがつながっていても，またつなげてしまう．

k-okada, 2016-04-14:
ロボットは何でもいいんですが，
```
$ rtmlaunch nextage_ros_bridge nextage_ros_bridge_simulation.launch 
```
としてから，ロボット操作しようとして
```
$ rosrun nextage_ros_bridge nextage.py
```
とすると，
```
[rtm.py]    Connect sh.qOut - el.qRef
[rtm.py]    Connect el.q - HGcontroller0.qIn
[rtm.py]    Connect HGcontroller0.qOut - HiroNX(Robot)0.qRef
[rtm.py]    Connect HiroNX(Robot)0.servoState - el.servoStateIn
[rtm.py]    Connect HiroNX(Robot)0.q - sh.currentQIn
[rtm.py]    Connect HiroNX(Robot)0.q - fk.q
[rtm.py]    Connect sh.qOut - fk.qRef
[rtm.py]    Connect seq.qRef - sh.qIn
[rtm.py]    Connect seq.tqRef - sh.tqIn
[rtm.py]    Connect seq.basePos - sh.basePosIn
[rtm.py]    Connect seq.baseRpy - sh.baseRpyIn
[rtm.py]    Connect seq.zmpRef - sh.zmpIn
[rtm.py]    Connect seq.optionalData - sh.optionalDataIn
[rtm.py]    Connect sh.basePosOut - seq.basePosInit
[rtm.py]    Connect sh.basePosOut - fk.basePosRef
[rtm.py]    Connect sh.baseRpyOut - seq.baseRpyInit
[rtm.py]    Connect sh.baseRpyOut - fk.baseRpyRef
[rtm.py]    Connect sh.qOut - seq.qInit
[rtm.py]    Connect sh.zmpOut - seq.zmpRefInit
[rtm.py]    Connect HiroNX(Robot)0.q - el.qCurrent
```
とかひょうじされてポートを接続してくれます．
でも言う一回，nexgtage.py を実行すると同じように表示されます．で問題は，
最初の時は，
```
rtcat -l localhost: 15005/sh.
rtc | grep Connect | uniq -c
      2  +Connected to  /localhost:15005/HiroNX(Robot)0.rtc:q
      2  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:qRef
      2  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:tqRef
      2  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:basePos
      2  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:baseRpy
```
だったのに，２回めには
```
$ rtcat -l localhost:15005/sh.rtc | grep Connect | uniq -c
      3  +Connected to  /localhost:15005/HiroNX(Robot)0.rtc:q
      3  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:qRef
      3  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:tqRef
      3  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:basePos
      3  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:baseRpy
```
みたいになって接続がパラレルにされている状況になります．
あれ，ConnectPortのまえにIs Already Connectedみたいな表示が出ないのなんで？と思ってみてみると，
https://github.com/fkanehiro/hrpsys-base/commit/5205eb47ef4284b779bfa6d191f04f366a1c87a5#diff-00040ddccd3d05c29c4cc32c61e7e024R535
誰だコンナ変更したのは！！！！とおもったら自分でした．ということで，
https://github.com/fkanehiro/hrpsys-base/pull/978　で直しました．

最近いろいろ問題が出ているのはこれかもしれません @snozawa 
さっきは最初っから２つ同じ接続があったりしましたが（つまりnextage.pyを上げなくてもlaunchでhrpsys_config_tools.pyか何かが２回よばれたのでしょう），上のPRで全て一になったので，これで良いともいます．

```
$ rtcat -l localhost:15005/sh.rtc | grep Connect | uniq -c
      1  +Connected to  /localhost:15005/HiroNX(Robot)0.rtc:q
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:qRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:tqRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:basePos
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:baseRpy
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:zmpRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:optionalData
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/el.rtc:qRef
      1  +Connected to  /localhost:15005/fk.rtc:qRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:qInit
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/log.rtc:sh_qOut
      1  +Connected to  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:mcangle
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/ic.rtc:qRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/log.rtc:sh_tqOut
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:basePosInit
      1  +Connected to  /localhost:15005/fk.rtc:basePosRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/log.rtc:sh_basePosOut
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/ic.rtc:basePosIn
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:baseRpyInit
      1  +Connected to  /localhost:15005/fk.rtc:baseRpyRef
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/log.rtc:sh_baseRpyOut
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/ic.rtc:baseRpyIn
      1  +Connected to  /localhost:15005/HrpsysSeqStateROSBridge0.rtc:baseTform
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/seq.rtc:zmpRefInit
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/log.rtc:sh_zmpOut
      1  +Connected to  /localhost:15005/kokada-t440s.host_cxt/manager.mgr/StateHolderServiceROSBridge.rtc:StateHolderService
```

k-okada, 2016-04-14:
あと，毎回nextage.pyをあげるんではなくて，デバッグ用途にはhrpsyspyというコマンドを使うことを提案していますがあまりユーザはいなようです．

```
$ hrpsyspy 
configuration ORB with localhost:15005
[hrpsys.py] wait for RTCmanager : kokada-t440s
[hrpsys.py] wait for HiroNX(Robot)0 : <hrpsys.rtm.RTcomponent instance at 0x7f3c467730e0> ( timeout 0 < 10)
[hrpsys.py] findComps -> RobotHardware : <hrpsys.rtm.RTcomponent instance at 0x7f3c467730e0> isActive? = True 
[hrpsys.py] simulation_mode : True
Starting hrpsys python interface for HiroNX(Robot)0
 Installed rtc are ...  seq   sh   fk   rmfo   ic   el   log  
 Use 'hcf' instance to access to the hrpsys controller
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: hcf
Out[1]: <hrpsys.hrpsys_config.HrpsysConfigurator instance at 0x7f3c463a2488>
```
となって，僕的には使いやすいです．

k-okada, 2016-04-14:
さすがに今日は無理だけど、頑張ってテスト作ろうと思うのでマージはちょっと待って下さい

snozawa, 2016-04-15:
> 最近いろいろ問題が出ているのはこれかもしれません @snozawa
> さっきは最初っから２つ同じ接続があったりしましたが（つまりnextage.pyを上げなくてもlaunchでhrpsys_config_tools.pyか何かが２回よばれたのでしょう），上のPRで全て一になったので，これで良いともいます．

なるほど。
これ自体は結構前の変更ですね（１年ほどまえ）

snozawa, 2016-04-15:
> あと，毎回nextage.pyをあげるんではなくて，デバッグ用途にはhrpsyspyというコマンドを使うことを提案していますがあまりユーザはいなようです．

これは、hrpsys_tools_config.pyの上位互換版ですか？
（機能的に・ちゃんとhrpsys-baseにコミットされてる・..etc）
いいですねこれ！

k-okada, 2016-04-15:
https://github.com/fkanehiro/hrpsys-base/pull/574
これも高度一年前の変更です．

k-okada, 2016-04-15:
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_tools/scripts/hrpsys_tools_config.py
とは微妙に使い方がちがっていて，hrpsyspyはインタプリタですので，一個もRTCの立ち上げ接続はしないつもりです．
toolconfigのほうは最初の起動スクリプトですから，ipythonにはせずに落とすべきかなと思います．

snozawa, 2016-04-15:
> とは微妙に使い方がちがっていて，hrpsyspyはインタプリタですので，一個もRTCの立ち上げ接続はしないつもりです．
> toolconfigのほうは最初の起動スクリプトですから，ipythonにはせずに落とすべきかなと思います．

なるほど、わかりました。
多分今まで個人的に、hrpsys_tools_configをインタプリタ的につかってました
実際の起動は各ロボットごとに継承したもので接続したいときがあり、起動はロボットごとに継承したconfig.pyで行い、
hrpsys_tools_config.pyでは起動・接続が行われるけど、
（このissueの内容がなおれば）既に起動・接続されてるRTCがある場合には
そのまま素通りしてきてくれたためです。

k-okada, 2016-04-15:
ロボットごとに継承したconfigを使っていて，チョット違うつなぎ方をしているときに
hrpsys_tools_config.pyを使うと，チョット違っているところが反映されなくて，変なことに
なることが理屈上はあり得るので，いやだなぁ，と思っていました．実際はわかっている人が
やっているから，そんなに問題はないのかもしれないけど．

snozawa, 2016-04-15:
なるほど，了解です．
> 実際はわかっている人がやっているから，そんなに問題はないのかもしれないけど．

はい，これもありますし，hrpsys_tools_config.pyを使うのは簡易に確認したいときのみで，ちゃんとしたpythonからの確認や使い方をしたいときは各ロボットごとのhrpsys_config.pyをつかってたというのもあります．