fkanehiro/hrpsys-base #980: robotがroboになっている

k-okada, 2016-04-15:
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L945
ですが，
waitForRTCManagerAndRoboHardware
になっています．RobotHardwareのつもりだった．
どうしよう．．．

iory, 2016-04-19:
これは調べると
https://github.com/search?q=waitForRTCManagerAndRoboHardware&ref=opensearch&type=Code
そこまで多くの場所に書いてあるよう記述じゃないので，修正のPR送ればいいのでしょうか？
それとも，なにか深い理由があって変更は難しいことなのでしょうか？

k-okada, 2016-04-20:
start-jsk/rtmros_choreonoid – walking.py
jsk-ros-pkg/euslib – connect_port_sim.py

とか把握できていないところで使われているので，結構色々なところで使われているなぁ，という印象です．

    def waitForRTCManagerAndRobotHardware(self, robotname="Robot",
managerhost=nshost):

    def waitForRTCManagerAndRoboHardware(self, robotname="Robot",
managerhost=nshost):
                print self.configurator_name, "\e[1;31m
waitForRTCManagerAndRoboHardware has renamed to
waitForRTCManagerAndRoboHardware: Please update your code\e[0m"
               return waitForRTCManagerAndRoboHardware(self,
robotname="Robot", managerhost=nshost)

とかしておいて，気がついたところを直しておく，というのがいいと思いいます

そうでないと，これを修正すると，ユーザ全員に最新版を使ってね，となって，最新版にあげてコンパイルして失敗する，ということがありそうです．

iory, 2016-04-20:
なるほど．
レポジトリを超えて複数で使われているので，変更してしまうと大変なことになるということですか．
renameして以前のwaitForRTCManagerAndRoboHardwareを使うと黄色で警告がでるようにPRを送りました．