fkanehiro/hrpsys-base #986: hrpsyspyのhcf.servonOn() deサーボが立ち上がらない

iory, 2016-04-26:


snozawa, 2016-04-26:
@orikuma 
サーボはhrpsys_config.py系のpythonスクリプトからどうやってあげていましたか？

@iory 
rtcdの出力結果を添付してください。

iory, 2016-04-26:
rtcdの出力は以下です．
```
 $ rosrun hrp4r_hrpsys hrp4r_start_rtcd.sh
sending SIGKILL to rtcd...
Running on art-linux.
Running on art-linux.
rtcd.sh: no process found
rtcd: no process found
iob is unlocked by 4200
grxuser   4156  2.0  0.2   4308  1388 pts/0    S+   11:49   0:00 /bin/bash /home/grxuser/ros/fuerte/rtm-ros-robotics/rtmros_hrp4r/hrp4r_hrpsys/scripts/hrp4r_start_rtcd.sh
grxuser   4202  0.0  0.1   3364   820 pts/0    S+   11:49   0:00 grep rtcd.sh
grxuser   4156  2.0  0.2   4308  1388 pts/0    S+   11:49   0:00 /bin/bash /home/grxuser/ros/fuerte/rtm-ros-robotics/rtmros_hrp4r/hrp4r_hrpsys/scripts/hrp4r_start_rtcd.sh
grxuser   4204  0.0  0.1   3364   812 pts/0    S+   11:49   0:00 grep rtcd
Running on art-linux.
ln: cannot remove `hrpEC.so': Permission denied
ln: cannot remove `libhrp_servo.so': Permission denied
`io_main' -> `io_main_art'
iob is unlocked by 4218
Warning: format is changed to S16_BE

Tue Apr 26 11:49:19 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e3000000100000000000000e0000000010102000f0000003133332e31312e3231362e3232310000418c00000b0000004e616d6553657276696365000500000000000000080000000100000000545441010000001c00000001000000010001000100000001000105090101000100000009010100030000001a000000010000000f0000003133332e31312e3231362e32323100009d3a0000025454413f0000000100000009000000687270343030316300000000270000002f746d702f6f6d6e692d677278757365722f3030303030343233312d31343631363338393539000003545441080000002fd71e5701001087
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
ready
hrpExecutionContext for ART is registered
loading /opt/grx/HRP4R/model/HRP4Rmain.wrl
The "equivalentInertia" field of the Joint node is obsolete.
Humanoid node
Joint nodeWAIST
  Segment node BODY
    AccelerationSensorgsensor
    Gyrogyrometer
  Joint nodeR_HIP_Y
      Segment node R_HIP_Y_LINK
      Joint nodeR_HIP_R
          Segment node R_HIP_R_LINK
          Joint nodeR_HIP_P
              Segment node R_HIP_P_LINK
              Joint nodeR_KNEE_P
                  Segment node R_KNEE_P_LINK
                  Joint nodeR_ANKLE_P
                      Segment node R_ANKLE_P_LINK
                      Joint nodeR_ANKLE_R
                          Segment node R_ANKLE_R_LINK
                            ForceSensorrfsensor
                          Joint nodeR_FOOT
                            Segment node R_FOOT_LINK
  Joint nodeL_HIP_Y
      Segment node L_HIP_Y_LINK
      Joint nodeL_HIP_R
          Segment node L_HIP_R_LINK
          Joint nodeL_HIP_P
              Segment node L_HIP_P_LINK
              Joint nodeL_KNEE_P
                  Segment node L_KNEE_P_LINK
                  Joint nodeL_ANKLE_P
                      Segment node L_ANKLE_P_LINK
                      Joint nodeL_ANKLE_R
                          Segment node L_ANKLE_R_LINK
                            ForceSensorlfsensor
                          Joint nodeL_FOOT
                            Segment node L_FOOT_LINK
  Joint nodeCHEST_P
      Segment node CHEST_P_LINK
[... 641 lines cut ...]
[rtm.py]    Connect RobotHardware0.servoState - log.RobotHardware0_servoState (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = rfsensorOut to sh_rfsensorOut
[rtm.py]    Connect sh.rfsensorOut - log.sh_rfsensorOut (dataflow_[log] Log cleared
[el] setServoErrorLimit 0.1[rad] for all joints
[el] setServoErrorLimit 0.54[rad] for L_HAND_J0
[el] setServoErrorLimit 0.54[rad] for L_HAND_J1
[el] setServoErrorLimit 0.54[rad] for R_HAND_J1
[el] setServoErrorLimit 0.54[rad] for R_HAND_J0
type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = lfsensorOut to sh_lfsensorOut
[rtm.py]    Connect sh.lfsensorOut - log.sh_lfsensorOut (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = off_rfsensor to rmfo_off_rfsensor
[rtm.py]    Connect rmfo.off_rfsensor - log.rmfo_off_rfsensor (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[hrpsys.py]   setupLogger : record type = TimedDoubleSeq, name = off_lfsensor to rmfo_off_lfsensor
[rtm.py]    Connect rmfo.off_lfsensor - log.rmfo_off_lfsensor (dataflow_type=Push, subscription_type=flush, bufferlength=1, push_rate=1000, push_policy=new)
[hrpsys.py] setup joint groups
[hrpsys.py] initialized successfully

StateHolder::goActual()
StateHolder::goActual()
StateHolder::goActual()
StateHolder::goActual()
StateHolder::goActual()
StateHolder::goActual()
```

iory, 2016-04-26:
```
hcf.rh_svc.power(jname, OpenHRP.RobotHardwareService.SWITCH_ON)
```
というようなことをすると入ると思います．
hrpsyspy (https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L1718) でのservoOnはこれをするようになっていないのは仕様ですか

snozawa, 2016-04-26:
今JAXON,HRP２ではここのservoOnは使ってないので、
体内環境だと動かないかもしれない（GUIがでる）

https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/hrpsys_ros_bridge/hrpsys_dashboard.py#L188-L192
のように(上記はrosserviceごしですが）
power On
goActual
servo On
を１秒sleepいれて実行すると入るかな？

iory, 2016-04-26:
なるほど．
rosを通さないservo onは
```
hcf.rh_svc.power(jname ,OpenHRP.RobotHardwareService.SWITCH_ON)
for rtc in hcf.getJointAngleControllerList():
     rtc.stop();
     rtc.start();
time.sleep(1)
hcf.sh_svc.goActual()
hcf.rh_svc.servo(jname,OpenHRP.RobotHardwareService.SWITCH_ON)
```
でできました．
rosごしのservoONも戻ったら試してみます．
serviceを介して行うことも内部では同じようにhcfでやっているということでよろしいでしょうか？

snozawa, 2016-04-26:
servoOnは
https://github.com/fkanehiro/hrpsys-base/issues/485
こちらが進んでなくて、JAXON, HRP2で現役で使っているservoOn関数は
それぞれのロボットで（ほぼ同じ内容を）かいていて、
ここにcommitできてない状況です。

サーボONくらいが確認できたら、一度jaxon, hrp2同様に
HRP4Rのhrpsys_config.pyを一度見直すと良いかもしれません。
hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials以下に
hrp4r_hrpsys_config.pyがあって、それを継承して実機のものができるかんじになります。

snozawa, 2016-04-26:
>  rosごしのservoONも戻ったら試してみます．

ROSごしのservoOnは、現状動いてないのでいいです。
（StableRTCでは未確認ですが動くと思いますが、UnStableRTCではgoActual的なものを実行しなければならず、今のところこれがROSから呼ぶことができてないです）

ros service越しで行うのと、hcfでおこなうのとではちょっと違います。
JAXON,HRP2ともにhcfのほうでやっています。
idlにかいてある関数はros service ,hcfいずれからもよべます。
hcfのstart, stopなどもっと細かいopenrtmの指令はhcfからしか現状よべてないです。

closed 2017-03-28 by iory