fkanehiro/hrpsys-base #1006: #1005のテストで、いくつかのrtcでcreateCompに失敗する

rkoyama1623, 2016-06-06:
https://github.com/fkanehiro/hrpsys-base/pull/1005 のテスト11番が通らないのですが、
調べてみると、
hrpsys/sample/SampleRobot/samplerobot_reference_force_updater.py
hcfの中のいくつかのrtcを作れないというエラーが出ています。
エラー文を出しているコードは[ここ](https://github.com/rkoyama1623/hrpsys-base/blob/rfu_support_both_arms_pr/python/hrpsys_config.py#L563)です。

rfuのみならず、rmfoなどのrtcも作成に失敗しています。
そのため、rfuのテストで、rfuのポートをLoggerに接続しようとすると、rfuのインスタンスがNoneであるためにテストに失敗します。

どのような対策があり得るでしょうか？

以下、travisのログのエラー箇所の抜粋です。

全体のリンクは、
https://travis-ci.org/fkanehiro/hrpsys-base/jobs/135045094
https://s3.amazonaws.com/archive.travis-ci.org/jobs/135045094/log.txt
です。

```
[hrpsys.py] creating components
[hrpsys.py]   eval : [self.seq, self.seq_svc, self.seq_version] = self.createComp("SequencePlayer","seq")
[hrpsys.py] create Comp -> SequencePlayer : <hrpsys.rtm.RTcomponent instance at 0x3301638> (315.1.8)
[hrpsys.py] create CompSvc -> SequencePlayer Service : <OpenHRP._objref_SequencePlayerService instance at 0x33729e0>
[hrpsys.py]   eval : [self.sh, self.sh_svc, self.sh_version] = self.createComp("StateHolder","sh")
[hrpsys.py] create Comp -> StateHolder : <hrpsys.rtm.RTcomponent instance at 0x336e1b8> (315.1.8)
[hrpsys.py] create CompSvc -> StateHolder Service : <OpenHRP._objref_StateHolderService instance at 0x3307ab8>
[hrpsys.py]   eval : [self.fk, self.fk_svc, self.fk_version] = self.createComp("ForwardKinematics","fk")
[hrpsys.py] create Comp -> ForwardKinematics : <hrpsys.rtm.RTcomponent instance at 0x3306cb0> (315.1.8)
[hrpsys.py] create CompSvc -> ForwardKinematics Service : <OpenHRP._objref_ForwardKinematicsService instance at 0x3301e60>
[hrpsys.py]   eval : [self.tf, self.tf_svc, self.tf_version] = self.createComp("TorqueFilter","tf")
[hrpsys.py] create Comp -> TorqueFilter : <hrpsys.rtm.RTcomponent instance at 0x32ffc20> (315.1.8)
("can't find a service named", 'service0')
[hrpsys.py]   eval : [self.kf, self.kf_svc, self.kf_version] = self.createComp("KalmanFilter","kf")
[hrpsys.py] create Comp -> KalmanFilter : <hrpsys.rtm.RTcomponent instance at 0x3372a70> (315.1.8)
[hrpsys.py] create CompSvc -> KalmanFilter Service : <OpenHRP._objref_KalmanFilterService instance at 0x33028c0>
[hrpsys.py]   eval : [self.vs, self.vs_svc, self.vs_version] = self.createComp("VirtualForceSensor","vs")
[hrpsys.py] create Comp -> VirtualForceSensor : <hrpsys.rtm.RTcomponent instance at 0x3306830> (315.1.8)
[hrpsys.py] create CompSvc -> VirtualForceSensor Service : <OpenHRP._objref_VirtualForceSensorService instance at 0x336ecf8>
[hrpsys.py]   eval : [self.rmfo, self.rmfo_svc, self.rmfo_version] = self.createComp("RemoveForceSensorLinkOffset","rmfo")
[hrpsys.py] ?[31mFail to createComps 'NoneType' object has no attribute 'ref'?[0m
[hrpsys.py]   eval : [self.es, self.es_svc, self.es_version] = self.createComp("EmergencyStopper","es")
[hrpsys.py] ?[31mFail to createComps CORBA.UNKNOWN(omniORB.UNKNOWN_UserException, CORBA.COMPLETED_MAYBE)?[0m
[hrpsys.py]   eval : [self.rfu, self.rfu_svc, self.rfu_version] = self.createComp("ReferenceForceUpdater","rfu")
[hrpsys.py] ?[31mFail to createComps CORBA.UNKNOWN(omniORB.UNKNOWN_UserException, CORBA.COMPLETED_MAYBE)?[0m
[hrpsys.py]   eval : [self.ic, self.ic_svc, self.ic_version] = self.createComp("ImpedanceController","ic")
[hrpsys.py] create Comp -> ImpedanceController : <hrpsys.rtm.RTcomponent instance at 0x3371560> (315.1.8)
[hrpsys.py] create CompSvc -> ImpedanceController Service : <OpenHRP._objref_ImpedanceControllerService instance at 0x336ba70>
[hrpsys.py]   eval : [self.abc, self.abc_svc, self.abc_version] = self.createComp("AutoBalancer","abc")
[hrpsys.py] create Comp -> AutoBalancer : <hrpsys.rtm.RTcomponent instance at 0x33724d0> (315.1.8)
[hrpsys.py] create CompSvc -> AutoBalancer Service : <OpenHRP._objref_AutoBalancerService instance at 0x336bb48>
[hrpsys.py]   eval : [self.st, self.st_svc, self.st_version] = self.createComp("Stabilizer","st")
[hrpsys.py] create Comp -> Stabilizer : <hrpsys.rtm.RTcomponent instance at 0x3303f38> (315.1.8)
[hrpsys.py] create CompSvc -> Stabilizer Service : <OpenHRP._objref_StabilizerService instance at 0x336b440>
[hrpsys.py]   eval : [self.co, self.co_svc, self.co_version] = self.createComp("CollisionDetector","co")
[hrpsys.py] create Comp -> CollisionDetector : <hrpsys.rtm.RTcomponent instance at 0x3370440> (315.1.8)
[hrpsys.py] create CompSvc -> CollisionDetector Service : <OpenHRP._objref_CollisionDetectorService instance at 0x3372e60>
[hrpsys.py]   eval : [self.tc, self.tc_svc, self.tc_version] = self.createComp("TorqueController","tc")
[hrpsys.py] create Comp -> TorqueController : <hrpsys.rtm.RTcomponent instance at 0x3307b00> (315.1.8)
[hrpsys.py] create CompSvc -> TorqueController Service : <OpenHRP._objref_TorqueControllerService instance at 0x3306518>
[hrpsys.py]   eval : [self.te, self.te_svc, self.te_version] = self.createComp("ThermoEstimator","te")
[hrpsys.py] create Comp -> ThermoEstimator : <hrpsys.rtm.RTcomponent instance at 0x3370638> (315.1.8)
("can't find a service named", 'service0')
[hrpsys.py]   eval : [self.hes, self.hes_svc, self.hes_version] = self.createComp("EmergencyStopper","hes")
[hrpsys.py] ?[31mFail to createComps CORBA.UNKNOWN(omniORB.UNKNOWN_UserException, CORBA.COMPLETED_MAYBE)?[0m
[hrpsys.py]   eval : [self.el, self.el_svc, self.el_version] = self.createComp("SoftErrorLimiter","el")
[hrpsys.py] create Comp -> SoftErrorLimiter : <hrpsys.rtm.RTcomponent instance at 0x33714d0> (315.1.8)
[hrpsys.py] create CompSvc -> SoftErrorLimiter Service : <OpenHRP._objref_SoftErrorLimiterService instance at 0x336ed88>
[hrpsys.py]   eval : [self.tl, self.tl_svc, self.tl_version] = self.createComp("ThermoLimiter","tl")
[hrpsys.py] create Comp -> ThermoLimiter : <hrpsys.rtm.RTcomponent instance at 0x3370560> (315.1.8)
("can't find a service named", 'service0')
[hrpsys.py]   eval : [self.bp, self.bp_svc, self.bp_version] = self.createComp("Beeper","bp")
[hrpsys.py] ?[31mFail to createComps CORBA.UNKNOWN(omniORB.UNKNOWN_UserException, CORBA.COMPLETED_MAYBE)?[0m
[hrpsys.py]   eval : [self.log, self.log_svc, self.log_version] = self.createComp("DataLogger","log")
[hrpsys.py] create Comp -> DataLogger : <hrpsys.rtm.RTcomponent instance at 0x3306200> (315.1.8)
[hrpsys.py] create CompSvc -> DataLogger Service : <OpenHRP._objref_DataLoggerService instance at 0x3302cb0>
```

mmurooka, 2016-06-06:
ログを見るとこのテストのhrpsysのバージョンは315.1.8のようですが，
ReferenceForceUpdaterやその他生成に失敗しているrtcは，このバージョンには無いためにエラーになっているようです．

https://github.com/rkoyama1623/hrpsys-base/blob/rfu_support_both_arms_pr/sample/SampleRobot/samplerobot_reference_force_updater.py#L79-L80
でバージョンチェック分岐の前で`init()`を呼んでいて，
この`init()`のなかで存在しないrtcを作ろうとしています．
`init()`もバージョンチェックの中にいれれば解決するはずです．

ついでに，https://github.com/fkanehiro/hrpsys-base/pull/1005/commits/73ff65d5605dc41fb17c889b821c95051537620d の変更は，
他の部分と空行や空白の統一性がないので，
それも直しておくといいと思います．

rkoyama1623, 2016-06-07:
ご指摘の通り、hrpsysのバージョンの問題でした。
バージョンチェックのif文を入れて対応できました。
https://github.com/fkanehiro/hrpsys-base/pull/1005/files#diff-c06d1e7d899277cafd6fd01b4efee81eR23
ありがとうございました。

空白・空行の方も直しました。

closed 2016-06-07 by rkoyama1623