fkanehiro/hrpsys-base #1019: 14版のテストがおとおりずらくないですかね？

k-okada, 2016-06-30:


k-okada, 2016-06-30:
https://github.com/start-jsk/rtmros_common/pull/905
のおかげで通らなくなってしまった　@orikuma

orikuma, 2016-06-30:
start-jsk/rtmros_common#905 はHrpsysSeqStateROSBridgeのodomがpublishされる周期をtestするものですが, いくつか疑問点が有ります.
1. testされている値について
odomのtfをtestするhztest_tfは50±20Hzでtestしているはずで, 500±200Hzはhztest_jsのはずですが, hztest_jsが見当たらないようです. launchが古いということはありうるでしょうか?
参考: https://github.com/start-jsk/rtmros_common/pull/905/commits/0a29d1cb96fe3df3573474039b027ed188c99b53
2. HrpsysSeqStateROSBridgeについて
そもそもHrpsysSeqStateROSBridgeが500Hzで回ってないように見えます.
他のrtcの処理が重いという可能性はありそうでしょうか?

snozawa, 2016-07-01:
14番は普段から通り辛い気がしてましたが、
start-jsk/rtmros_common#905
のmergeの日付をみると`on 2 Feb`ですね。
通らなくなってから、何かいもrestartすると通ってるようです。

k-okada, 2016-07-10:
https://s3.amazonaws.com/archive.travis-ci.org/jobs/141171806/log.txt
を見ると，testファイルは
https://raw.githubusercontent.com/start-jsk/rtmros_common/1.3.1/hrpsys_ros_bridge/test/test-samplerobot.test
なので，

> odomのtfをtestするhztest_tfは50±20Hzでtestしているはずで, 500±200Hzはhztest_jsのはずですが, hztest_jsが見当たらないようです. launchが古いということはありうるでしょうか?

なので，testが古いというよりはhydroのバージョンで＋1.3.1のtest-samplerobot.test をつかっているということになります．なので，
https://github.com/start-jsk/rtmros_common/pull/905
は関係ないか．

k-okada, 2016-07-10:
それにしても通らなすぎに思えるけど．．．