fkanehiro/hrpsys-base #1034: リレーのONOFFに関して扱うコード

snozawa, 2016-08-25:
リレーのONOFFを扱うコードはありますでしょうか。

具体的には、サーボONしてロボットを稼働していて、
リレーがOFFして切ったら、サーボもOFFしてほしい、というものです。
そうすると、万一手順を誤ってリレーONしてしまっても、すでにサーボがOFFなので
危険でないというフェイルセーフになります。

こういった機能はすでにRobotHardwareか何かにありますでしょうか。
また、iobかなにかで、リレーのreadに相当するようなものはありますでしょうか。

よろしくお願いいたします。

fkanehiro, 2016-08-25:
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/io/iob.h#L179-L207
のread_power_state, read_power_command, write_power_commandが関連するAPIとなります。
RobotHardwareでは、リレーOFFを検知してサーボをOFFにするような機能は入っていません。

大抵の場合はサーボON中にリレーOFFにすると、角度偏差過大となりサーボOFFになります。

角度偏差過大でちゃんとサーボOFFになるか？というのを確認したり、時々リレーがOFFでもサーボONにしたい場合があるため、リレーOFF＝サーボOFFとはしていません。

snozawa, 2016-08-25:
早速のご回答ありがとうございます。
電気的にONな状態を表すのに
- 機械スイッチ（リレー）によるON/OFF
- ソフトウェア的なものによるON/OFF

があると思っており、電気的にONとなるのは上記２つがONの場合のみ、と思ってます。
read_power_command等の関数は後者ということでよろしいでしょうか。

今回知りたいのは、前者の機械スイッチのON/OFFを取得したいといったものでした。
これはもしかしたらロボットごとにことなるiobの実装次第なきがしますが、
read_digital_input的なものを利用するのでしょうか。

> 大抵の場合はサーボON中にリレーOFFにすると、角度偏差過大となりサーボOFFになります。
> 角度偏差過大でちゃんとサーボOFFになるか？というのを確認したり、時々リレーがOFFでもサーボONにしたい場合があるため、リレーOFF＝サーボOFFとはしていません。

こちらもありがとうごじます。
おっしゃるとおり、角度偏差が過大なときにはサーボがOFFになります。
一方で、サーボONの状態でリレーON=>リレーOFFになり、
「角度偏差が過大」というしきい値未満ではあるけど
万一リレーONにしてしまうと暴走したりヒューズが飛ぶケースがこちらで有りました。
なので、こちらとしてはリレーOFFになった場合はサーボOFFにしたいケースがあります。

そうしますと、RobotHardwareの中でリレーOFF＝＞サーボOFFにという機能をつくり、
> 角度偏差過大でちゃんとサーボOFFになるか？というのを確認したり、時々リレーがOFFでもサーボONにしたい場合があるため、リレーOFF＝サーボOFFとはしていません。

互換性のためデフォルトOFFににしておくのが良いでしょうか。
それとも、サーボOFF用のデータポートをRobotHardwareにつくり、外のRTCでリレーOFF=>サーボOFFとなる
判定をしてデータポートを出力する、Robothadwareはデータポートが来た時にサーボOFFする、
といった形のほうがよろしいでしょうか。

fkanehiro, 2016-08-25:
リレーOFFでサーボONにしたいのは特殊なケースなので、デフォルトONでOFFにもできるようになっていれば良いような気がします。
その場合には機械スイッチのON/OFFを知る必要はなく、機械スイッチとソフトスイッチのANDが分かれば十分のように思いますがどうでしょうか。

snozawa, 2016-08-25:
> その場合には機械スイッチのON/OFFを知る必要はなく、機械スイッチとソフトスイッチのANDが分かれば十分のように思いますがどうでしょうか。

はい、そのように思います。
その場合、機械スイッチとソフトスイッチのANDに関する情報は、何になりますでしょうか。
（powerはソフトスイッチ単体の２値フラグと理解しています）

fkanehiro, 2016-08-25:
read_power_stateがそれのつもりです。実際にモータドライバにモータ駆動電源が供給されているか否か（分かればですが）、を返すようにiobを実装します。

snozawa, 2016-08-25:
なるほど、了解です。
実はこちらのあるクローズドなロボット（iobもクローズド）では
read_power_stateが「機械スイッチとソフトスイッチのAND」に実装されておらず、
つねに０を返す仕様になってました。
別途「機械スイッチとソフトスイッチのAND」の値を取得するIOはありそうですので、
これをread_power_stateの中で呼びながらなにかするようにしてみようかと思います。

ところで、IOに関係してread_digital_input関数がRobotHardwareの中では
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/robot.cpp#L740
でよばれており、これは
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/RobotHardwareService_impl.cpp#L144
のサービスポートでのみ呼ばれています。
read_digital_input関数を、他のread_xx系関数と同様にonExecuteの中でかならずよばれるようにしつつ、
サービスポートのreadDigitalInputではすでにread_digital_inputで取得されたメンバ変数かなにかを参照するようにしても、問題ないでしょうか。
（ところで、サービスポートのようなread_xx系の関数は実時間スレッドと別なところからよんでても大丈夫なのでしょうか？）

fkanehiro, 2016-08-25:
特に問題ないように思います。
別スレッドから読んでも大丈夫かどうかはread_xxの中身で排他制御をしているか否か次第かと。
一度RobotHardwareに吸い上げる場合は、RobotHardware内で排他制御が必要になりますね。

snozawa, 2016-08-25:
わかりました。
こちらの方向で一旦PRを作ってみます。

snozawa, 2016-08-31:
度々失礼します。

https://github.com/fkanehiro/hrpsys-base/blob/master/lib/io/iob.h#L8-L9
でON=1, OFF=0とされており、read_servo_stateもこれに準拠していると思います
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/RobotHardware/robot.cpp#L571-L572

read_power_stateも同様に、ON/OFFの２値をかえして、
ONのときに１、OFFのときに０を返すような仕様になってますでしょうか。
よろしくお願いします。

fkanehiro, 2016-08-31:
はい、そうですね。

snozawa, 2016-08-31:
ありがとうございます

snozawa, 2016-09-26:
こちら一段落しましたので、一旦closeします。
ありがとうございました。

closed 2016-09-26 by snozawa