fkanehiro/hrpsys-base #1071: Impedanceが入った状態でpotentio-vectorをangle-vectorに送ると振動が起きる

ban-masa, 2016-11-15:
HRP2で腕にimpedance制御を入れた状態で*ri*のpotentio-vectorを*robot*のangle-vectorに入れて、
その後*ri*のangle-vectorに*robot*のangle-vectorを送ると（riのpotentio-vectorをriのangle-vectorに送るのと同義）、HRP2の腕が一瞬ガタッと振動するという現象を確認しました。

正確には以下のプログラムで上述の現象が発生しました。
https://gist.github.com/ban-masa/d671498cad38e5d5c3351e6996a6169a

時系列としましては、
reset-poseから足を上げたポーズに推移した後にpotentio-vectorを*ri*のangle-vectorに送ったタイミングで１回めの振動発生、
足を上げたポーズからreset-poseに推移するために*ri*のangle-vectorにreset-poseを送ったタイミングで２回めの振動発生
となっております。

impedance制御を腕に入れない状態ではこの振動は発生せず、またreset-poseの状態でpotentio-vectorを*ri*のangle-vectorに送ってもこの振動は発生しませんでした。
hrpsys-simulatorで再現するかどうかの検証も行いましたが、再現しないようでした。

rtcdのログ(log.txt)の該当部分（start impedance以降）を調べましたが
```
537:672:[el] velocity limit over RARM_JOINT3(21), qvel=2.23776, lvlimit =-2.23759, uvlimit =2.23759, servo_state = ON
538:673:[el] velocity limit over LARM_JOINT3(29), qvel=-2.23776, lvlimit =-2.23759, uvlimit =2.23759, servo_state = ON
539:674:[el] velocity limit over LARM_JOINT3(29), qvel=-2.24161, lvlimit =-2.23759, uvlimit =2.23759, servo_state = ON
540:675:[el] velocity limit over RARM_JOINT3(21), qvel=2.23776, lvlimit =-2.23759, uvlimit =2.23759, servo_state = ON
541:676:[el] velocity limit over LARM_JOINT3(29), qvel=-2.23776, lvlimit =-2.23759, uvlimit =2.23759, servo_state = ON
550:685:[el] velocity limit over RARM_JOINT3(21), qvel=-2.23776, lvlimit =-2.23759, uvlimit =2.23759, servo_state = ON
```
という出力しかありませんでした。
[log.txt](https://github.com/fkanehiro/hrpsys-base/files/591506/log.txt)

YoheiKakiuchi, 2016-11-15:
エラー報告としては、hrpsysのログがあるともっとよいですね。
~~~
(send *ri* :set-log-maxlength (* 300 250)) ;; hrpsysの周期(250Hz) * データをとりたい秒数 (300秒)
(send *ri* :start-log)
;;; ここで エラーが発生する動作をさせる
(send *ri* :save-log "/tmp" :make-directory t) ;; エラーが発生したらセーブ
;; make-directory t すると /tmp/xxxxx の下にログファイルができるのでそれをまとめて、googledriveなどに上げる
~~~

という手順です。

ban-masa, 2016-11-17:
hrpsysのログを取りました
[https://drive.google.com/drive/folders/0B-LWtxVDNqCOWW5yOTNxWWdQeDQ?usp=sharing](url)
[https://drive.google.com/drive/folders/0B-LWtxVDNqCOY1J0N2VjOVotZEE?usp=sharing](url)
上がすでに述べたriのpotentio-vectorをangle-vectorに入れた際のログで、
下が少し異なる足の姿勢を作ってrobotのangle-vectorをriのangle-vectorに送ったものです。
実際に実行したeuslispは各ログディレクトリのreadme.lに記載してあります。

下の方ではriのpotentio-vectorをriのangle-vectorに送っていないにも関わらず、振動が起きました。
関節を微小距離動かすことよりも、足をあげる姿勢をとること自体に問題があるのかもしれないです。

ログの関節角比率や関節速度比率を見てみましたが、関節角の限界に到達したりはしていないようでした。

ban-masa, 2016-11-17:
リンクが正しく貼れていなかったようなので修正しました

ban-masa, 2016-11-17:
rtcがstate-holder->emergency-stopper->impedance-controllerとつながっているので
２つのうちの上のログについてstate-holderのqOutとimpedance-controllerの出力qを比較しました。
```
(send *ri* :angle-vector (send *ri* :state :potentio-vector) 3000)
```
のタイミングでstate-holderのqOutが緩やかに変化しているのに対し、impedance-controllerのqはかなり尖った出力になっているようです。

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ban-masa, 2016-11-18:
hrpsys-simulatorでは再現しないと書きましたが、もう一度ログを取ってちゃんと調べたところ再現しました。
また、その際にhrpsys-simulatorが以下のようなエラーを出力し続けて固まりました。
```
[ic, JointPathEx] ERROR nan/inf is found
```
その時のhrpsysのログは以下になります。rarmのjoint-angleの指令値のicの前後での変化もグラフにして置いてあります。

https://drive.google.com/drive/folders/0B-LWtxVDNqCObG9SaUdNc1FFNlk?usp=sharing

やはりimpedance-controllerが原因のようです

snozawa, 2016-11-21:
ところで、
```
(send *ri* :set-impedance-controller-param :arms :use-sh-base-rpy t)
```
としてみて同様の際限コードを走らせるとどうなるかな？

ban-masa, 2016-11-21:
```
(send *ri* :set-impedance-controller-param :arms :use-sh-base-pos-rpy t)
```
とした後にやってみると再現しなくなりました。
振動が起きたり、hrpsys-simulatorが固まることもなく足をあげた姿勢をとれました。

snozawa, 2016-11-21:
お、なるほど、了解です。
今はImpedanceController指令値情報として、
- StateHolderからのbasePos, rotを利用するか
- ２本足で立ってるとして、fix-leg-to-coords相当のことをして地面を足につけるか

の２択があって、デフォルトは後者になってますが、そこがあやしかったですね。
取り急ぎ前者で動かして見てください。
ただ、現状この場合、ref force指令値をいれると微妙に意図した向きからずれるかもしれない。

ban-masa, 2016-11-21:
すみません。
現状自分のプログラムでは振動が起きるような動作は回避するようにプログラムを書いているので、至急必要というわけではないです。
ただ、実機でも
```(send *ri* :set-impedance-controller-param :arms :use-sh-base-pos-rpy t)```
で振動が起きなくなるかどうかは検証してみようと思います。

ban-masa, 2016-11-21:
上記の対処で実機でも振動は起きなくなりました。

ban-masa, 2017-02-02:
```(send *ri* :set-impedance-controller-param :arms :use-sh-base-pos-rpy t)```で振動は起きなくなることまでは判明したので、一度閉じます

closed 2017-02-02 by ban-masa