fkanehiro/hrpsys-base #1078: Building 315.10.1 failure on QNX

130s, 2016-12-06:
315.10.1 on QNX が成功している方がいればご存知のことを教えて頂けると助かります．

以下，ビルド用スクリプト (独自で作っているものですみませんが) の出力結果です．

```
cd /tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger && /usr/qnx650/host/qnx6/x86/usr/bin/QCC  -x c++ -DDataLogger_EXPORTS -DROBOT_IOB_VERSION=2 -I/usr/pkg/include -O3 -DNDEBUG -I/tmp/hrpsys-source-315.10.1-2016-12-05/build -I/opt/jsk/include -I/usr/pkg/include -I/opt/jsk/include/coil-1.1 -I/opt/jsk/include/openrtm-1.1 -I/opt/jsk/include/openrtm-1.1/rtm/idl -I/opt/jsk/include/OpenHRP-3.1 -I/opt/jsk/include/eigen3   -DHRPSYS_PACKAGE_VERSION="\"315.10.1\"" -o CMakeFiles/DataLogger.dir/DataLogger.o -c /tmp/hrpsys-source-315.10.1-2016-12-05/src/hrpsys-base-315.10.1/rtc/DataLogger/DataLogger.cpp
In file included from /tmp/hrpsys-source-315.10.1-2016-12-05/src/hrpsys-base-315.10.1/rtc/DataLogger/DataLogger.cpp:10:
/tmp/hrpsys-source-315.10.1-2016-12-05/build/hrpsys/util/Hrpsys.h:22: error: 'std::localtime' has not been declared
/tmp/hrpsys-source-315.10.1-2016-12-05/build/hrpsys/util/Hrpsys.h:28: error: 'std::toupper' has not been declared
cc: /usr/qnx650/host/qnx6/x86/usr/lib/gcc/i486-pc-nto-qnx6.5.0/4.4.2/cc1plus error 1
make[2]: *** [rtc/DataLogger/CMakeFiles/DataLogger.dir/DataLogger.o] Error 1
make[2]: Leaving directory `/tmp/hrpsys-source-315.10.1-2016-12-05/build'
make[1]: *** [rtc/DataLogger/CMakeFiles/DataLogger.dir/all] Error 2
make[1]: Leaving directory `/tmp/hrpsys-source-315.10.1-2016-12-05/build'
make: *** [all] Error 2
```

もうちょっと長いログ

```
ubuntu$ ./compile-hrpsys-qnx.sh 192.168.149.128 barstow 315.10.1
:
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/SoftErrorLimiter.cpp
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/SoftErrorLimiter.h
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/SoftErrorLimiter.txt
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/SoftErrorLimiterComp.cpp
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/SoftErrorLimiterService_impl.cpp
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/SoftErrorLimiterService_impl.h
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/beep.cpp
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/beep.h
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/robot.cpp
  inflating: hrpsys-base-315.10.1/rtc/SoftErrorLimiter/robot.h
   creating: hrpsys-base-315.10.1/rtc/Stabilizer/
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/CMakeLists.txt
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/Integrator.cpp
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/Integrator.h
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/Stabilizer.cpp
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/Stabilizer.h
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/Stabilizer.txt
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/StabilizerComp.cpp
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/StabilizerService_impl.cpp
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/StabilizerService_impl.h
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/TwoDofController.cpp
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/TwoDofController.h
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/ZMPDistributor.h
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/testTwoDofController.cpp
  inflating: hrpsys-base-315.10.1/rtc/Stabilizer/testZMPDistributor.cpp
   creating: hrpsys-base-315.10.1/rtc/StateHolder/
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/CMakeLists.txt
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolder.cpp
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolder.h
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolder.txt
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolder.xml
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolderComp.cpp
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolderService_impl.cpp
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/StateHolderService_impl.h
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/TimeKeeperService_impl.cpp
  inflating: hrpsys-base-315.10.1/rtc/StateHolder/TimeKeeperService_impl.h
   creating: hrpsys-base-315.10.1/rtc/ThermoEstimator/
  inflating: hrpsys-base-315.10.1/rtc/ThermoEstimator/CMakeLists.txt
  inflating: hrpsys-base-315.10.1/rtc/ThermoEstimator/MotorHeatParam.h
  inflating: hrpsys-base-315.10.1/rtc/ThermoEstimator/ThermoEstimator.cpp
  inflating: hrpsys-base-315.10.1/rtc/ThermoEstimator/ThermoEstimator.h
  inflating: hrpsys-base-315.10.1/rtc/ThermoEstimator/ThermoEstimator.txt
  inflating: hrpsys-base-315.10.1/rtc/ThermoEstimator/ThermoEstimatorComp.cpp
   creating: hrpsys-base-315.10.1/rtc/ThermoLimiter/
  inflating: hrpsys-base-315.10.1/rtc/ThermoLimiter/CMakeLists.txt
  inflating: hrpsys-base-315.10.1/rtc/ThermoLimiter/ThermoLimiter.cpp
  inflating: hrpsys-base-315.10.1/rtc/ThermoLimiter/ThermoLimiter.h
  inflating: hrpsys-base-315.10.1/rtc/ThermoLimiter/ThermoLimiter.txt
[... 1040 lines cut ...]
cd /tmp/hrpsys-source-315.10.1-2016-12-05/build && /usr/pkg/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/hrpsys-source-315.10.1-2016-12-05/src/hrpsys-base-315.10.1 /tmp/hrpsys-source-315.10.1-2016-12-05/src/hrpsys-base-315.10.1/rtc/DataLogger /tmp/hrpsys-source-315.10.1-2016-12-05/build /tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger /tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger/CMakeFiles/DataLogger.dir/DependInfo.cmake --color=
Dependee "/tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger/CMakeFiles/DataLogger.dir/DependInfo.cmake" is newer than depender "/tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger/CMakeFiles/DataLogger.dir/depend.internal".
Dependee "/tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger/CMakeFiles/DataLogger.dir/depend.internal".
Scanning dependencies of target DataLogger
make[2]: Leaving directory `/tmp/hrpsys-source-315.10.1-2016-12-05/build'
make -f rtc/DataLogger/CMakeFiles/DataLogger.dir/build.make rtc/DataLogger/CMakeFiles/DataLogger.dir/build
make[2]: Entering directory `/tmp/hrpsys-source-315.10.1-2016-12-05/build'
/usr/pkg/bin/cmake -E cmake_progress_report /tmp/hrpsys-source-315.10.1-2016-12-05/build/CMakeFiles 9
[ 38%] Building CXX object rtc/DataLogger/CMakeFiles/DataLogger.dir/DataLogger.o
cd /tmp/hrpsys-source-315.10.1-2016-12-05/build/rtc/DataLogger && /usr/qnx650/host/qnx6/x86/usr/bin/QCC  -x c++ -DDataLogger_EXPORTS -DROBOT_IOB_VERSION=2 -I/usr/pkg/include -O3 -DNDEBUG -I/tmp/hrpsys-source-315.10.1-2016-12-05/build -I/opt/jsk/include -I/usr/pkg/include -I/opt/jsk/include/coil-1.1 -I/opt/jsk/include/openrtm-1.1 -I/opt/jsk/include/openrtm-1.1/rtm/idl -I/opt/jsk/include/OpenHRP-3.1 -I/opt/jsk/include/eigen3   -DHRPSYS_PACKAGE_VERSION="\"315.10.1\"" -o CMakeFiles/DataLogger.dir/DataLogger.o -c /tmp/hrpsys-source-315.10.1-2016-12-05/src/hrpsys-base-315.10.1/rtc/DataLogger/DataLogger.cpp
In file included from /tmp/hrpsys-source-315.10.1-2016-12-05/src/hrpsys-base-315.10.1/rtc/DataLogger/DataLogger.cpp:10:
/tmp/hrpsys-source-315.10.1-2016-12-05/build/hrpsys/util/Hrpsys.h:22: error: 'std::localtime' has not been declared
/tmp/hrpsys-source-315.10.1-2016-12-05/build/hrpsys/util/Hrpsys.h:28: error: 'std::toupper' has not been declared
cc: /usr/qnx650/host/qnx6/x86/usr/lib/gcc/i486-pc-nto-qnx6.5.0/4.4.2/cc1plus error 1
make[2]: *** [rtc/DataLogger/CMakeFiles/DataLogger.dir/DataLogger.o] Error 1
make[2]: Leaving directory `/tmp/hrpsys-source-315.10.1-2016-12-05/build'
make[1]: *** [rtc/DataLogger/CMakeFiles/DataLogger.dir/all] Error 2
make[1]: Leaving directory `/tmp/hrpsys-source-315.10.1-2016-12-05/build'
make: *** [all] Error 2
Connection to 192.168.149.128 closed.
+ echo =====
=====
+ echo '$ tar -xvzf /tmp/hrpsys-315.10.1-qnx-2016-12-05; cd hrpsys-source-315.10.1-2016-12-05/build; make install'
$ tar -xvzf /tmp/hrpsys-315.10.1-qnx-2016-12-05; cd hrpsys-source-315.10.1-2016-12-05/build; make install
```

130s, 2016-12-06:
上記ビルド用スクリプトはこれまで`315.2.8`までしかコンパイル動作確認していませんでしたが，`315.8.0`，`315.9.0`はコンパイル自体は成功しているように見えます．

`315.10.x` は個人的には差し当たり不要になったので閉じます．

k-okada, 2016-12-18:
jenkinsのqnxのチェックがちゃんと動いていなかったので、コンパイル通らなかったらfailとするようにしました．
基本的には、https://github.com/fkanehiro/hrpsys-base/pull/1085/files みたいにstd関数を追加するんだとおもいます．using namespace stdを全部にかけてもいいかもしれませんが、QNXのコンパイルが必須な人は少ないので、
今はこの方法担っています．コンパイルがとおらなかったら、ここを参考になしましょう．

130s, 2016-12-21:
https://github.com/fkanehiro/hrpsys-base/pull/1085 を含む次期リリース (315.10.2?) が出たらそレを使って試します．315.10.1 の問題は解決されるわけではないので，close (won't fix か？) します．

closed 2016-12-21 by 130s

emijah, 2016-12-21:
これの事だったんですね。