fkanehiro/hrpsys-base #1079: Versioning is not correct in 315.9.0 and 315.8.0

130s, 2016-12-08:
バージョン確認に，https://github.com/fkanehiro/hrpsys-base/issues/423#issuecomment-67145190 で教わった次のコマンドを使っていますが，手元で作った`315.{9, 8}.0`のバイナリで次のようになります．

```
$ strings /opt/jsk/lib/RobotHardware.so | grep ^[0-9]*\\.[0-9]*\\.[0-9]*$
315.7.0
```

https://github.com/fkanehiro/hrpsys-base/blob/315.9.0/CMakeLists.txt#L107-L109 をみると確かに`315.7.0`となっています．

同様に [tag 315.8.0](https://github.com/fkanehiro/hrpsys-base/blob/315.8.0/CMakeLists.txt#L86) でも`315.7.0`のようです ([tag 315.10.0](https://github.com/fkanehiro/hrpsys-base/blob/315.10.0/CMakeLists.txt)では治っていそう)．

130s, 2016-12-08:
https://github.com/fkanehiro/hrpsys-base/pull/1080 の方法は失敗だったので非 admin としての解決を諦めました．

admin 権限を持つ方に tag 315.{8, 9}.0 を削除し，ソース変更後再度 tag して頂く，とかになるでしょうか．

130s, 2016-12-16:
> admin 権限を持つ方に tag 315.{8, 9}.0 を削除し，ソース変更後再度 tag して頂く，とかになるでしょうか．

- tag を作り直す事は問題無さそうです ([例](http://stackoverflow.com/questions/8044583/how-can-i-move-a-tag-on-a-git-branch-to-a-different-commit))
- 更新したコードを master ブランチ上で，次のバージョンに含まれるコミットより前に挿入したい気がしますが，その方法はよくわかってないです．[git-filter-branch](https://git-scm.com/docs/git-filter-branch) とかを使えば出来そうという話も聞きましたが，私はちょっと理解に時間がかかりそうなので手を付けるのを止めておきます．

k-okada, 2016-12-18:
遅くなりました．
https://gist.github.com/k-okada/ec2832b6a60cf14bd27ea224782ce870
https://gist.github.com/k-okada/a2d2b06824f6123ef79176b01c82b574
で変更できると思います．これで出来たbranchをgit push -f origin master したものを
https://github.com/k-okada/hrpsys-base/commits/master
におきました．8.0以降全てのSHAが変わっているので、これを採用すべきかは議論の余地はあるとはおもいますが、
まずは、git filter branch の例ということで．．．．