fkanehiro/hrpsys-base #1095: IIRFilterの初期値0以外にしたい時に使うIIRFilter::resetが効かない

ishiguroJSK, 2017-02-17:
標準のままだと0以外の値で`filter.reset(999.9);`などしても0.0からなめらかに目標値に行ってしまうようなのですが，
```
void IIRFilter::reset(double initial_input)
{
    m_previous_values.assign(m_dimension, initial_input);
}
```
における「m_previous_values」は正しく所望の初期値に変更できているのは確認できています．
なので，
```
double IIRFilter::passFilter(double input)
{
    if (! m_initialized) {
        return 0.0;
    }
    double feedback, filtered;
    // calcurate retval
    feedback = m_fb_coefficients[0] * input;
    for (int i = 0; i < m_dimension; i++) {
        feedback += m_fb_coefficients[i + 1] * m_previous_values[i];
    }
    filtered = m_ff_coefficients[0] * feedback;
    for(int i = 0; i < m_dimension; i++) {
        filtered += m_ff_coefficients[i + 1] * m_previous_values[i];
    }
    // update previous values
    m_previous_values.push_front(feedback);
    m_previous_values.pop_back();
    return filtered;
}
```
の部分を見ていたのですが，じっくり読んでも[教科書的なIIRの式](https://ja.wikipedia.org/wiki/%E7%84%A1%E9%99%90%E3%82%A4%E3%83%B3%E3%83%91%E3%83%AB%E3%82%B9%E5%BF%9C%E7%AD%94)と対応付けが出来なくてお手上げです．
これは中間的な変数を必要としないスマートな実装ということなのでしょうか？

一応自力で[教科書的な実装](https://github.com/fkanehiro/hrpsys-base/commit/7785afd54f66221447d88d036c5f3d3f283c4dcf)をしてIIRFilter::resetが正しく動作するのは確認しました．

YoheiKakiuchi, 2017-02-18:
@orikuma 
PR前の実装でinitial_valueを０以外にする方法はあるのかな？
ないなら、 @ishiguroJSK の実装でいいように思います。

orikuma, 2017-02-20:
遅くなりました.
結論から言うと, 以前の実装のreset関数が間違っています.
正しく直せば @ishiguroJSK さんのものと同じ結果が得られますが,
可読性の観点から @ishiguroJSK の実装のほうが良さそうかなと思っています.
(以前の実装のメリットは多分バッファが一つ減るくらいのはずです)

--以下詳細--
以前の私の実装は, 以下の文献で言うDirectForm-IIの実装になっています.
@ishiguroJSK さんの実装は多分DirectForm-Iの実装で, そうだとすれば内容的には等価です.
http://www.eas.uccs.edu/~mwickert/ece2610/lecture_notes/ece2610_chap8.pdf
DirectForm-IIの実装の場合, m_previous_valueに保存するべき値はwに相当する値であり,
出力yとは異なります.
以前の実装ではここでyを保存してしまっていたので, reset時の値が0でないと正しい値が得られませんでした.
ここの値をyがinitial_inputに収束するときのw(feedback)の値になるように修正してやればよいです.
(修正したものが下記)
https://github.com/fkanehiro/hrpsys-base/compare/master...orikuma:fix-iir-filter-reset

ishiguroJSK, 2017-02-21:
@orikuma さんありがとうございます．
オフラインでちらっと話に上がったように，コメントに参考URLを載せたプルリクをお願いします．

ishiguroJSK, 2017-03-11:
マージされたのでcloseします．

closed 2017-03-11 by ishiguroJSK