fkanehiro/hrpsys-base #1098: STが切れない時がある

YoheiKakiuchi, 2017-02-24:
sverviceでstopStabilizerを呼んだ時に、
```transition_time == 0``` でないと、
STが切れないようになっていると思います。
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/Stabilizer/Stabilizer.cpp#L1744

この場合、足が離れて、transitionが終わるまでの間にstopStabilizerを送ると、
なにもせずに返って来てSTが入ったままになります。
パラメータでIDLEになったかどうかは確認できるので、
stopStabilizerし、IDLEになるまでstopStabilizerし続けるというアプリケーション側の回避方法はあります。

また、まれなケースですが、足がついたり離れたりする(airと判定されるかされないかギリギリの接触状態になる）
ときは、airとstのtransitionを繰り返してしまいstopStabilizerが受け付けられるタイミングが
ほぼないという状況にもなります。

stopStabilizerの ```transition_time == 0``` は必要でしょうか？

idlを変更することになりますが、
```transition_time != 0``` の時に、```waitSTTransition()``` をしてstopできるようになるまで待つようなwait_transitionフラグをサービスにつけるというのはどうでしょうか？

snozawa, 2017-02-27:
> この場合、足が離れて、transitionが終わるまでの間にstopStabilizerを送ると、
> なにもせずに返って来てSTが入ったままになります。

こちらの症状ですが、正しくは
- 体内のst.rtcでは、（足が離れたときの空中判定に起因する）停止処理は続いている。そのため、ちょっとまてばMODE_IDLE状態になる
- eusなどの外部のプログラムから、通常のstop-stのときは「停止処理の指令＋停止するまでwait」となってるのに対し、今回の場合は「停止処理とwaitがスルーされてかえってくる」ため、eusなどではstop-stをおくった直後にはMODE_IDLEでない場合がある

といったことでしょうか。
また、

> transition_time != 0 の時に、waitSTTransition() をしてstopできるようになるまで待つようなwait_transitionフラグをサービスにつけるというのはどうでしょうか？

とあわせて、st側の解決策として、waitTransition関数をif文から取り出し、「停止処理」をかけるかどうかはif文でチェックするけど、「waitする」は常に行う（他の誰かが既に停止してても、必ずまつ）ようにすることで、
常にstop-st後ではMODE_IDLEとなることが保証されそうなきがするのですが、
こんな感じの解決策で良いでしょうか。

> また、まれなケースですが、足がついたり離れたりする(airと判定されるかされないかギリギリの接触状態になる）
> ときは、airとstのtransitionを繰り返してしまいstopStabilizerが受け付けられるタイミングが
> ほぼないという状況にもなります。

これは転倒中でしょうか。それとももう少しイージーなケースでも再現するでしょうか。

> stopStabilizerの transition_time == 0 は必要でしょうか？

停止処理のための補間みたいなことをしているので、単純にこのif文から該当部分をぬくのは難しいかと思います。

YoheiKakiuchi, 2017-02-27:
> そのため、ちょっとまてばMODE_IDLE状態になる
ここは、MODE_IDLEではなくてMODE_AIRになっています。

MODE_STの時に足が離れると、判定されたら即座にMODE_AIRに（パラメータ的には）なって、
遷移中の間(デフォルトでは2秒) stop/startが受け付けられなくなります。

その後、遷移が終わっても、MODE_AIRのままです。
（stopは何も処理されずに返ってきているため）

> とあわせて、st側の解決策として、waitTransition関数をif文から取り出し、「停止処理」をかけるかどうかはif文でチェックするけど、「waitする」は常に行う（他の誰かが既に停止してても、必ずまつ）ようにすることで、
> 常にstop-st後ではMODE_IDLEとなることが保証されそうなきがするのですが、こんな感じの解決策で良いでしょうか。

解決策としては、これでいいように思います。MODE_IDLEとなることが保証されるので。

snozawa, 2017-03-09:
@YoheiKakiuchi さん
こちらOKそうでしたら、closeをお願いします。

closed 2017-03-09 by YoheiKakiuchi