fkanehiro/hrpsys-base #1112: CMakeLists.txtの https://github.com/fkanehiro/hrpsys-base/commit/e335e24f614d799214ae92d363e919456bbd0cbb のcommit以降ROS_PACKAGE_PATHにバックスラッシュが入る

mmurooka, 2017-03-10:
https://github.com/fkanehiro/hrpsys-base/commit/e335e24f614d799214ae92d363e919456bbd0cbb のcommit以降、hrpsys自体のbuildは通るのですが、ROS_PACKAGE_PATHにバックスラッシュが紛れ込んで、他のパッケージのbuildが通らなくなっているようです。

```bash
[http://hrp2017v:10017][192.168.96.184] leus@callisto:~$ source ~/ros/indigo_parent/devel/setup.bash 
[http://hrp2017v:10017][192.168.96.184] leus@callisto:~$ echo $ROS_PACKAGE_PATH
/home/leus/ros/indigo_parent/src/jsk-ros-pkg/geneus:/home/leus/ros/indigo_parent/src/euslisp/Euslisp:/home/leus/ros/indigo_parent/src/euslisp/jskeus:/home/leus/ros/indigo_parent/src/openrtm_aist:/home/leus/ros/indigo_parent/src/openhrp3:/home/leus/ros/indigo_parent/src/hrpsys:/home/leus/ros/indigo_parent/src/openrtm_aist_python:/home/leus/ros/indigo_parent/src/rtctree:/home/leus/ros/indigo_parent/src/rtshell:/home/leus/ros/indigo_parent/src/rtsprofile:/home/leus/ros/indigo_parent/devel/\:/home/leus/ros/indigo_parent/devel/share/hrpsys:/opt/ros/indigo/share:/opt/ros/indigo/stacks
[http://hrp2017v:10017][192.168.96.184] leus@callisto:~$ echo $ROS_PACKAGE_PATH | sed -e "s/:/\n/g"
/home/leus/ros/indigo_parent/src/jsk-ros-pkg/geneus
/home/leus/ros/indigo_parent/src/euslisp/Euslisp
/home/leus/ros/indigo_parent/src/euslisp/jskeus
/home/leus/ros/indigo_parent/src/openrtm_aist
/home/leus/ros/indigo_parent/src/openhrp3
/home/leus/ros/indigo_parent/src/hrpsys
/home/leus/ros/indigo_parent/src/openrtm_aist_python
/home/leus/ros/indigo_parent/src/rtctree
/home/leus/ros/indigo_parent/src/rtshell
/home/leus/ros/indigo_parent/src/rtsprofile
/home/leus/ros/indigo_parent/devel/\
/home/leus/ros/indigo_parent/devel/share/hrpsys
/opt/ros/indigo/share
/opt/ros/indigo/stacks
```
の中で
```
/home/leus/ros/indigo_parent/devel/\
/home/leus/ros/indigo_parent/devel/share/hrpsys
```
とあるように、
hrpsysのひとつ前のパッケージの最後にバックスラッシュが入るようになりました。

おそらくこれが原因で、
例えば、jsk_perceptionのcatkin buildで以下のようなエラーが生じます。
エラーの中で
```
File name too long
```
と言われていて、これはROS_PACKAGE_PATHの中に紛れ込んだバックスラッシュによって、コロンがファイル名の一部とみなされてしまったのではないかと予想しています。

hrpsysの該当commitを戻すと、
ROS_PACKAGE_PATHにバックスラッシュが入らなくなり、jsk_perceptionのbuildもできるようになりました。

以下、jsk_perceptionをbuildしたときのエラーです。
```
leus@callisto:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception$ catkin bt --force-cmake --start-with-this
==> Expanding alias 'bt' from 'catkin bt --force-cmake --start-with-this' to 'catkin b --this --force-cmake --start-with-this'
==> Expanding alias 'b' from 'catkin b --this --force-cmake --start-with-this' to 'catkin build --this --force-cmake --start-with-this'
-------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/leus/ros/indigo_parent/devel:/opt/ros/indigo
Workspace:                   /home/leus/ros/indigo
-------------------------------------------------------------------------------
Source Space:       [exists] /home/leus/ros/indigo/src
Log Space:          [exists] /home/leus/ros/indigo/logs
Build Space:        [exists] /home/leus/ros/indigo/build
Devel Space:        [exists] /home/leus/ros/indigo/devel
Install Space:      [unused] /home/leus/ros/indigo/install
DESTDIR:            [unused] None
-------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------
[... 27 lines cut ...]
________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << jsk_perception:cmake /home/leus/ros/indigo/logs/jsk_perception/build.cmake.004.log
OPENCV 2.4.8 FOUND
OPENMP FOUND
CMake Warning at /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception/CMakeLists.txt:119 (message):
  DEPRECATION WARNING: srv files in jsk_pcl_ros are deprecated.  Please use
  those in jsk_recognition_msgs.

CMake Warning (dev) at CMakeLists.txt:253 (message):
  robot_self_filter is found.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:348 (message):
  Node/Nodelets without rostest:
  node_scripts/non_maximum_suppression.py;node_scripts/apply_context_to_label_probability;node_scripts/matchtemplate.py;node_scripts/fisheye_ray.py;node_scripts/binpack_rect_array.py;node_scripts/image_time_diff.py;node_scripts/random_forest_server.py;node_scripts/solidity_rag_merge.py;node_scripts/unwrap_histogram_with_range_array.py;node_scripts/fast_rcnn.py;src/polygon_array_color_likelihood.cpp;src/polygon_array_color_histogram.cpp;src/sparse_image_encoder.cpp;src/sparse_image_decoder.cpp;src/background_substraction_nodelet.cpp;src/fisheye_to_panorama.cpp;src/grabcut_nodelet.cpp;src/rgb_decomposer.cpp;src/hsv_decomposer.cpp;src/lab_decomposer.cpp;src/ycc_decomposer.cpp;src/contour_finder.cpp;src/snake_segmentation.cpp;src/color_histogram_label_match.cpp;src/unapply_mask_image.cpp;src/single_channel_histogram.cpp;src/morphological_operator.cpp;src/morphological_operator.cpp;src/morphological_operator.cpp;src/morphological_operator.cpp;src/morphological_operator.cpp;src/morphological_operator.cpp;src/morphological_operator.cpp;src/rect_to_mask_image.cpp;src/mask_image_to_roi.cpp;src/roi_to_rect.cpp;src/roi_to_mask_image.cpp;src/polygon_to_mask_image.cpp;src/polygon_array_to_label_image.cpp;src/rect_to_roi.cpp;src/project_image_point.cpp;src/sliding_window_object_detector.cpp;src/gaussian_blur.cpp;src/tabletop_color_difference_likelihood.cpp;src/saliency_map_generator_node.cpp;src/skeletonization_nodelet.cpp;src/robot_to_mask_image.cpp
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/leus/ros/indigo/build/jsk_perception; catkin build --get-env jsk_perception | catkin env -si  /usr/bin/cmake /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/leus/ros/indigo/devel/.private/jsk_perception -DCMAKE_INSTALL_PREFIX=/home/leus/ros/indigo/install; cd -
........................................................................................................................................................................................................................................................
________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << jsk_perception:make /home/leus/ros/indigo/logs/jsk_perception/build.make.014.log
/bin/sh: 1: ROS_PACKAGE_PATH=/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception:/home/leus/ros/indigo/src/multimaster_experimental/app_manager:/home/leus/ros/indigo/src/multimaster_experimental/app_manager_tutorial:/home/leus/ros/indigo/src/ar_tools/ar_tools:/home/leus/ros/indigo/src/arm_navigation_msgs:/home/leus/ros/indigo/src/ar_tools/artoolkit:/home/leus/ros/indigo/src/ar_tools/ar_pose:/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/assimp_devel:/home/leus/ros/indigo/src/ros-planning/moveit_robots/atlas_moveit_config:/home/leus/ros/indigo/src/ros-planning/moveit_robots/atlas_v3_moveit_config:/home/leus/ros/indigo/src/audio_common/audio_common:/home/leus/ros/indigo/src/audio_common/audio_common_msgs:/home/leus/ros/indigo/src/audio_common/audio_capture:/home/leus/ros/indigo/src/audio_common/audio_play:/home/leus/ros/indigo/src/ros-planning/moveit_robots/baxter/baxter_ikfast_left_arm_plugin:/home/leus/ros/indigo/src/ros-planning/moveit_robots/baxter/baxter_ikfast_right_arm_plugin:/home/leus/ros/indigo/src/ros-planning/moveit_robots/baxter/baxter_moveit_config:/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_3rdparty/bayesian_belief_networks:/home/leus/ros/indigo/src/jsk-ros-pkg/jsk-ros-pkg-unreleased/sandbox/cfs_sensor:/home/leus/ros/indigo/src/ros-planning/moveit_robots/clam_moveit_config:/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/collad [... cut ...] ros/indigo_parent/src/openrtm_aist_python:/home/leus/ros/indigo_parent/src/rtctree:/home/leus/ros/indigo_parent/src/rtshell:/home/leus/ros/indigo_parent/src/rtsprofile:/home/leus/ros/indigo_parent/devel/\:/home/leus/ros/indigo_parent/devel/share/hrpsys:/opt/ros/indigo/share:/opt/ros/indigo/stacks: File name too long
make[2]: *** [/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception/sample/rimokon-pose.launch] Error 2
make[1]: *** [CMakeFiles/eusmodel_template.dir/all] Error 2
```

k-okada, 2017-03-10:
あれ、
でも、戻すと今度はhttps://github.com/start-jsk/rtmros_common/pull/998 でトラブルわけだけど、どうしよう．

k-okada, 2017-03-10:
http://stackoverflow.com/questions/11593180/cmake-how-to-output-semicolon-as-command-options-in-add-custom-target
か．
https://cmake.org/pipermail/cmake/2002-May/002827.html
つければ両方行けるのか．

後は結構な変更になるけど、この部分はbuild_type cmakeなのにroscdさせたかったからだったと思うので、package.xmlのexport build_type を消してもいんじゃ無いかという気もしてきた．ダメかな？どうだろう．

snozawa, 2017-03-10:
build_typeなしだとどうなるんでしたでしょうか。
`$(find hrpsys)`
や
`pkg-config hrpsys-base --variable=prefix`
もbuild_typeが必要だったりしますか？
（これらはsamplerobot.launchやサンプルpyとかで使われ手相です）

mmurooka, 2017-03-10:
> http://stackoverflow.com/questions/11593180/cmake-how-to-output-semicolon-as-command-options-in-add-custom-target

の作戦でやる例は
https://github.com/fkanehiro/hrpsys-base/compare/master...mmurooka:fix-semicolon
のような感じでしょうか．

手元の環境で，ROS_PACKAGE_PATHにバックスラッシュが入る問題は起きなかったので，
あとは

> でも、戻すと今度はstart-jsk/rtmros_common#998 でトラブルわけだけど、どうしよう．

が大丈夫なことを確認できればいいのですが，
これは手元でも再現するはずでしょうか．新しいcmakeなどが必要でしょうか．

k-okada, 2017-03-11:
そうするとcatkinになるので、catkin_package()をcmakeに書く必要がある．とすると、
こんどはhrpsys-baseがcatkinに依存するので、それはよくないと．
でも、もしかするとcatkin_package()なしで、build_typeをなくするとなにになるかなあ、と．あるいは、find_pacage(catkin)
CATKIN_FOUNDでなにかするか．．．
何れにせよ、難しいですねぇ．．．

k-okada, 2017-03-11:
2017年3月11日 0:04 Masaki Murooka <notifications@github.com>:
>
> 手元の環境で，ROS_PACKAGE_PATHにバックスラッシュが入る問題は起きなかったので，
> あとは
>
> でも、戻すと今度はstart-jsk/rtmros_common#998 でトラブルわけだけど、どうしよう．
>
> が大丈夫なことを確認できればいいのですが，
> これは手元でも再現するはずでしょうか．新しいcmakeなどが必要でしょうか．

https://github.com/jsk-ros-pkg/jsk_travis/blob/ae5c2bdf6afdad7b21b21e98d09567340f1aca58/travis.sh#L104-L105

を実行したらエラー環境を再現できることになると思います．

--
◉ Kei Okada

snozawa, 2017-03-11:
> そうするとcatkinになるので、catkin_package()をcmakeに書く必要がある．とすると、 こんどはhrpsys-baseがcatkinに依存するので、それはよくないと． でも、もしかするとcatkin_package()なしで、build_typeをなくするとなにになるかなあ、と．あるいは、find_pacage(catkin) CATKIN_FOUNDでなにかするか．．． 何れにせよ、難しいですねぇ．．．

なるほど、find_package(catkin)できればcatkinで、そうでないならこれまでどおり、
が一番よさそうに感じましたが、いかがでしょうか。

k-okada, 2017-03-11:
それでうごけば、というのと、
hrpsys-base側から見たら、普通にcmake でうごくようにしていて、ROSに依存させたくないよね、
というのがあるだろうから、エラーがでるだけだからよいとしても、find_packge(catkin)を入れるのは
センスうたがわれないか、というところでしょうか．あとは、１６．０４にすると、いとせずにcatkinがはいるので、
cmake だけ使いたい人が困らないか、というのもチェック項目が．
このロジックがとおると、hrpsys-baseに .l のコードをばんばんいれておいて、EUSDIRがない人は
無視するから良いでしょ、という話も通用しちゃいますね．
とはいえ、すでにrostestいれちゃっているので、まぁよい、という話もある．

まぁ、そもそも考えるとすると今のパッケージのデザインが今一つでgazeboスタイルで、
- hrpsys は ROS とは関係なく動くようにする
- となると、テスト関係が問題になる．これはhrpsys/opnertmでうまい仕組みがあるとよい．
- gazebo_ros てきなかんじで、ROSパッケージ内で動くようにするのは、別のパッケージが管理すれば良い．
というのが良かった、ということでしょうか．とは言え、結局うまくテストは知らせる方法がないので、
いまのキメラでもしょうがいないかなぁ．と．

もちろん、hrpsysをROSのスタイルで開発できるようにする、という話もあるけど、世の中そんなに単純
じゃなくて、eusもあるし、もっと一般的な問題ではpipとrosとどうしたらいい、というのもあるので、
こういうところで苦労して経験値をためるのは良いことかな．と．

snozawa, 2017-03-11:
> find_packge(catkin)を入れるのは センスうたがわれないか

現状もfind_package(catkin)がはいってるので、
（根本としてそれどうなのという議論とは別に）今回のissueの問題を直すのには、
https://github.com/fkanehiro/hrpsys-base/issues/1112#issuecomment-285837064
の有無でセンス感はかわらない、と理解してたのですがあってますか？
（依存度がかわらないと理解してた、という意味）

> まぁ、そもそも考えるとすると今のパッケージのデザインが今一つでgazeboスタイルで、 - hrpsys は ROS とは関係なく動くようにする - となると、テスト関係が問題になる．これはhrpsys/opnertmでうまい仕組みがあるとよい． - gazebo_ros てきなかんじで、ROSパッケージ内で動くようにするのは、別のパッケージが管理すれば良い． [...]

なるほど、やはりけっこう難しい問題ですね。
gazebo_rosは、向かしのstart-jsk/hrpsysと同じような形式なんでしたっけ？
現状は、start-jsk/hrpsysの方法、hrpsys-baseで現状やっている方法、jsk_3rdpartyのような方法などがありそうですが、
どれがオススメなんでしょうか（実はjsk_3rdpartyがどういうスタイルか僕はわかってない気がします）。
また上記以外で、JSKでは試してないけど「これはキテる」と思われる他の代替手法もございますでしょうか。

mmurooka, 2017-03-11:
https://github.com/fkanehiro/hrpsys-base/pull/1114 だとcatkinがaptでもソース（travis）でも大丈夫なので，
現状のorigin/masterの仕様に近いまま，現状の動かない状態は解決できるはずです．

snozawa, 2017-05-02:
@mmurooka 
これもうOKそうでしょうか。

mmurooka, 2017-05-02:
https://github.com/fkanehiro/hrpsys-base/pull/1114 で解決しましたのでcloseします．

closed 2017-05-02 by mmurooka