fkanehiro/hrpsys-base #1131: [question] What is setWrenches' argument?

130s, 2017-05-04:
https://github.com/fkanehiro/hrpsys-base/pull/434/files#diff-6204f002204dd9ae80f203901f155fa9R44 をみると setWrenches 第一引数の配列の要素は 24 個あります．https://github.com/fkanehiro/hrpsys-base/pull/1120/commits/ac34b71431f6335eabde8c28edd5ecce8371541a で PR 出してますが最後の6つは ref_{force, moment} だとすると，最初の18個は何でしょうか？

```
hcf.seq_svc.setWrenches([0,0,0,0,0,0,
                         0,0,0,0,0,0,
                         0,0,0,0,0,0,
                         10,10,-10,0,0,0,], 2.0);
```

sample robot は関節が29あるようで，arm だけだとしても14なので，18という数字が判りません．
```
term-1$ rtmlaunch hrpsys_ros_bridge samplerobot.launch

term-2$ rostopic echo /joint_states/name -n 1
['RLEG_HIP_R', 'RLEG_HIP_P', 'RLEG_HIP_Y', 'RLEG_KNEE', 'RLEG_ANKLE_P', 'RLEG_ANKLE_R', 'RARM_SHOULDER_P', 'RARM_SHOULDER_R', 'RARM_SHOULDER_Y', 'RARM_ELBOW', 'RARM_WRIST_Y', 'RARM_WRIST_P', 'RARM_WRIST_R', 'LLEG_HIP_R', 'LLEG_HIP_P', 'LLEG_HIP_Y', 'LLEG_KNEE', 'LLEG_ANKLE_P', 'LLEG_ANKLE_R', 'LARM_SHOULDER_P', 'LARM_SHOULDER_R', 'LARM_SHOULDER_Y', 'LARM_ELBOW', 'LARM_WRIST_Y', 'LARM_WRIST_P', 'LARM_WRIST_R', 'WAIST_P', 'WAIST_R', 'CHEST']
$ a=('RLEG_HIP_R', 'RLEG_HIP_P', 'RLEG_HIP_Y', 'RLEG_KNEE', 'RLEG_ANKLE_P', 'RLEG_ANKLE_R', 'RARM_SHOULDER_P', 'RARM_SHOULDER_R', 'RARM_SHOULDER_Y', 'RARM_ELBOW', 'RARM_WRIST_Y', 'RARM_WRIST_P', 'RARM_WRIST_R', 'LLEG_HIP_R', 'LLEG_HIP_P', 'LLEG_HIP_Y', 'LLEG_KNEE', 'LLEG_ANKLE_P', 'LLEG_ANKLE_R', 'LARM_SHOULDER_P', 'LARM_SHOULDER_R', 'LARM_SHOULDER_Y', 'LARM_ELBOW', 'LARM_WRIST_Y', 'LARM_WRIST_P', 'LARM_WRIST_R', 'WAIST_P', 'WAIST_R', 'CHEST')
$ echo ${#a[*]}
29
```

snozawa, 2017-05-09:
これは、6 x センサの数になってます。
SampleRobotの場合は、４つ手足に力センサがVRMLで定義されてるので、6x4=24になっています。
また、センサの順番はVRMLのsensorIdできまります。
この例の場合は、最後の６個の要素はrhsensor（右手）に相当します。
（細かいですが、ホントはヒューマノイドロボットのこの状況の場合足先にも体重分の力がかかるので、バランス的には足先にもref forceが設定してあるのが正しいですが、腕にimpedanceを入れるというサンプルなので今はこうしています）

wrenchの説明は
http://fkanehiro.github.io/hrpsys-base/dd/d91/SequencePlayer.html
のloadPatternのところにもありますが、idlにも書いておいた方がわかりやすかったですね。

130s, 2017-05-09:
@snozawa ありがとうございます．だいぶ頓珍漢な質問をしていたことが判りました．．．

https://github.com/fkanehiro/hrpsys-base/pull/1120 で Python I/F に本件の[説明を足しました](https://github.com/fkanehiro/hrpsys-base/pull/1120/commits/6544fef8d74a27f6a01db787b94c930e560b77ff)．
(Hiro-NEXTAGE ユーザはほとんど／ほぼすべて Python 経由で hrpsys 使うのでそうさせて頂きました)．

closed 2017-05-09 by 130s