fkanehiro/hrpsys-base #1149: SimpleFullbodyInverseKinematicsSolverのm_robotをpublicにしたりエンドエフェクタ位置を取得する関数を追加したい

ishiguroJSK, 2017-06-04:
AutoBalancer.hの
```
class SimpleFullbodyInverseKinematicsSolver
{
private:
    // Robot model for IK
    hrp::BodyPtr m_robot;
```
のm_robotをpublicかpublic constメンバにして頂けると，
このクラスを生成以降，
```
fik->ikp["rleg"].target_p0
とか
fik->m_robot->calcForwardKinematics();
```
みたいに統一的にアクセスできるのですがダメでしょうか？
あと
```
hrp::Vector3 getEEPos(const std::string& limb_name){
  return ikp[limb_name].target_link->p + ikp[limb_name].target_link->R * ikp[limb_name].localPos;
}
hrp::Matrix33 getEERot(const std::string& limb_name){
  return ikp[limb_name].target_link->R * ikp[limb_name].localPos;
}
```
みたいなエンドエフェクタの位置姿勢が取得できるメソッドを追加したいです
(毎度`fik->ikp["rleg"].target_link->p+ fik->ikp["rleg"].target_link->R * fik->ikp["rleg"].localPos`と書くのがしんどいので・・・)

snozawa, 2017-06-06:
> みたいに統一的にアクセスできるのですがダメでしょうか？

これはAutoBalancer.cpp内部のSimpleFullbodyInverseKinematicsSolver以外の
箇所でつかうときに、面倒じゃないようにするということかな？

また、
> みたいなエンドエフェクタの位置姿勢が取得できるメソッドを追加したいです

これもAutoBalancer.cpp内で使うために、AutoBalancerに追加するということかな？
これは追加するので良い気がします
（ただ、微妙にlocalPosの他に色々たしてる箇所もあるので、この関数がそのまま使えない箇所もありそう）

ishiguroJSK, 2017-06-06:
AutoBalancer.hのSimpleFullbodyInverseKinematicsSolverクラスを別のRTCでインクルードして使ったりしているのですが，現在だとSimpleFullbodyInverseKinematicsSolverのインスタンス(へのポインタ)fikとm_robotに対して，欲しい情報によってfik->としたりm_robot->としたりしているのですが，fikの中にm_robotを保持しているので，fik->とfik->m_robot->と統一できたら色々関数の引数に渡したいときにもfik一つで全情報渡せて嬉しいかなぁと思っていました．
getEEPosなどはAutoBalancer.hのSimpleFullbodyInverseKinematicsSolverのクラスメンバに追加すると便利だと思っています

snozawa, 2017-06-06:
> AutoBalancer.hのSimpleFullbodyInverseKinematicsSolverクラスを別のRTCでインクルードして使ったりしているのですが，現在だとSimpleFullbodyInverseKinematicsSolverのインスタンス(へのポインタ)fikとm_robotに対して，欲しい情報によってfik->としたりm_robot->としたりしているのですが，fikの中にm_robotを保持しているので，fik->とfik->m_robot->と統一できたら色々関数の引数に渡したいときにもfik一つで全情報渡せて嬉しいかなぁと思っていました．

まず、他のから使うのであれば、SimpleFullbodyxxxはAutoBalancer.h以外の場所にあるのが良いですね。
もっとも単純にはrtc/AutoBlancer/以下にSimpleFulbodyxxxx.h,cppなどを追加するかんじでしょうか。

それで、fikをつかうプログラムはfikの初期化時に引数でm_robotを与えてると思うので、
そのプログラムではm_robot自体をもってるんだよね？
- 今までは fik/m_robotと記述してた
- 今後は fik/fik->m_robotと記述したい

ということなのかな？わざわざfik->m_robotとするのはかえって面倒ではないかな？
また、fikのメンバを外から積極的に使うのはあまりきれいではないような気がします。

> 色々関数の引数に渡したいときにもfik一つで全情報渡せて嬉しいかなぁと思っていました．

情報をわたすといった部分も、具体的に何がしたいかがわかるとよいのかも

> getEEPosなどはAutoBalancer.hのSimpleFullbodyInverseKinematicsSolverのクラスメンバに追加すると便利だと思っています

このへんの必要そうなメンバは追加するので良い気がします。

ishiguroJSK, 2017-06-06:
fik生成後にfikだけにある情報とm_robotだけにある情報それぞれ初期化したりするときなどに別々の変数に初期化しにいくのが冗長かなぁと思っていましたが，ただの美学なのでやっぱりいいです．
ということは今回の変更はSimpleFulbodyxxxx.hに分離すること(.cppにするほどではないと思っていますが)とgetEEPos関数を追加することでよろしいでしょうか(命名規則なども)？

snozawa, 2017-06-06:
> ということは今回の変更はSimpleFulbodyxxxx.hに分離すること(.cppにするほどではないと思っていますが)とgetEEPos関数を追加することでよろしいでしょうか(命名規則なども)？

そうですね。
前者は.hくらいでも良いかなとは思います。
後者は、(そもそもすでにEEとかいてるプログラムもあるのでアレですが）一応
`getEndEffectorPos`とかの中身のわかる名前でお願いします。

ishiguroJSK, 2017-06-07:
#1150 でマージされたのでcloseします

closed 2017-06-07 by ishiguroJSK