fkanehiro/hrpsys-base #1161: rtcの守るべきマナー(JointAngleController)

YoheiKakiuchi, 2017-06-20:
特に getJointAngleControllerList(self): で返ってくるrtcについては、
qRef(inport)とq(output)を持っていて順次qRefとqをつないでいくことを前提にしていると
思うのですが、守るべきマナーはあるでしょうか？

ぱっと見に思いつくところとしては、以下があります。

1. 中身の機能が働いていない場合は q = qRef を出力する
2. qRefが来ない場合 qを出力しない
3. サービスからenable/disableできる(?)

snozawa, 2017-06-20:
ご指摘ありがとうございます。
実はあまり深く考えてなかった＋議論されてなかった点です。

現状当てはまる＋暗黙のうちに考えてたルールは、おっしゃるものも含めて
- 中身の機能が働いていない場合は q = qRef を出力する
- サービスからenable/disableできる（遷移もちゃんと不連続にならないようにする）
- `hcf.xx.stop()`(つまりonDeactivate)でdisable処理を行う（servoOn時にdisableされてないと、StateHolderのgoActualの後工程でも常に関節角を上がいてしまい、サーボエラーでサーボがはいらないため）

くらいだと思ってます(3つめはRFUが対応してなかったので直します）

> qRefが来ない場合 qを出力しない

これは、RTCによってまちまちですが、ホントはだしたほうが良い気がしてきました。

YoheiKakiuchi, 2017-06-20:
> > qRefが来ない場合 qを出力しない
> これは、RTCによってまちまちですが、ホントはだしたほうが良い気がしてきました。

出すなら、
a. 一度出していたら、前回と同じものを出す、
b. 一度も出していない（初期状態）なら、モデルとjoint lengthが同じで、すべての要素が0のangle-vectorを出力
でしょうか。

a.はそれほど問題になりそうにないですね。
現状では通常運用で、一度値が来て後に来なくなるのは上流側に問題がある場合であって、
その時点で全体的に見ればエラーになっている

b.はactivate直後は必ずこの状態になって、上流から関節角度がこなくても下流に関節角度を流す。
一応、stateholderがactivateされた直後で、かつ、すべてのrtcの立ち上げ前にgoActualされれば
一度も関節角度0のベルトルを扱う必要はなくなるが、現状でも実質そうなっていないので、問題ないように思う。

ちなみに、qRefがこなくてもqを出す場合の利点は何でしょうか？

snozawa, 2017-06-20:
> b.はactivate直後は必ずこの状態になって、上流から関節角度がこなくても下流に関節角度を流す。
> 一応、stateholderがactivateされた直後で、かつ、すべてのrtcの立ち上げ前にgoActualされれば
> 一度も関節角度0のベルトルを扱う必要はなくなるが、現状でも実質そうなっていないので、問題ないように思う。

これ何でしょうか。
activate直後であってもonExecute通常実行時でも、接続が正常にされてる場合は、
qRefには前のRTCがwrite()した結果がはいっていてqに出力されます

それから、超すいません、
> これは、RTCによってまちまちですが、ホントはだしたほうが良い気がしてきました。

これは私の日本語が誤ってました。
RTCによってまちまちですが、qRefが出力されてないときはqを出力しないほうが良いです。
また、qRefが着ていない
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/SoftErrorLimiter/SoftErrorLimiter.cpp#L400-L405
というのは、あくまで設定・接続か何かに不備があり、そうでなければqRefがでてて
上記のようなワーニングがでないです。

ワーニングをだしたほうがよい、とかこうとして誤りました。