fkanehiro/hrpsys-base #1166: stride_parameterの追加に関して

snozawa, 2017-06-23:
@YutaKojio
GaitGeneratorの歩幅制限に関する
https://github.com/fkanehiro/hrpsys-base/pull/1162/files
のＰＲで，昔からあったコードでたぶん（ＪＳＫ内部コードからの）移植時に
何か間違ってた部分を修正しました．

これに伴い，デフォルトで使われてる歩幅制限が
`前，横，旋回，後`
の４つになってるのに加え，横を内側外側にわけようかと思ってます．
ところで，上記if文をみると，内股制限は横（dy）成分だけでなく
旋回(dth)成分も制限してたように思います．

なので，４つの成分でなく
`前，横外側，旋回外側，後，横内側，旋回内側`
と６つのパラメタがいいのではと思ってたりします．

他にも着地位置補正のパラメタなどは５になってると思います．

まずはstride_parameterを４＝＞６にするのはどう思います？

ちなみに，長さをかえるとして，外から４が与えらえても問題なくうごくように
互換性を保つようなコードにはすると思います．

よろしくお願いします．

YutaKojio, 2017-06-23:
> まずはstride_parameterを４＝＞６にするのはどう思います？
> ちなみに，長さをかえるとして，外から４が与えらえても問題なくうごくように
> 互換性を保つようなコードにはすると思います．

今までのコードと互換性があるならパラメータが6つに増えるのはいいと思います．

着地位置補正のパラメタ(```stride_limitation_for_circle_type```) は今の５つのままで大丈夫です．
（もし変更が必要でしたら６つになっても大丈夫です．）

snozawa, 2017-06-26:
ありがとうございます。
それでは6このパラメタにする方向で検討します。

snozawa, 2017-08-04:
Solved via https://github.com/fkanehiro/hrpsys-base/pull/1185.

closed 2017-08-04 by snozawa