fkanehiro/hrpsys-base #1202: hrpsys-simulatorなどの関節トルクについて

snozawa, 2017-09-04:
関節トルクについて質問があります。
(hrpsys-simualtorを使っている点、および中身に関してはopenhrp3/choreonoid双方に関連する質問になりそうでしたので、こちらに失礼します）

hrpsys-simulatorで全部の関節をTORQUEモードで実行してシミュレーションするときに、
(openhrp3プロジェクトファイルに記載したら追加される）JointTorqueのInPort/OutPortは
何のトルクとなっているかという点について質問がございます。

値を見ると、関節トルク（リンク側最終出力のトルク）でなく、（ものによって違うとは思いますが）100倍くらいオーダが違う大きい値がはいっているようでした。

- 関節トルク(joint->u)は、リンク側の最終出力トルクでなく、モータ側の入力トルクのようなものになりますでしょうか（かもしくは、gearRatio, rotoInertiaのかけ合わさった何かしらの値でしょうか）
- もし上記があっている場合、リンク側の最終出力トルクを実測トルクとして取得する方法はございますでしょうか。
- TORQUEモードの場合、指令トルクと出力トルク（つまり、InportのトルクとOutPortのトルク）は同じ値がはいっているという理解でよろしいでしょうか。

よろしくお願いいたします

fkanehiro, 2017-09-04:
いずれも関節のトルク(Link::u)となっています。
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/util/PortHandler.cpp#L162
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/util/PortHandler.cpp#L178
100倍くらいオーダが違うというのは何と何を比較した場合の話になりますでしょうか？

snozawa, 2017-09-05:
(A) 軌道計画時の関節トルク
(B) シミュレーションの力計測値・ロボット状態から逆動力学で求めたトルク
(C) hrpsys-simulatorのトルクのポート（Link::u）のトルク

などをそれぞれ比較しました。
(A)と(B)はだいたい一致しておりましたが、（C)が違ってるようでした。

たとえば、一歩時間0.45[s]の歩行シミュレーションで
（B)の結果
[image]
（C)の結果
[image]
関節エラー値
[image]
で、この例では（B)と（C)が概ね４倍程度違ってそうでした。
関節エラーが大きくなるときにトルクが増えているようで、この例でなくジャンプのような動作では
より大きい違い（ケタが違う）になってました。
また、PDゲインとこのエラー値を掛け合わせたものは（C)とあってるようでした。

なので、（C)のシミュレータから得られるLink::uの値はPDサーボの値がそのままはいっているようで、
関節角度エラーが大きくなると大きいようでした。

またこのシミュレーションは、sample1_bush.wrlで関節トルクモード、足にはブッシュありでおこなっておりました。
（B)での計測力は、ブッシュの奥（ブッシュよりも本体側）なので、もしかしたらこれも関係ありますでしょうか。

fkanehiro, 2017-09-05:
はい、トルク制御モードの場合、Link::uは外から与えたトルク指令そのものになりますね。
(B)の計算がリジッドな足首で、力センサから得た力が床から足平に作用している力として計算しているなら、それなりの差を生む原因にはなるのではないでしょうか。