fkanehiro/hrpsys-base #1207: openhrp3が複数存在するとhrpsysが下位のopenhrp3に依存してしまう

ishiguroJSK, 2017-10-12:
aptのros-indigo-openhrp3 + ~/catkin_ws/src/openhrp3(最新の変更)&hrpsys(最新の変更)
や 
~/ros/indigo_parent/src/openhrp3 + ~/catkin_ws/src/openhrp3(最新の変更)&hrpsys(最新の変更)
など
の構成時にhrpsysをビルドしようとすると下位のopenhrp3に依存しているようで，
openhrp3に入れた最新の変更が反映されずビルドが失敗します．

下位のopenhrp3を消したりすれば所望の動作になります．
catkin configやroscd openhrp3, roscd hrpsysは正しい結果になっています．
心当たりなどありますでしょうか？

k-okada, 2017-10-16:
状況がわかるコンパイル中のメッセージcmakeのlog、足りなければprint文入れてみてください。cmakeではmessage になります

ishiguroJSK, 2017-10-17:
どの部分が下位のopenhrp3を参照しに行っているのか見当もついていないので，
ビルドメッセージそのままですが，リンカが引数の増えたcalcJacobianを参照しに行って失敗しているようです．

```
$ catkin b
==> Expanding alias 'b' from 'catkin b' to 'catkin build'
-------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/indigo
Workspace:                   /home/ishiguro/catkin_ws/debug
-------------------------------------------------------------------
Source Space:       [exists] /home/ishiguro/catkin_ws/debug/src
Log Space:         [missing] /home/ishiguro/catkin_ws/debug/logs
Build Space:        [exists] /home/ishiguro/catkin_ws/debug/build
Devel Space:        [exists] /home/ishiguro/catkin_ws/debug/devel
Install Space:      [unused] /home/ishiguro/catkin_ws/debug/install
DESTDIR:            [unused] None
-------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                                                                                
[build] Updating package table.                                                                                                                           
Starting  >>> catkin_tools_prebuild                                                                                                                       
Finished  <<< catkin_tools_prebuild                [ 1.8 seconds ]                                                                                        
Starting  >>> openhrp3                                                                                                                                    
__________________________________________________________________________________________________________________________________________________________
Warnings   << openhrp3:cmake /home/ishiguro/catkin_ws/debug/logs/openhrp3/build.cmake.000.log                                                             
CMake Warning (dev) in hrplib/hrpCorba/CMakeLists.txt:
  Syntax Warning in cmake code at

    /home/ishiguro/catkin_ws/debug/src/openhrp3/hrplib/hrpCorba/CMakeLists.txt:38:55

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in hrplib/hrpCorba/CMakeLists.txt:
  Syntax Warning in cmake code at

    /home/ishiguro/catkin_ws/debug/src/openhrp3/hrplib/hrpCorba/CMakeLists.txt:44:55

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at server/UtDynamicsSimulator/CMakeLists.txt:15 (link_directories):
  This command specifies the relative path

    sDIMS

  as a link directory.

  Policy CMP0015 is not set: link_directories() treats paths relative to the
  source dir.  Run "cmake --help-policy CMP0015" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/ishiguro/catkin_ws/debug/build/openhrp3; catkin build --get-env openhrp3 | catkin env -si  /usr/bin/cmake /home/ishiguro/catkin_ws/debug/src/openhrp3 --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/ishiguro/catkin_ws/debug/devel; cd -
[... 714 lines cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [bin/StabilizerComp] エラー 1
make[1]: *** [rtc/Stabilizer/CMakeFiles/StabilizerComp.dir/all] エラー 2
CMakeFiles/AutoBalancerComp.dir/AutoBalancer.cpp.o: 関数 `FullbodyInverseKinematicsSolver::solveFullbodyIK(boost::shared_ptr<hrp::Body>, std::vector<IKConstraint, std::allocator<IKConstraint> > const&)' 内:
AutoBalancer.cpp:(.text._ZN31FullbodyInverseKinematicsSolver15solveFullbodyIKEN5boost10shared_ptrIN3hrp4BodyEEERKSt6vectorI12IKConstraintSaIS6_EE[_ZN31FullbodyInverseKinematicsSolver15solveFullbodyIKEN5boost10shared_ptrIN3hrp4BodyEEERKSt6vectorI12IKConstraintSaIS6_EE]+0x7ee): `hrp::JointPath::calcJacobian(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const' に対する定義されていない参照です
CMakeFiles/AutoBalancerComp.dir/__/ImpedanceController/JointPathEx.cpp.o: 関数 `hrp::JointPathEx::calcInverseKinematics2Loop(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, double, double, double, Eigen::Matrix<double, -1, 1, 0, -1, 1> const*)' 内:
JointPathEx.cpp:(.text+0x5efb): `hrp::JointPath::calcJacobian(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const' に対する定義されていない参照です
JointPathEx.cpp:(.text+0x60d8): `hrp::JointPath::calcJacobian(Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const' に対する定義されていない参照です
collect2: error: ld returned 1 exit status
make[2]: *** [bin/AutoBalancerComp] エラー 1
[... cut ...]
[build]   Abandoned: None.                                                                                                                                
[build]   Failed:    1 packages failed.                                                                                                                   
[build] Runtime: 1 minute and 28.1 seconds total.                                                                                                         
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

ishiguroJSK, 2017-10-17:
hrpsys/CMakeLists.txtの
```
# OpenHRP                                                                                                                                                 
find_package(OpenHRP REQUIRED)
include_directories(${OPENHRP_INCLUDE_DIRS})                                                                                                            
link_directories(${OPENHRP_LIBRARY_DIRS}) 
```
で
```
message(STATUS " ${OPENHRP_INCLUDE_DIRS}")                                                            
message(STATUS " ${OPENHRP_LIBRARY_DIRS}")
```
として`cmake .`すると
```
--  ;/opt/ros/indigo/include ;/opt/ros/indigo/include/coil-1.1 ;/opt/ros/indigo/include/openrtm-1.1 ;/opt/ros/indigo/include/openrtm-1.1/rtm/idl ;/home/ishiguro/catkin_ws/debug/devel/include/OpenHRP-3.1 ;/usr/include/eigen3
--  ;/usr/lib/atlas-base/atlas;;/opt/ros/indigo/lib;;/home/ishiguro/catkin_ws/debug/devel/lib
```
と表示されて，この順序でいいんだっけ？と思って
```
include_directories(/home/ishiguro/catkin_ws/debug/devel/include/OpenHRP-3.1 ;/opt/ros/indigo/include ;/opt/ros/indigo/include/coil-1.1 ;/opt/ros/indigo/\
include/openrtm-1.1 ;/opt/ros/indigo/include/openrtm-1.1/rtm/idl ;/usr/include/eigen3)
link_directories(/home/ishiguro/catkin_ws/debug/devel/lib; /usr/lib/atlas-base/atlas;;/opt/ros/indigo/lib)
```
と順番を変えて試してみたら一応ビルドは通りました
openhrp3のpkg-configの不備？

ishiguroJSK, 2017-10-17:
以下のopenhrp3/openhrp3.1.pc.in
```
prefix=@OPENHRP_DIR@
exec_prefix=${prefix}/bin
libdir=${prefix}/lib
includedir=-I${prefix}/@RELATIVE_HEADERS_INSTALL_PATH@ -I@EIGEN_INCLUDE_DIR@
idl_dir=${prefix}/@RELATIVE_IDL_INSTALL_PATH@
link_shared_files=@PKG_CONF_LINK_SHARED_FILES@
link_static_files=@PKG_CONF_LINK_STATIC_FILES@
link_depend_dirs=@PKG_CONF_LINK_DEPEND_DIRS@
link_depend_files=@PKG_CONF_LINK_DEPEND_FILES@
link_depend_options=@PKG_CONF_LINK_DEPEND_OPTS@
cflag_defs=@PKG_CONF_DEFS@
cflag_options=@PKG_CONF_CXXFLAG_OPTIONS@

Name: @PACKAGE_NAME@
Description: @OPENHRP_SHORT_DESCRIPTION@
URL: @OPENHRP_URL@
Version: @OPENHRP_VERSION_MAJOR@.@OPENHRP_VERSION_MINOR@.@OPENHRP_VERSION_MICRO@
Requires: @PKG_CONF_REQUIRES@
Libs: ${link_depend_dirs} -L${libdir} ${link_shared_files} ${link_depend_options} ${link_depend_files}
Libs.private: ${link_static_files}
Cflags: ${cflag_defs} ${cflag_options} ${includedir}
```
が展開されると以下のdevel/lib/pkgconfig/openhrp3.1.pcとして展開されるので，
```
prefix=/home/ishiguro/catkin_ws/debug/devel
exec_prefix=${prefix}/bin
libdir=${prefix}/lib
includedir=-I${prefix}/include/OpenHRP-3.1 -I/usr/include/eigen3
idl_dir=${prefix}/share/OpenHRP-3.1/idl
link_shared_files=-lhrpCollision-3.1 -lhrpModel-3.1 -lhrpPlanner-3.1 -lhrpUtil-3.1
link_static_files=-lhrpCorbaStubSkel-3.1
link_depend_dirs=-L/usr/lib/atlas-base/atlas -L/usr/lib/x86_64-linux-gnu -L/opt/ros/indigo/lib
link_depend_files=-lblas -llapack -lboost_filesystem -lboost_signals -lboost_program_options -lboost_regex -ljpeg -lpng -lz -luuid -ldl -lp\
thread -lomniORB4 -lomnithread -lomniDynamic4 -lRTC -lcoil
link_depend_options=-export-dynamic
cflag_defs=
cflag_options=-Wall -fPIC -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -O2 -I/opt/ros/indigo/inclu\
de -I/opt/ros/indigo/include/coil-1.1 -I/opt/ros/indigo/include/openrtm-1.1 -I/opt/ros/indigo/include/openrtm-1.1/rtm/idl

Name: OpenHRP3.1
Description: Platform software development of robots and simulations.
URL: http://www.openrtp.jp/openhrp3/
Version: 3.1.9
Requires:
Libs: ${link_depend_dirs} -L${libdir} ${link_shared_files} ${link_depend_options} ${link_depend_files}
Libs.private: ${link_static_files}
Cflags: ${cflag_defs} ${cflag_options} ${includedir}
```
この時点で`Libs: ${link_depend_dirs} -L${libdir}`の部分などで/optと~/catkin_wsが入れ替わっているので，
```
Libs: -L${libdir} ${link_depend_dirs} ${link_shared_files} ${link_depend_options} ${link_depend_files}
Libs.private: ${link_static_files}
Cflags: ${cflag_defs} ${includedir} ${cflag_options}
```
とすることで正解ではないでしょうか？

fkanehiro, 2017-10-17:
なるほど、良さそうですね。

k-okada, 2017-10-17:
すばらしい．PRを送ってください．

ishiguroJSK, 2017-10-18:
マージされたのでcloseします．
対応ありがとうございました．

closed 2017-10-18 by ishiguroJSK