fkanehiro/hrpsys-base #1243: RT-Linux vs ART-Linux

yoshimalucky, 2018-05-15:
質問失礼します。

足首に6軸力センサを追加した以下のロボットにRTM-ROSシステムを導入するプロジェクトを進めています
https://github.com/ROBOTIS-GIT/ROBOTIS-OP3

実機の体内ではLinux-Mint(Ubuntu16.04互換?)が動いていて、
hrpsys-baseを入れる前にRT-LinuxとART-Linuxどちらを入れるべきか迷っていますが、
どちらがいいでしょうか？

こちらのインストール手順にもリアリタイムOSについては全く触れられていないので、
どのように自分のシステムに合わせて選べばいいかという手がかりが得られずにいます。
https://raw.githubusercontent.com/fkanehiro/hrpsys-base/master/INSTALL

いくつか話を見聞きした限りでは、RT-Linuxの方が早い制御周期が使われているので、
RT-Linuxの方がいいのかと漠然と思ってはいますが、その理由がわからないので、
安易に入れてもいいか迷っています。

お手数ですが、どなたか知見をお持ちの方がいらっしゃれば、ご回答お願いいたします。

yoshimalucky, 2018-05-15:
ここ (https://github.com/fkanehiro/hrpsys-base/tree/master/ec/hrpEC) に

hrpEC (RT-Linux向け?
hrpEC-art (ART-Linux用

というものがあるので、これらを選ぶ際の指標みたいなものがあるのかという質問です

fkanehiro, 2018-05-15:
ART-Linuxはすでにメンテされていないと思いますので、検討の対象とならないと思います。
もう一つの方はPOSIXのAPIを呼んでいるだけですので、low-latencyカーネルやRT-Preemptパッチが当たったカーネルで使用することができます。
リアルタイム性が重要であればRT-Preemptカーネルでしょうか。
https://wiki.linuxfoundation.org/realtime/documentation/howto/applications/preemptrt_setup

7675t, 2018-05-15:
対象システムについてはよく知らないのですが，low-latencyだと結局ジッタの最大値は大きいままなので，それが問題になる場合には制御周期が比較的長い(>10ms)場合でも，PREEMPT_RTパッチがオススメです．

ちなみに，"RT-Linux"と書くと，20年ぐらい前に存在した製品"RT-Linux"とかぶるので，"PREEMPT_RTパッチ"と書いた方が混乱しないと思います．

yoshimalucky, 2018-05-16:
ここでいうRT-Linuxというのは、おそらく、PREEMPT_RTのことで合っていると思います。
自分もRT-Linuxでググっても、@7675t 氏がおっしゃるようなものが出てきてしまって、有益な情報が得られませんでした。
研究室の内部資料にARTとRTの制御周期とECについて書かれていて、略記によってこのように混乱しましたが、RTはPREEMPT_RTのことを指しているのかと思います。

ひとまず、ここで紹介されているリンク先のものを使わせていただこうと思います。
ご回答ありがとうございました。

closed 2018-05-16 by yoshimalucky