fkanehiro/hrpsys-base #1250: <setJointAngles/playPattern>OfGroupを一度呼ぶとsetJointAnglesで動かせなくなる

pazeshun, 2018-06-18:
Euslispにおいて、`(send *ri* :angle-vector (send *robot* :angle-vector) 5000 :ctype :rarm-controller)`という様なcontroller指定の指令を一度でも送ってしまうと、その後`(send *ri* :angle-vector (send *robot* :angle-vector) 5000)`の様な:default-controllerの指令を送っても動かなくなる問題の元原因がこれです。

k-okada, 2018-06-19:
これは，hrpsysのプラグインの仕組として，
- SequencePlayer (全身）
が基本的には動いていて，その後，GroupのSequencerを起動すると
- SequencePlayer (全身） >  SequencePlayer (上半身）
と後に作られたgroupが上書きされます．
新しく出来た上半身のシーケンサを外したければ以下のコードを呼ぶ必要が有ります．
https://github.com/fkanehiro/hrpsys-base/blob/master/idl/SequencePlayerService.idl#L153

pazeshun, 2018-06-19:
なるほど、理解しました。
Euslispの方を治すには、
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/HrpsysJointTrajectoryBridge.cpp#L489
のようにjoint trajectory action経由でfullbodyの指令を送る部分で、groupが上書きされていた場合は`removeJointGroup`を呼べばよい気がするので、やってみます

k-okada, 2018-06-19:
> hrpsys_ros_bridge/src/HrpsysJointTrajectoryBridge.cpp#L489

いや，これはユーザが明示的にremoveするのでいいとおもいます．

pazeshun, 2018-06-19:
> いや，これはユーザが明示的にremoveするのでいいとおもいます．

なるほど、https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/euslisp/rtm-ros-robot-interface.l#L438-L442
を使って、
```
(send *ri* :remove-joint-group "rarm")
```
とするということですかね。

iory, 2018-06-19:
途中から失礼します。

> いや，これはユーザが明示的にremoveするのでいいとおもいます．

これはどういった意図でしょうか？

k-okada, 2018-06-19:
音に応じて首だけ動かすプログラムなどが走っている状況を想定しています

iory, 2018-06-19:
> 音に応じて首だけ動かすプログラムなどが走っている状況を想定しています

なるほど、わかりました。

pazeshun, 2018-06-19:
> 音に応じて首だけ動かすプログラムなどが走っている状況を想定しています

なるほど。
GroupのSequencerを起動したあとの挙動について、全身のSequencePlayerに指令を送っても全軸動かないと誤解していましたが、他のgroupについては動かせることがわかりました。

ありがとうございました。

closed 2018-06-19 by pazeshun